IDEC-ezWheel/swd_ros_controllers #5: Failure to start swd_diff_drive_controller

fig-shimooka, 2022-03-16:
I am trying to get the SWD starter kit setup on a x86_64 PC that I have prepared myself.
I want to start swd_diff_drive_controller following the instructions in README.md.
When I run it, I get the following error
What is the cause of this?
Perhaps it is because the service file(ezw-dbus-user-session, ezw-swd-left, etc...) is not running, but where can I get the service file from?

```
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://x86:42357/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /swd_diff_drive_controller/base_frame: base_link
 * /swd_diff_drive_controller/baseline_m: 0.485
 * /swd_diff_drive_controller/command_timeout_ms: 500
 * /swd_diff_drive_controller/control_mode: Twist
 * /swd_diff_drive_controller/have_backward_sls: False
 * /swd_diff_drive_controller/left_swd_config_file: /opt/ezw/usr/etc/...
 * /swd_diff_drive_controller/odom_frame: odom
 * /swd_diff_drive_controller/positive_polarity_wheel: Right
 * /swd_diff_drive_controller/pub_freq_hz: 50
 * /swd_diff_drive_controller/publish_odom: True
 * /swd_diff_drive_controller/publish_safety_functions: True
 * /swd_diff_drive_controller/publish_tf: True
 * /swd_diff_drive_controller/right_swd_config_file: /opt/ezw/usr/etc/...
 * /swd_diff_drive_controller/wheel_max_speed_rpm: 95
 * /swd_diff_drive_controller/wheel_safety_limited_speed_rpm: 40

NODES
  /
    swd_diff_drive_controller (swd_ros_controllers/swd_diff_drive_controller)

ROS_MASTER_URI=http://localhost:11311

process[swd_diff_drive_controller-1]: started with pid [51043]
[ INFO] [1647417727.125643313]: Ready !
[ INFO] [1647417727.130635324]: Initializing swd_diff_drive_controller, node name : /swd_diff_drive_controller
[ INFO] [1647417727.152577877]: Motors config files, right : /opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini, left : /opt/ezw/usr/etc/ezw-smc-core/swd_left_config.ini
[default] contextId = 12
[default] nodeId = 5
[default] HWConfigurationEntry = SWD_RIGHT
[default] HWConfigurationFile = /opt/ezw/data/configuration.json
[default] SWCompatibilityFile = /opt/ezw/firmwares/compatibility.json
[default] CANOpenEDSFile = /opt/ezw/usr/etc/ezw-canopen-dico/swd_core.eds
[default] targetVelocityTimeoutMs = 300
[default] maxMaintenanceSpeedValue = 400
2022-3-16 17:2:7;;SMCDS;WARN;load;55;Config;Invalid type for canDevice => can0 by default;
[default] contextId = 12
[default] nodeId = 5
[default] HWConfigurationEntry = SWD_RIGHT
[default] HWConfigurationFile = /opt/ezw/data/configuration.json
[default] SWCompatibilityFile = /opt/ezw/firmwares/compatibility.json
[default] CANOpenEDSFile = /opt/ezw/usr/etc/ezw-canopen-dico/swd_core.eds
[default] targetVelocityTimeoutMs = 300
[default] maxMaintenanceSpeedValue = 400
2022-3-16 17:2:7;;SMCDS;INFO;getHWConfigurationMember;90;HardwareHelper;Search for entry SWD_RIGHT / nbStepRevolutionElectype Integer into /opt/ezw/data/configuration.json;
2022-3-16 17:2:7;;SMCDS;INFO;getHWConfigurationMember;111;HardwareHelper;nbStepRevolutionElec : 6;
2022-3-16 17:2:7;;SMCDS;INFO;getHWConfigurationMember;90;HardwareHelper;Search for entry SWD_RIGHT / nbPolePairtype Integer into /opt/ezw/data/configuration.json;
2022-3-16 17:2:7;;SMCDS;INFO;getHWConfigurationMember;111;HardwareHelper;nbPolePair : 5;
2022-3-16 17:2:7;;SMCDS;INFO;getHWConfigurationMember;145;HardwareHelper;Search for entry SWD_RIGHT / reduction type Double into /opt/ezw/data/configuration.json;
2022-3-16 17:2:7;;SMCDS;INFO;getHWConfigurationMember;166;HardwareHelper;reduction : 14.000000;
2022-3-16 17:2:7;;SMCDS;INFO;getHWConfigurationMember;145;HardwareHelper;Search for entry SWD_RIGHT / diameter type Double into /opt/ezw/data/configuration.json;
2022-3-16 17:2:7;;SMCDS;INFO;getHWConfigurationMember;166;HardwareHelper;diameter : 125.000000;
2022-3-16 17:2:7;;SMCDS;INFO;getHWConfigurationMember;200;HardwareHelper;Search for entry SWD_RIGHT / device type String into /opt/ezw/data/configuration.json;
2022-3-16 17:2:7;;SMCDS;INFO;getHWConfigurationMember;221;HardwareHelper;device : /dev/ttyUSBY;
2022-3-16 17:2:7;;APP;INFO;init;76;DBusClientDelegate;Checking DBUS proxy availability : commonapi.ezw.canopenservice;
2022-3-16 17:4:7;;APP;ERROR;init;83;DBusClientDelegate;Checking DBus proxy availability timeout (120s) for proxy commonapi.ezw.canopenservice;
[ERROR] [1647417847.246376733]: Failed initializing right motor, CONTEXT_ID: 12, EZW_ERR: SMCService : COSDBusClient::init() return error code : 5
2022-3-16 17:4:7;;APP;ERROR;setTargetVelocity;1037;SMC Controller is not initialized;
2022-3-16 17:4:7;;APP;ERROR;setTargetVelocity;1037;SMC Controller is not initialized;
[ERROR] [1647417847.257613902]: FATAL ERROR, exception 'Failed initializing right motor'
[swd_diff_drive_controller-1] process has died [pid 51043, exit code 1, cmd /home/fig/ros1_swd/install/lib/swd_ros_controllers/swd_diff_drive_controller __name:=swd_diff_drive_controller __log:=/home/fig/.ros/log/4b4eadea-a4f9-11ec-b2e5-ad701015f6b0/swd_diff_drive_controller-1.log].
log file: /home/fig/.ros/log/4b4eadea-a4f9-11ec-b2e5-ad701015f6b0/swd_diff_drive_controller-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

GMezWheel, 2022-03-18:
It seems that **ezw-canopen.service** is not started.
```
2022-3-16 17:2:7;;APP;INFO;init;76;DBusClientDelegate;Checking DBUS proxy availability : commonapi.ezw.canopenservice;
2022-3-16 17:4:7;;APP;ERROR;init;83;DBusClientDelegate;Checking DBus proxy availability timeout (120s) for proxy commonapi.ezw.canopenservice;
[ERROR] [1647417847.246376733]: Failed initializing right motor, CONTEXT_ID: 12, EZW_ERR: SMCService : COSDBusClient::init() return error code : 5
```
`> systemctl start ezw-canopen`

_NOTA : the dependency with ezw-stack has to be removed from **ezw-canopen** systemctl service._

This dbus service allows **swd_ros_controller** to communicate using CANopen with the SWD motors (left and right).

At minimal, the following systemctl service have to be started:
```
> systemctl -a | grep ezw
ezw-dbus-user-session (optional if a dbus session already exists. eg graphical dbus session)
ezw-canopen
````

If the previous systemctl ezw services are not installed in the system, maybe the **swd-starter-kit** ubuntu package is not installed:
`> sudo apt install swd-starter-kit`

The dbus session information created by  **ezw-dbus-user-session** service shall be loaded like that.
`> export $(cat /opt/ezw/data/tmp/SYSTEMCTL_dbus.id)`

fig-shimooka, 2022-03-18:
Thank you very much.
I did exactly what you said and it worked.

stakakusu, 2022-06-14:
>Also, the service file and the shell script that runs in it (/opt/sbi/*.sh) do not exist after installing sed-service.
>Therefore, I copied them from the IPC in the SWC starter kit.
for whom may don't have access to starter-kit package, I attached zipped "/opt/sbin" folder
[sbin.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/8899429/sbin.zip)
.

GMezWheel, 2022-06-14:
Hi @stakakusu,
The sbin.zip and services.zip files seem to be OK.

At minimal, swd_diff_drive_controller needs that the following services have to be started : 
* ezw-canopen.service
* ezw-dbus-user-session.service
* ezw-swd-left.service
* ezw-swd-right.service

see #7

suguruhamada1003910, 2022-07-21:
[SWD_SK_Data.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/9157371/SWD_SK_Data.zip)

Necessary files are uploaded.