IDEC-ezWheel/swd_ros_controllers #7: The power of the right motor is transmitted to the left motor.

fig-shimooka, 2022-03-18:
I would like to run the SWD kit with x86-64 CPU Board.
For some reason, when I send any value to angular.z in cmd_vel, it only goes straight ahead and backward (no error is generated).
I cannot make a turn.
It seems that the indicated value of the right motor is being transmitted to the left motor.

Also, when I use swd_left_4_commissioning.py, I get an error (same for Right).
The ezw-swd-left is running.

-------------------
$ python3 swd_left_4_commissioning.py
Traceback (most recent call last):
File "/usr/lib/python3/dist-packages/dbus/bus.py", line 177, in activate_name_owner
return self.get_name_owner(bus_name)
File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
return self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'commonapi.SmcService.NMT.v1_0_commonapi.ezw.smcservice.swd_left': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "swd_left_4_commissioning.py", line 178, in
main(sys.argv[1:])
File "swd_left_4_commissioning.py", line 100, in main
commissioning.create_dbus_clients(instance_id)
File "/opt/ezw/commissioning/commissioning.py", line 196, in create_dbus_clients
nmt_client = NMTDBusClient(instance_id)
File "/opt/ezw/usr/lib/smcdbusclient/nmt.py", line 49, in init
self.dbus_object = self.bus.get_object(
File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
return self.ProxyObjectClass(self, bus_name, object_path,
File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 250, in init
self._named_service = conn.activate_name_owner(bus_name)
File "/usr/lib/python3/dist-packages/dbus/bus.py", line 182, in activate_name_owner
self.start_service_by_name(bus_name)
File "/usr/lib/python3/dist-packages/dbus/bus.py", line 277, in start_service_by_name
return (True, self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name commonapi.SmcService.NMT.v1_0_commonapi.ezw.smcservice.swd_left was not provided by any .service files

---------------------------

GMezWheel, 2022-03-18:
For the commissionin, does the service of ezw-canopen is started ?

At minimal, the following systemctl service have to be started for commissioning:
```
> systemctl -a | grep ezw
ezw-dbus-user-session (optional if a dbus session already exists. eg graphical dbus session)
ezw-stack
ezw-canopen
ezw-swd-left
ezw-swd-right
````

fig-shimooka, 2022-03-18:
Yes, those services have been activated.

  ezw-canopen.service                                                                                        loaded    active     running            EZW CANOPEN SERVICE                              
  ezw-dbus-user-session.service                                                                              loaded    active     running            EZW DBUS USER SESSION SERVICE                    
  ezw-stack.service                                                                                          loaded    active     running            EZW APP STARTUP - PARENT SERVICE                 
  ezw-swd-left.service                                                                                       loaded    active     running            EZW SWD LEFT SERVICE                             
  ezw-swd-right.service                                                                                      loaded    active     running            EZW SWD RIGHT SERVICE

GMezWheel, 2022-03-18:
Have you export the dbus session information created by ezw-dbus-user-session service before start swd_ros_controller ?
> export $(cat /opt/ezw/data/tmp/SYSTEMCTL_dbus.id)

fig-shimooka, 2022-03-18:
SYSTEMCTL_dbus.id file has not been created.
I created it because there was no tmp folder. I then rebooted, but it is not created.

fig-shimooka, 2022-03-18:
Some files are taken from the arm version because "apt get installswd-starter-kit" failed to run on x86-64.

GMezWheel, 2022-03-18:
Can you check EZTMP variable into /opt/ezw/lib/function.res ?

I think that the session dbus is into /tmp and not into /opt/ezw/data/tmp.

Thus, the correct command is :
`export $(cat /opt/ezw/data/tmp/SYSTEMCTL_dbus.id)`

fig-shimooka, 2022-03-18:
EZTMP was allocated /tmp.
The cat /opt/ezw/sbin/sce-ros-bringup.sh file was exported correctly.

```
#!/bin/bash

export $(cat /tmp/SYSTEMCTL_dbus.id)

# Add this line to your local ~/.swd_ros_env to be able to
# access ROS topics from external machine on WiFi
# export ROS_IP=10.10.0.1
SWD_ROS_ENV="$HOME/.swd_ros_env"
SWD_USER_WS_DIR="$HOME/ros1_swd/"
SWD_BRINGUP_PKG="swd_starter_kit_bringup"
SWD_BRINGUP_LAUNCH_FILE="starter_kit.launch"

# Load custom environnement settings if any
if [[ -f $SWD_ROS_ENV ]]; then
    echo "Found custom environnement config at $SWD_ROS_ENV"
    source $SWD_ROS_ENV
fi

source /opt/ros/noetic/setup.bash

# Load the custom workspace if specified
CUSTOM_SETUP="$SWD_USER_WS_DIR/devel/setup.bash"

if [[ -f $CUSTOM_SETUP ]]; then
    source $CUSTOM_SETUP
fi
export ROS_MASTER_URI=http://10.0.0.2:11311
export ROS_HOSTNAME=10.0.0.2
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opt/ezw/usr/lib

roslaunch $SWD_BRINGUP_PKG $SWD_BRINGUP_LAUNCH_FILE

```

GMezWheel, 2022-03-18:
Logs files are into /tmp/sce-*.log

Is everything OK into canopen, swd-left and swd-right log files ?

eg:
```
`$ cat /tmp/sce-swd-right.log
##########################
Starting Release Version : 0.1.2_swd_sk
2022-03-14 04:21:27
##########################
loading ini variables ...                                                        OK 
Run on ROBOT With Role Master  : Starting 

FOUND /tmp/SYSTEMCTL_dbus.id
Exporting following variables :
DBUS_SESSION_BUS_ADDRESS=unix:abstract=/tmp/dbus-riZKNCFCau,guid=11e440353d8bd549678cfa5e622efaf8
DBUS_SESSION_BUS_PID=691
```

fig-shimooka, 2022-03-18:
The path for SYSTEMCTL_dbus.id in the commissioning.py file was incorrect.
After correcting this, swd_left_4_commissioning.py could be executed.
However, the AMR can still only go straight ahead.

```

SYSTEMCTL_dbus.id detected
create_dbus_clients(swd_left) : OK
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=287, can_id2=288)
SRDOParameters(valid=False, sct=25, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=259, can_id2=260)
SRDOParameters(valid=True, sct=255, srvt=100, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=265, can_id2=266)
SRDOParameters(valid=False, sct=25, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=263, can_id2=264)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=265, can_id2=266)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=267, can_id2=268)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=269, can_id2=270)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=271, can_id2=272)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=273, can_id2=274)
SRDOParameters(valid=True, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=352, can_id2=353)
fig@x86:/opt/ezw/commissioning$ python3 swd_right_5_commissioning.py
SYSTEMCTL_dbus.id detected
create_dbus_clients(swd_right) : OK
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=287, can_id2=288)
SRDOParameters(valid=False, sct=25, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=259, can_id2=260)
SRDOParameters(valid=True, sct=255, srvt=100, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=352, can_id2=353)
SRDOParameters(valid=False, sct=25, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=263, can_id2=264)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=265, can_id2=266)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=267, can_id2=268)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=269, can_id2=270)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=271, can_id2=272)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=273, can_id2=274)
SRDOParameters(valid=True, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=265, can_id2=266)
```

GMezWheel, 2022-03-18:
OK. Nice !

Can you check that the both motors have a CANopen nodeid different ?
*  left : 4
* right : 5

```swd_sk@nuvo: ~ $ candump can0 | grep 70
  can0  705   [1]  7F
  can0  704   [1]  7F
  can0  705   [1]  7F
  can0  704   [1]  7F
...
```

fig-shimooka, 2022-03-18:
Thank you!
This is the result.

```

fig@x86:/opt/ezw/commissioning$ candump can0 |grep 70
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
```

GMezWheel, 2022-03-18:
OK. 
701 corresponds to canopen master node.

Can you echo your cmd_vel message ?
`$ ros topic echo cmd_vel`

fig-shimooka, 2022-03-18:
The nodeId in left_drive_config.ini was 16.
Is this the cause?
$ cat /opt/ezw/usr/etc/ezw-smc-core/left_drive_config.ini
# SMC Drive service config file
contextId = 12
nodeId = 16

$ cat /opt/ezw/usr/etc/ezw-smc-core/swd_left_config.ini
# SMC Drive service config file
contextId = 12
nodeId = 4

linear:
  x: -0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.6316610276699066
---
linear:
  x: -0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.6316610276699066
---
linear:
  x: -0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.6316610276699066

Candump results when cmd_vel->angular.z is fluctuating
$ candump can0 |grep 70
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05
  can0  705   [1]  05
  can0  385   [8]  27 00 70 03 00 00 44 00
  can0  701   [1]  05
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  384   [8]  27 00 70 04 00 00 44 00
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  384   [8]  27 00 70 04 00 00 44 00
  can0  384   [8]  27 00 70 04 00 00 44 00
  can0  384   [8]  27 00 70 04 00 00 44 00
  can0  384   [8]  27 00 70 04 00 00 44 00
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  484   [8]  27 00 70 FF FF FF 44 00
  can0  485   [8]  27 00 70 FF FF FF 44 00
  can0  485   [8]  27 00 70 FF FF FF 44 00
  can0  484   [8]  27 00 70 FF FF FF 44 00
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  385   [8]  27 00 70 03 00 00 44 00
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  384   [8]  27 00 70 03 00 00 44 00
  can0  704   [1]  05
  can0  701   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05

GMezWheel, 2022-03-18:
Yes, your rigths.

NodeId shall be 4 into /opt/ezw/usr/etc/ezw-smc-core/left_drive_config.ini

fig-shimooka, 2022-03-18:
Renumbered.
Result remains the same, cannot turn

  can0  705   [1]  05
  can0  384   [8]  27 00 70 00 00 00 44 00
  can0  385   [8]  27 00 70 00 00 00 44 00
  can0  701   [1]  05
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  384   [8]  27 00 70 00 00 00 44 00
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05
  can0  385   [8]  27 00 70 FF FF FF 44 00
  can0  701   [1]  05
  can0  384   [8]  27 00 70 FF FF FF 44 00
  can0  705   [1]  05
  can0  701   [1]  05
  can0  704   [1]  05
  can0  705   [1]  05

GMezWheel, 2022-03-18:
Normal, I think the good config file name is : swd_left_config.ini

fig-shimooka, 2022-03-18:
I will send you my ezw folder.
I checked, only swd_left_config.ini was used.
[ezw.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/8303991/ezw.zip)

fig-shimooka, 2022-03-18:
I want to check the data of the operating volume sent to the left and right motors (I want to know if the same data is sent to the left and right motors).
Can I check this in candump?

GMezWheel, 2022-03-18:
Yes. 
can-id : 504 (left) et 505 (right)

GMezWheel, 2022-03-18:
In the swd_sk user home directory, there is a remote.py script.

You can connect to the swd-left dbus session with :
`$ ./remote.py swd_left`

fig-shimooka, 2022-03-18:
Thank you.
I have seen the data.
It appears that the same data is being sent in both cases when I turn the wheel.
[image]

Remote.py results.
/opt/ezw/usr/sbin$ ./remote.py swd_left
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 721485826 .................................................................... [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 0.9.0 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : OPERATION_ENABLED  ........................................................... [OK]
NbError              : 1 ............................................................................ [OK]
LastError            : 4210 : Excess_temperature_device ............................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, True, True, True, True, True, True  .............................. [OK]
Peripheral           : True, True, True, True, True, True  .......................................... [OK]
VelocityModeSW       : internal_limit_active: True .................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 147 .......................................................................... [OK]
OdometryValue        : 137 .......................................................................... [OK]
<COM> Switch NodeID = i Restore param = j
<NMT> RESET_NODE = r
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Diagnostic> a
<SW> s
<quit> 0

/opt/ezw/usr/sbin$ ./remote.py swd_right
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 724434945 .................................................................... [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 0.9.0 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : OPERATION_ENABLED  ........................................................... [OK]
[... cut ...]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 48 ........................................................................... [OK]
OdometryValue        : 44 ........................................................................... [OK]
<COM> Switch NodeID = i Restore param = j
<NMT> RESET_NODE = r
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Diagnostic> a
<SW> s
<quit> 0

GMezWheel, 2022-03-18:
Can you send me the version of all your debian ezw packages ?
I will build a similar config to check the problem ?

fig-shimooka, 2022-03-18:
The version of swd-services is available here.
swd-services:
  Installed: 0.2.0
  Candidate: 0.2.0
  Version table:
 *** 0.2.0 500
        500 http://packages.ez-wheel.com:8081/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status

swd_ros_controllers and swd_starter_kit_bringup got the latest source code from git.

I would like it to be composed if you don't mind.

fig-shimooka, 2022-03-18:
Source code for ros_ws is available here.
[ros1_swd.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/8304230/ros1_swd.zip)

GMezWheel, 2022-03-18:
Thanks. I'll be able to test your environment.

fig-shimooka, 2022-03-18:
Thank you. Very much appreciated!

GMezWheel, 2022-03-18:
Can you check /opt/ezw/usr/etc/ezw-canopen-service config files ?
=> these files seems differents between ezw.zip and ros1_swd.zip.

The good ones are the following: 
* ros1_swd.zip\ros1_swd\ezw\usr\etc\LinuxMaster.csv => OK
* ros1_swd.zip\ros1_swd\ezw\usr\etc\nodes_som.csv => OK
* ros1_swd.zip\ros1_swd\ezw\usr\etc\ezw-canopen-service : pdo_mapping.csv => OK

fig-shimooka, 2022-03-19:
The /opt/ezw/usr/etc folder is here.
All three csv files were different.
The ros1_swd.zip/ros1_swd/ezw/usr/etc/ezw-canopen-service folder was overwritten with /opt/ezw/usr/etcezw-canopen-service.
The system was then rebooted, but the result remains the same.

[etc.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/8308579/etc.zip)

```
## candump can0 | grep 70
  can0  701   [1]  05
  can0  704   [1]  05
  can0  705   [1]  05
  can0  701   [1]  05

## candump can0 |grep 50
  can0  505   [4]  0F 00 E4 FE
  can0  504   [4]  0F 00 E4 FE
  can0  505   [4]  0F 00 E4 FE
  can0  504   [4]  0F 00 E4 FE
  can0  505   [4]  0F 00 E4 FE
  can0  504   [4]  0F 00 E4 FE
  can0  505   [4]  0F 00 E4 FE
  can0  504   [4]  0F 00 E4 FE
  can0  505   [4]  0F 00 E4 FE
  can0  504   [4]  0F 00 E4 FE
  can0  505   [4]  0F 00 E4 FE
  can0  504   [4]  0F 00 E4 FE

---
linear:
  x: -0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.45693452358245845
---
linear:
  x: -0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.45693452358245845

```

fig-shimooka, 2022-03-19:
There is a difference in the MasterName in pdo_mapping.csv between the arm and x86-64 versions.
It is named swd_right(or right)_controlword in the arm version and smc_test_controlword in the x86-64 version.
The difference also affects the library.
For example, /opt/ezw/usr/include/ezw-canopen/app_od.h is affected.
I want to configure SWD in an x86-64 environment.
"apt get install" did not install swd-starter-kit.
So I quote some /opt/ezw files from IMX6 PC(arm).
I assume this is the cause.
Can you distribute swd-starter-kit for x86-64 as well?

## pdo_mapping.csv ( x86-64)
```
NodeID,Address,Name,MasterAddress,MasterName

# Drive 0x10
16,0x60400010,controlword,0x60030010,smc_drive_controlword
16,0x60420010,vl_target_velocity,0x60040010,smc_drive_vl_target_velocity
16,0x20000020,ezw_smc_status_word,0x60000020,smc_drive_statusword
16,0x20308008,supported_fsa_actions,0x21000008,smc_drive_supported_fsa_actions
16,0x606C0020,velocity_actual_value,0x60010020,smc_drive_velocity_actual_value
16,0x60640020,position_value,0x60020020,smc_drive_position_value
16,0x60410010,statusword,0x600d0010,smc_drive_cia402_statusword

# Test 0x05
5,0x60400010,controlword,0x61030010,smc_test_controlword
5,0x60420010,vl_target_velocity,0x61040010,smc_test_vl_target_velocity
5,0x20000020,ezw_smc_status_word,0x61000020,smc_test_statusword
5,0x20308008,supported_fsa_actions,0x21010008,smc_test_supported_fsa_actions
5,0x606C0020,velocity_actual_value,0x61010020,smc_test_velocity_actual_value
5,0x60640020,position_value,0x61020020,smc_test_position_value
5,0x60410010,statusword,0x61050010,smc_test_cia402_statusword

# Test 0x04
4,0x60400010,controlword,0x61030010,smc_test_controlword
4,0x60420010,vl_target_velocity,0x61040010,smc_test_vl_target_velocity
4,0x20000020,ezw_smc_status_word,0x61000020,smc_test_statusword
4,0x20308008,supported_fsa_actions,0x21010008,smc_test_supported_fsa_actions
4,0x606C0020,velocity_actual_value,0x61010020,smc_test_velocity_actual_value
4,0x60640020,position_value,0x61020020,smc_test_position_value
4,0x60410010,statusword,0x61050010,smc_test_cia402_statusword
```
## pdo_mapping.csv ( armh)
```
NodeID,Address,Name,MasterAddress,MasterName

# SWD Left 0x4

# TPDO
4,0x60400010,controlword,0x61030010,swd_left_controlword
4,0x60420010,vl_target_velocity,0x61040010,swd_left_vl_target_velocity
# RPDO
4,0x606C0020,velocity_actual_value,0x61010020,swd_left_velocity_actual_value
4,0x60640020,position_value,0x61020020,swd_left_position_value
4,0x60410010,statusword,0x61050010,swd_left_statusword

# SWD Right 0x05

# TPDO
5,0x60400010,controlword,0x61090010,swd_right_controlword
5,0x60420010,vl_target_velocity,0x610a0010,swd_right_vl_target_velocity
# RPDO
5,0x606C0020,velocity_actual_value,0x61070020,swd_right_velocity_actual_value
5,0x60640020,position_value,0x61080020,swd_right_position_value
5,0x60410010,statusword,0x610b0010,swd_right_statusword
```

## app_od.h(armh)
```
/**
 * Copyright (C) 2021, EZ-Wheel S.A.S.
[... 41 lines cut ...]

#include "co_datatype.h"
#include "ezw_types.h"

#define APP_OD_OCTET_STRING_MAX_SIZE 32

/* Insert application variables that are used in the OD here */

extern uint8_t smc_drive_supported_fsa_actions; /**< 0x2100:0x00 (smc_drive_supported_fsa_actions) */
extern uint8_t smc_test_supported_fsa_actions; /**< 0x2101:0x00 (smc_test_supported_fsa_actions) */
extern uint32_t smc_drive_statusword; /**< 0x6000:0x00 (smc_drive_statusword) */
extern int32_t smc_drive_velocity_actual_value; /**< 0x6001:0x00 (smc_drive_velocity_actual_value) */
extern int32_t smc_drive_position_value; /**< 0x6002:0x00 (smc_drive_position_value) */
extern uint16_t smc_drive_controlword; /**< 0x6003:0x00 (smc_drive_controlword) */
extern int16_t smc_drive_vl_target_velocity; /**< 0x6004:0x00 (smc_drive_vl_target_velocity) */
extern uint16_t smc_drive_cia402_statusword; /**< 0x600d:0x00 (smc_drive_cia402_statusword) */
extern uint32_t smc_test_statusword; /**< 0x6100:0x00 (smc_test_statusword) */
extern int32_t smc_test_velocity_actual_value; /**< 0x6101:0x00 (smc_test_velocity_actual_value) */
extern int32_t smc_test_position_value; /**< 0x6102:0x00 (smc_test_position_value) */
extern uint16_t smc_test_controlword; /**< 0x6103:0x00 (smc_test_controlword) */
extern int16_t smc_test_vl_target_velocity; /**< 0x6104:0x00 (smc_test_vl_target_velocity) */
extern uint16_t smc_test_cia402_statusword; /**< 0x6105:0x00 (smc_test_cia402_statusword) */

#endif /* APP_OD_H */
```

GMezWheel, 2022-03-19:
I think there is a problem with the swd-services package installed.

Indeed, all the previous bad config files are contained in swd-services package (**/opt/ezw/usr** directory) and not in swd-starter-kit package (**/opt/ezw** directory).
Thus, if you install **swd-services_0.2.0_amd64.deb**, the files in /opt/ezw/usr/etc will be overwritten by the correct ones : 
```
# SWD Left 0x4
# SWD Right 0x5
```

fig-shimooka, 2022-03-19:
The ezw folder was deleted and swd-service was installed again.
The result was no different.

```
NodeID,Address,Name,MasterAddress,MasterName

# Drive 0x10
16,0x60400010,controlword,0x60030010,smc_drive_controlword
16,0x60420010,vl_target_velocity,0x60040010,smc_drive_vl_target_velocity
16,0x20000020,ezw_smc_status_word,0x60000020,smc_drive_statusword
16,0x20308008,supported_fsa_actions,0x21000008,smc_drive_supported_fsa_actions
16,0x606C0020,velocity_actual_value,0x60010020,smc_drive_velocity_actual_value
16,0x60640020,position_value,0x60020020,smc_drive_position_value
16,0x60410010,statusword,0x600d0010,smc_drive_cia402_statusword

# Test 0x05
5,0x60400010,controlword,0x61030010,smc_test_controlword
5,0x60420010,vl_target_velocity,0x61040010,smc_test_vl_target_velocity
5,0x20000020,ezw_smc_status_word,0x61000020,smc_test_statusword
5,0x20308008,supported_fsa_actions,0x21010008,smc_test_supported_fsa_actions
5,0x606C0020,velocity_actual_value,0x61010020,smc_test_velocity_actual_value
5,0x60640020,position_value,0x61020020,smc_test_position_value
5,0x60410010,statusword,0x61050010,smc_test_cia402_statusword

# Test 0x04
4,0x60400010,controlword,0x61030010,smc_test_controlword
4,0x60420010,vl_target_velocity,0x61040010,smc_test_vl_target_velocity
4,0x20000020,ezw_smc_status_word,0x61000020,smc_test_statusword
4,0x20308008,supported_fsa_actions,0x21010008,smc_test_supported_fsa_actions
4,0x606C0020,velocity_actual_value,0x61010020,smc_test_velocity_actual_value
4,0x60640020,position_value,0x61020020,smc_test_position_value
4,0x60410010,statusword,0x61050010,smc_test_cia402_statusword
```

GMezWheel, 2022-03-19:
Sorry. You're right.

The files are not correct in the package.

I'll rebuild a new version of this package for amd64.

fig-shimooka, 2022-03-19:
Thank you. It helps.
If you don't mind, I would like to know the completion schedule.

GMezWheel, 2022-03-19:
In two days (coming monday).

fig-shimooka, 2022-03-20:
Sometimes the execution of swd_left(or right)_NODE_ID_commissioning.py fails.
The timing of the failure varies each time.
Are other nodes interrupting?
It occurs whether ezw-ros-bringup.service is stopped or moved.

I will wait if it will be improved in the next distribution package.

## result
```

$ python3 swd_left_4_commissioning.py 
SYSTEMCTL_dbus.id detected
create_dbus_clients(swd_left) : OK
Restore factory parameters : OK
Reset to apply parameters : OK
setNetworkParameters(node_id=4) : OK
setTPDOCommunicationParameters(PDOId.PDO_1) : Failed

Commissioning failed !
$ python3 swd_left_4_commissioning.py 
SYSTEMCTL_dbus.id detected
create_dbus_clients(swd_left) : OK
Restore factory parameters : OK
Reset to apply parameters : OK
setNetworkParameters(node_id=4) : OK
setTPDOCommunicationParameters(PDOId.PDO_1) : OK
setTPDOCommunicationParameters(PDOId.PDO_2) : OK
setTPDOCommunicationParameters(PDOId.PDO_3) : OK
setTPDOCommunicationParameters(PDOId.PDO_4) : OK
setRPDOCommunicationParameters(PDOId.PDO_1) : OK
setRPDOCommunicationParameters(PDOId.PDO_2) : OK
setRPDOCommunicationParameters(PDOId.PDO_3) : OK
setRPDOCommunicationParameters(PDOId.PDO_4) : OK
setPolarityParameters(True) : OK
getSRDOParameters(SRDO_1) : OK
setSRDOParameters(SRDO_1) : Failed

```

## swd_left_4_commissioning.py
```
from enum import Enum
import sys
import time

sys.path.append(".")

import commissioning

from smcdbusclient.communication import BlocId
from smcdbusclient.nmt import NMTCommand

from smcdbusclient.safe_motion import SafetyControlWordId, SafetyFunctionId

from smcdbusclient.srdo import SRDOId, SRDOParameters

def update_SRDO_parameters():
    # SRDO_9
    # communication parameters (RX)
    srdoParam = SRDOParameters()
    srdoParam.can_id1 = 0x109
    srdoParam.can_id2 = 0x10A
    srdoParam.valid = 1
    srdoParam.sct = 255
    srdoParam.srvt = 100
    error = commissioning.srdo_client.setSRDOParameters(SRDOId.SRDO_9, srdoParam)
    commissioning.check("setSRDOParameters(SRDOId.SRDO_9)", error)

    # mapping parameters
    scw = SafetyControlWordId.CAN_2
[... 304 lines cut ...]
    # Update STO parameters
    #
    commissioning.update_STO_parameters(restart_acknowledge_behavior)

    #
    # Update SLS parameters
    #
    commissioning.update_SLS_parameters(sls_vl_limit, sls_vl_time_monitoring)

    # Save modified parameters
    error = commissioning.communication_client.storeParameters(BlocId.ALL)
    commissioning.check("storeParameters", 1)  # error)

    # Reset to apply parameters
    error = commissioning.nmt_client.setNMTState(NMTCommand.RESET_NODE)
    commissioning.check("setNMTState", error)

    # Exit with success
    print("\nCommissioning succeeded !")

if __name__ == "__main__":

    main(sys.argv[1:])

```

GMezWheel, 2022-03-21:
Hi @fig-shimooka,

During commissioning, swd firmware shall stay in NMT PREOP; but, swd_ros_controller force NMT OPER mode when running.
=> swd_ros_controller shall be stopped when commissioning
Is it embarrassing for you ?

GMezWheel, 2022-03-21:
swd-services v0.2.1 with correct CANopen dictionary is available.

fig-shimooka, 2022-03-21:
What I want to do is change velocity_limit_u32 with swd_ros_controller running.
If this is not possible, I will consider how to shut down swd_ros_controller once and then change it.

fig-shimooka, 2022-03-21:
Thanks for the "swd-service" update!

It is now being tested for operation.
The error has been occurring since I ran Commissioning yesterday.
Is there any way to initialise SWD Core?

```

$ python3 swd_left_4_commissioning.py
SYSTEMCTL_dbus.id detected
create_dbus_clients(swd_left) : OK
Traceback (most recent call last):
  File "swd_left_4_commissioning.py", line 178, in <module>.
    main(sys.argv[1:])
  File "swd_left_4_commissioning.py", line 108, in main
    ) = commissioning.srdo_client.getSRDOParameters(srdo)
  File "/opt/ezw/usr/lib/smcdbusclient/srdo.py", line 128, in getSRDOParameters
    a = (SRDODirection.fromDBus(r[0]), SRDOParameters.fromDBus(r[1]), int(r[2]), int(r[3]))
  File "/opt/ezw/usr/lib/smcdbusclient/srdo.py", line 44, in fromDBus
    return SRDODirection(int(dbus_obj))
  File "/usr/lib/python3.8/enum.py", line 339, in __call__.
    return cls.__new__(cls, value)
  File "/usr/lib/python3.8/enum.py", line 663, in __new__.
    raise ve_exc
ValueError: 0 is not a valid SRDODirection

```

fig-shimooka, 2022-03-21:
Hi @GMezWheel 
Thanks for the "swd-service" update!

It is now being tested for operation.
The error has been occurring since I ran Commissioning yesterday.
Is there any way to initialise SWD Core?

```

$ python3 swd_left_4_commissioning.py
SYSTEMCTL_dbus.id detected
create_dbus_clients(swd_left) : OK
Traceback (most recent call last):
  File "swd_left_4_commissioning.py", line 178, in <module>.
    main(sys.argv[1:])
  File "swd_left_4_commissioning.py", line 108, in main
    ) = commissioning.srdo_client.getSRDOParameters(srdo)
  File "/opt/ezw/usr/lib/smcdbusclient/srdo.py", line 128, in getSRDOParameters
    a = (SRDODirection.fromDBus(r[0]), SRDOParameters.fromDBus(r[1]), int(r[2]), int(r[3]))
  File "/opt/ezw/usr/lib/smcdbusclient/srdo.py", line 44, in fromDBus
    return SRDODirection(int(dbus_obj))
  File "/usr/lib/python3.8/enum.py", line 339, in __call__.
    return cls.__new__(cls, value)
  File "/usr/lib/python3.8/enum.py", line 663, in __new__.
    raise ve_exc
ValueError: 0 is not a valid SRDODirection

```

fig-shimooka, 2022-03-21:
The updateParameter also fails.
```

$ python3 swd_left_4_commissioning.py
SYSTEMCTL_dbus.id detected
create_dbus_clients(swd_left) : OK
Restore factory parameters : OK
Reset to apply parameters : OK
setNetworkParameters(node_id=4) : OK
setTPDOCommunicationParameters(PDOId.PDO_1) : OK
setTPDOCommunicationParameters(PDOId.PDO_2) : OK
setTPDOCommunicationParameters(PDOId.PDO_3) : OK
[commissioning.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/8315373/commissioning.zip)

setTPDOCommunicationParameters(PDOId.PDO_4) : OK
setRPDOCommunicationParameters(PDOId.PDO_1) : OK
setRPDOCommunicationParameters(PDOId.PDO_2) : OK
setRPDOCommunicationParameters(PDOId.PDO_3) : Failed
```

GMezWheel, 2022-03-21:
In CANopen standard, direction set to 0 (in the CANopen dictionary) means SRDO is disabled (invalid).

But, in SRDOParameters value, SRDO direction is either 1 (TX) or 2 (RX).
=> direction with 0 will raised an error : "0 is not a valid SRDODirection"
=> the validity is in flag params.valid 

The commissioning scripts will disable all SRDO
=> go to NMT PREOP
=> commissioning.disable_SRDO_parameters()
and configure SRDO 9 and 16.

fig-shimooka, 2022-03-21:
Where can I set up Directions?
I can't find it in commissioning.py.

fig-shimooka, 2022-03-21:
It is also no longer possible to set up the system.
This too has no known cause.
```

$ python3 swd_left_4_commissioning.py 
SYSTEMCTL_dbus.id detected
create_dbus_clients(swd_left) : OK
Restore factory parameters : OK
Reset to apply parameters : OK
setNetworkParameters(node_id=4) : Failed

$ python3 swd_left_4_commissioning.py 
SYSTEMCTL_dbus.id detected
create_dbus_clients(swd_left) : OK
Restore factory parameters : OK
Reset to apply parameters : OK
setNetworkParameters(node_id=4) : OK
setTPDOCommunicationParameters(PDOId.PDO_1) : OK
setTPDOCommunicationParameters(PDOId.PDO_2) : Failed

Commissioning failed !
```

GMezWheel, 2022-03-21:
swd_ros_controller has to be stopped for using commissioning scripts.

GMezWheel, 2022-03-21:
For commissioning SRDO, you can look at disable_SRDO_parameters in commissioning.py:
=> get SRDOParameters
=> set params.valid to True or False
=> set SRDOParameters

_NOTA : SRDO2/10/16 are TX and RX for the others_

fig-shimooka, 2022-03-21:
swd_ros_controller is deactivated
Also, disable_SRDO_parameters has params.valid set to True.
This still causes an error.

GMezWheel, 2022-03-21:
Maybe the both motors are using the same nodeId (16 by default).
`=> candump can0 | grep 7`
If this is the case, during commissioning one motor, the other one shall be off.

GMezWheel, 2022-03-21:
Indeed, the both motors have the same nodeId.
=> 0x10 => HB=710

The solution is : 
1) power off left motor.
2) modifying /opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini with nodeId = 16
3) commissioning right motor
4) modifying /opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini with nodeId = 5
5) power on left motor.
6) modifying /opt/ezw/usr/etc/ezw-smc-core/swd_left_config.ini with nodeId = 16
7) commissioning left motor
8) modifying /opt/ezw/usr/etc/ezw-smc-core/swd_left_config.ini with nodeId = 4

fig-shimooka, 2022-03-21:
The script was run with the left motor switched off.
The result was successful.

$ python3 swd_left_4_commissioning.py
SYSTEMCTL_dbus.id detected
create_dbus_clients(swd_left) : OK
Restore factory parameters : OK
Reset to apply parameters : OK
setNetworkParameters(node_id=4) : OK
setTPDOCommunicationParameters(PDOId.PDO_1) : OK
setTPDOCommunicationParameters(PDOId.PDO_2) : OK
setTPDOCommunicationParameters(PDOId.PDO_3) : OK
setTPDOCommunicationParameters(PDOId.PDO_4) : OK
setRPDOCommunicationParameters(PDOId.PDO_1) : OK
setRPDOCommunicationParameters(PDOId.PDO_2) : OK
setRPDOCommunicationParameters(PDOId.PDO_3) : OK
setRPDOCommunicationParameters(PDOId.PDO_4) : OK
setPolarityParameters(True) : OK
getSRDOParameters(SRDO_1) : OK
setSRDOParameters(SRDO_1) : OK
getSRDOParameters(SRDO_2) : OK
setSRDOParameters(SRDO_2) : OK
getSRDOParameters(SRDO_9) : OK
setSRDOParameters(SRDO_9) : OK
getSRDOParameters(SRDO_10) : OK
setSRDOParameters(SRDO_10) : OK
getSRDOParameters(SRDO_11) : OK
setSRDOParameters(SRDO_11) : OK
getSRDOParameters(SRDO_12) : OK
setSRDOParameters(SRDO_12) : OK
getSRDOParameters(SRDO_13) : OK
setSRDOParameters(SRDO_13) : OK
getSRDOParameters(SRDO_14) : OK
setSRDOParameters(SRDO_14) : OK
getSRDOParameters(SRDO_15) : OK
setSRDOParameters(SRDO_15) : OK
getSRDOParameters(SRDO_16) : OK
setSRDOParameters(SRDO_16) : OK
setSRDOConfigurationValidity() : OK
setSRDOParameters(SRDOId.SRDO_9) : OK
setSafetyControlWordMapping() : OK
setSRDOParameters(SRDOId.SRDO_16) : OK
setSafetyControlWordMapping() : OK
setSRDOConfigurationValidity() : OK
getVelocityModeParameters() : OK
setVelocityParameters() : OK
getSTOParameters(STOId.STO_1) : OK
setSTOParameters(STOId.STO_1) : OK
getLSParameters(SLSId.SLS_1) : OK
setSLSParameters(SLSId.SLS_1) : OK
storeParameters : OK
setNMTState : OK

Commissioning succeeded !

GMezWheel, 2022-03-21:
At restarting ezw-smc-service, it will load config.ini and use nodeId to discuss with motor.
=> in our case, nodeId is 16 (no more 4 or 5)

fig-shimooka, 2022-03-21:
For some reason, only the right motor fails to commission.
I rewrote the node id in swd_right_config.ini to 16, restarted ezw-swd-right.service and finally ran swd_right_5_commissioning.py.
```

$ python3 swd_right_5_commissioning.py
SYSTEMCTL_dbus.id detected
create_dbus_clients(swd_right) : OK
Restore factory parameters : OK
Reset to apply parameters : OK
setNetworkParameters(node_id=5) : Failed
```
```
$ ./remote.py swd_right
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x0 .......................................................................... [KO]
Vendor-Id            : 0x0 .......................................................................... [KO]
Product code         : 0x0 .......................................................................... [KO]
Serial number        : 0 ............................................................................ [KO]
Revision number      : 0 ............................................................................ [KO]
SWVersion            :  ............................................................................. [KO]
HWVersion            :  ............................................................................. [KO]
Calibrated           : False ........................................................................ [KO]
SRDO validity        : False ........................................................................ [KO]
NMT State            : UNKNOWN  ..................................................................... [OK]
PDS State            : NOT_READY_TO_SWITCH_ON  ...................................................... [KO]
NbError              : 0 ............................................................................ [KO]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [KO]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [KO]
SCW SAFEIN_1         : False, False, False, False, False, False, False, False  ...................... [KO]
Peripheral           : False, False, False, False, False, False  .................................... [KO]
VelocityModeSW       : internal_limit_active: False ................................................. [KO]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [KO]
Polarity             : velocity_polarity: False, position_polarity: False ........................... [KO]
TargetVelocity       : 0 ............................................................................ [KO]
VelocityDemand       : 0 ............................................................................ [KO]
VelocityActualValue  : 0 ............................................................................ [KO]
PositionValue        : 0 ............................................................................ [KO]
OdometryValue        : 0 ............................................................................ [K
```

GMezWheel, 2022-03-21:
Before recommissioning, launch remote.py to verify that all is ok with nodeId 16.

fig-shimooka, 2022-03-21:
result

 ```
$ ./remote.py swd_right
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x0 .......................................................................... [KO]
Vendor-Id            : 0x0 .......................................................................... [KO]
Product code         : 0x0 .......................................................................... [KO]
Serial number        : 0 ............................................................................ [KO]
Revision number      : 0 ............................................................................ [KO]
SWVersion            :  ............................................................................. [KO]
HWVersion            :  ............................................................................. [KO]
Calibrated           : False ........................................................................ [KO]
SRDO validity        : False ........................................................................ [KO]
NMT State            : UNKNOWN  ..................................................................... [OK]
PDS State            : NOT_READY_TO_SWITCH_ON  ...................................................... [KO]
NbError              : 0 ............................................................................ [KO]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [KO]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [KO]
SCW SAFEIN_1         : False, False, False, False, False, False, False, False  ...................... [KO]
Peripheral           : False, False, False, False, False, False  .................................... [KO]
VelocityModeSW       : internal_limit_active: False ................................................. [KO]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [KO]
Polarity             : velocity_polarity: False, position_polarity: False ........................... [KO]
TargetVelocity       : 0 ............................................................................ [KO]
VelocityDemand       : 0 ............................................................................ [KO]
VelocityActualValue  : 0 ............................................................................ [KO]
PositionValue        : 0 ............................................................................ [KO]
OdometryValue        : 0 ............................................................................ [KO]
```

fig-shimooka, 2022-03-21:
I set the node id in the ini file to 5 and restarted.
I then ran remote.py, but the result did not change.
The same was also true when I rebooted with node id set to 16.

fig-shimooka, 2022-03-21:
The left motor remains switched off.

GMezWheel, 2022-03-21:
Maybe the nodeId is 4 !

fig-shimooka, 2022-03-21:
```
$ ./remote.py swd_right
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 724434945 .................................................................... [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 0.9.0 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : PREOP  ....................................................................... [OK]
PDS State            : FAULT  ....................................................................... [OK]
NbError              : 1 ............................................................................ [OK]
LastError            : 4210 : Excess_temperature_device ............................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, False, False, False, False, True, True  .......................... [OK]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: True .................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: False, position_polarity: False ........................... [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 0 ............................................................................ [OK]
OdometryValue        : 0 ............................................................................ [OK]

$ ./remote.py swd_left
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 721485826 .................................................................... [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 0.9.0 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : PREOP  ....................................................................... [OK]
PDS State            : FAULT  ....................................................................... [OK]
NbError              : 0 ............................................................................ [OK]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
[... cut ...]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 5 ............................................................................ [OK]
OdometryValue        : 4 ............................................................................ [OK]
```

fig-shimooka, 2022-03-21:
The result of remote.py was as follows.
NMT State : PREOP  
PDS State : FAULT  
Running ezw-ros-bringup.service has not changed the state to NMT OPER

I added iptables to my system yesterday.
Is there a need to open ports?

GMezWheel, 2022-03-21:
Try to force NMT OPER for all CANopen node : 
`cansend can0 000#0100`

[38 more replies]

closed 2022-06-20 by ez-Support