IDEC-ezWheel/swd_ros_controllers #9: I want to update the SWD Core software version.

fig-shimooka, 2022-03-24:
Hello.
The SW Version of the SWD Core at hand seems to be out of date.
How can this be updated?
I want to update the SW Version to 0.9.0.

GMezWheel, 2022-03-24:
Hi @fig-shimooka,

You need to have a usb cable to flash a new firmware :
* M8 circular connector 5 core (eg ref : 1404461)
* USB Type A male connector

[image]
[image]
[image]

I will explain you the procedure to flash a new firmware when you will have the usb cable.

GMezWheel, 2022-03-24:
For information, a new version of SWD Core firmware is released : 1.0.1
This version needs new swd-services_0.1.2 package (with backward compatibility).
=> I suggest you to update this package

Then, I propose you to update your firmware tomorrow morning (9am french time). 
=> I will send you a link to download it.

ez-Support, 2022-03-25:
The new firmware is available here:
http://51.83.77.229:8000/s/tTpBai5ATZaWQkQ
You should download it.

ez-Support, 2022-03-25:
Then copy it to your SK (e.g. scp ~/Downloads/GM_05_SWD_v1.0.1_gm.bin swd_sk@10.10.0.1:/opt/ezw/firmwares/)
, thus your usb cable should be plugged, on one of the swd motor.
Have a look on the usb tty being used.
Then, you have to stop your ros-controller (e.g. with ezw-ros controller: "systemctl stop ezw-ros-controller.service")
Finally, flash the new binary using the bootloader, e.g.

sudo /opt/ezw/usr/bin/ezw-bootloader-tms /dev/ttyUSBX 1000000 /opt/ezw/firmwares/app_bin/gm_04_gm.bin --active_bootloader --active_application

fig-shimooka, 2022-03-28:
Sorry. The firmware download link has expired.
Please upload the firmware again.

GMezWheel, 2022-03-28:
http://51.83.77.229:8000/s/arw6Go8be8JRWge

fig-shimooka, 2022-03-28:
Thank you. I was able to update the firmware successfully.

closed 2022-03-28 by GMezWheel