IDEC-ezWheel/swd_ros_controllers #10: How to restore the SWD state to its start-up state

fig-shimooka, 2022-03-25:
How to restore the SWD state to its start-up state

Is there a way to restore the SWD state to the state immediately after start-up?
If the PC is rebooted with power supplied to the motor after a normal start-up, the PDS State sometimes becomes False.
The motor is rebooted ignoring the state of the motor. r

ez-Support, 2022-03-25:
The pc is the canopen master, so the two swd-motors turn to NMT PRE-OPERATIONAL, as the master was turned off.
When the ezw-ros-controller is started, it sends the commands to the motor, to go to NMT State OPERATIONNAL

fig-shimooka, 2022-03-28:
Sorry. It seems that ezw-ros-bringup failed to start. (Due to my environment).
After correcting my environment, it now works as expected.

closed 2022-03-28 by fig-shimooka