IDEC-ezWheel/swd_ros_controllers #11: Updating the firmware causes the ros-controller to die.

fig-shimooka, 2022-03-28:
I followed this procedure to update the SWD Core firmware.
https://github.com/ezWheelSAS/swd_ros_controllers/issues/9
After the update, I rebooted, but swd-ros-controller dies.
The swd-service has been removed and installed once.

```

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /swd_diff_drive_controller/base_frame: base_footprint
 * /swd_diff_drive_controller/baseline_m: 0.49
 * /swd_diff_drive_controller/command_timeout_ms: 500
 * /swd_diff_drive_controller/control_mode: Twist
 * /swd_diff_drive_controller/have_backward_sls: False
 * /swd_diff_drive_controller/left_swd_config_file: /opt/ezw/usr/etc/...
 * /swd_diff_drive_controller/odom_frame: odom
 * /swd_diff_drive_controller/positive_polarity_wheel: Right
 * /swd_diff_drive_controller/pub_freq_hz: 50
 * /swd_diff_drive_controller/publish_odom: True
 * /swd_diff_drive_controller/publish_safety_functions: True
 * /swd_diff_drive_controller/publish_tf: True
 * /swd_diff_drive_controller/right_swd_config_file: /opt/ezw/usr/etc/...
 * /swd_diff_drive_controller/wheel_max_speed_rpm: 95
 * /swd_diff_drive_controller/wheel_safety_limited_speed_rpm: 40
 * /urg_node/angle_max: 1.5707963
 * /urg_node/angle_min: -1.5707963
 * /urg_node/frame_id: laser_1
 * /urg_node/ip_address: 192.168.0.110
 * /xbox_joystick/autorepeat_rate: 50

NODES
  /
    swd_diff_drive_controller (swd_ros_controllers/swd_diff_drive_controller)
    urg_node (urg_node/urg_node)
    xbox_joystick (joy/joy_node)

auto-starting new master
process[master]: started with pid [13684]
ROS_MASTER_URI=http://192.168.0.1:11311

setting /run_id to db64a04c-ae7c-11ec-b905-b7e741d0abca
process[rosout-1]: started with pid [13710]
started core service [/rosout]
process[swd_diff_drive_controller-2]: started with pid [13717]
process[urg_node-3]: started with pid [13718]
process[xbox_joystick-4]: started with pid [13719]
[ INFO] [1648461183.504142290]: Ready !
[ INFO] [1648461183.522592913]: ###########################
[ INFO] [1648461183.525610888]: Initializing swd_diff_drive_controller, node name : /swd_diff_drive_controller
[ INFO] [1648461183.558645349]: Set Complete Safety Info
[ INFO] [1648461183.566188845]: Motors config files, right : /opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini, left : /opt/ezw/usr/etc/ezw-smc-core/swd_left_config.ini
[ WARN] [1648461183.584640697]: Couldn't set gain on joystick force feedback: Bad file descriptor
[swd_diff_drive_controller-2] process has died [pid 13717, exit code -11, cmd /home/fig/ros1_swd/devel/lib/swd_ros_controllers/swd_diff_drive_controller /swd_diff_drive_controller/cmd_vel:=/cmd_vel/master /swd_diff_drive_controller/odom:=/odom __name:=swd_diff_drive_controller __log:=/home/fig/.ros/log/db64a04c-ae7c-11ec-b905-b7e741d0abca/swd_diff_drive_controller-2.log].
log file: /home/fig/.ros/log/db64a04c-ae7c-11ec-b905-b7e741d0abca/swd_diff_drive_controller-2*.log
``````

``````
$ sudo apt-cache policy swd-services 
swd-services:
  Installed: 0.1.2
  Candidate: 0.1.2
  Version table:
 *** 0.1.2 500
        500 http://packages.ez-wheel.com:8081/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
```
[... 49 lines cut ...]
PDS State            : FAULT  ....................................................................... [OK]
NbError              : 1 ............................................................................ [OK]
LastError            : 800E : FAULT_ALIM_EXT ........................................................ [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, True, True, True, True, True, True  .............................. [OK]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 3 ............................................................................ [OK]
OdometryValue        : 2 ............................................................................ [OK]
```
[ezw_0_1_2.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/8361635/ezw_0_1_2.zip)

fig-shimooka, 2022-03-28:
Thank you.
I rebuilt it and it started.
However, SWD Core does not work after sending cmd_vel.
```

Mar 28 19:32:35 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463555.076087184]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:35 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463555.271587467]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:35 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463555.470312658]: NMT state machine is not OK.
Mar 28 19:32:35 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463555.470402112]: PDS state machine is not OK.
Mar 28 19:32:35 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463555.470459065]: NMT state machine is not OK, no valid SafetyFunctions message to publish
^AMar 28 19:32:35 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463555.677815247]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:35 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463555.874641021]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:36 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463556.078804272]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:36 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463556.265583637]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:36 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463556.475845960]: NMT state machine is not OK.
Mar 28 19:32:36 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463556.475940689]: PDS state machine is not OK.
Mar 28 19:32:36 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463556.476040468]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:36 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463556.688097138]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:36 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463556.870628116]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:37 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463557.067033984]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:37 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463557.270165374]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:37 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463557.475583928]: NMT state machine is not OK.
Mar 28 19:32:37 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463557.476072779]: PDS state machine is not OK.
Mar 28 19:32:37 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463557.476276027]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:37 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463557.683903524]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:37 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463557.879835952]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:38 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463558.090448774]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:38 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463558.292792272]: NMT state machine is not OK, no valid SafetyFunctions message to publish
Mar 28 19:32:38 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463558.477537551]: NMT state machine is not OK.
Mar 28 19:32:38 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463558.477959819]: PDS state machine is not OK.
Mar 28 19:32:38 x86 sce-ros-bringup.sh[3274]: [ WARN] [1648463558.478306471]: NMT state machine is not OK, no valid SafetyFunctions message to publish

```

fig-shimooka, 2022-03-28:
Should I do the commissioning again?

ez-Support, 2022-03-28:
Yes, you have to do the commissioning. As explained here: https://github.com/ezWheelSAS/swd_ros_controllers/issues/9#issuecomment-1078771109.

fig-shimooka, 2022-03-28:
Sorry. I missed it.
It works now.

fig-shimooka, 2022-03-28:
Error again.
I rebooted and the result is the same.
```

Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [ INFO] [1648467575.975247478]: Initializing swd_diff_drive_controller, node name : /swd_diff_drive_controller
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [ INFO] [1648467576.279676295]: Set Complete Safety Info
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [ INFO] [1648467576.329927936]: Motors config files, right : /opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini, left : /opt/ezw/usr/etc/ezw-smc-core/swd_left_confi>
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] contextId = 12
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] nodeId = 5
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] canDevice = can0
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] HWConfigurationEntry = SWD_RIGHT
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] HWConfigurationFile = /opt/ezw/data/configuration.json
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] SWCompatibilityFile = /opt/ezw/firmwares/compatibility.json
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] CANOpenEDSFile = /opt/ezw/usr/etc/ezw-canopen-dico/swd_core.eds
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] targetVelocityTimeoutMs = 300
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] maxMaintenanceSpeedValue = 400
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] contextId = 12
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] nodeId = 5
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] canDevice = can0
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] HWConfigurationEntry = SWD_RIGHT
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] HWConfigurationFile = /opt/ezw/data/configuration.json
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] SWCompatibilityFile = /opt/ezw/firmwares/compatibility.json
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] CANOpenEDSFile = /opt/ezw/usr/etc/ezw-canopen-dico/swd_core.eds
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] targetVelocityTimeoutMs = 300
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: [default] maxMaintenanceSpeedValue = 400
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:36;;SMCDS;INFO;getHWConfigurationMember;90;HardwareHelper;Search for entry SWD_RIGHT / nbStepRevolutionElectype Integer into /opt/ezw/data/configura>
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:36;;SMCDS;INFO;getHWConfigurationMember;111;HardwareHelper;nbStepRevolutionElec : 6;
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:36;;SMCDS;INFO;getHWConfigurationMember;90;HardwareHelper;Search for entry SWD_RIGHT / nbPolePairtype Integer into /opt/ezw/data/configuration.json;
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:36;;SMCDS;INFO;getHWConfigurationMember;111;HardwareHelper;nbPolePair : 5;
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:36;;SMCDS;INFO;getHWConfigurationMember;145;HardwareHelper;Search for entry SWD_RIGHT / reduction type Double into /opt/ezw/data/configuration.json;
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:36;;SMCDS;INFO;getHWConfigurationMember;166;HardwareHelper;reduction : 14.000000;
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:36;;SMCDS;INFO;getHWConfigurationMember;145;HardwareHelper;Search for entry SWD_RIGHT / diameter type Double into /opt/ezw/data/configuration.json;
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:36;;SMCDS;INFO;getHWConfigurationMember;166;HardwareHelper;diameter : 125.000000;
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:36;;SMCDS;INFO;getHWConfigurationMember;200;HardwareHelper;Search for entry SWD_RIGHT / device type String into /opt/ezw/data/configuration.json;
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:36;;SMCDS;INFO;getHWConfigurationMember;221;HardwareHelper;device : /dev/ttyUSBY;
Mar 28 20:39:36 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:36;;APP;INFO;init;76;DBusClientDelegate;Checking DBUS proxy availability : commonapi.ezw.canopenservice;
Mar 28 20:39:38 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:38;;APP;INFO;init;89;DBusClientDelegate;DBUS Proxy available : commonapi.ezw.canopenservice;
Mar 28 20:39:39 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:39;;SMCDS;INFO;init;83;Init Success;
Mar 28 20:39:39 x86 sce-ros-bringup.sh[2611]: [default] contextId = 12
Mar 28 20:39:39 x86 sce-ros-bringup.sh[2611]: [default] nodeId = 4
Mar 28 20:39:39 x86 sce-ros-bringup.sh[2611]: [default] canDevice = can0
Mar 28 20:39:39 x86 sce-ros-bringup.sh[2611]: [default] HWConfigurationEntry = SWD_LEFT
Mar 28 20:39:39 x86 sce-ros-bringup.sh[2611]: [default] HWConfigurationFile = /opt/ezw/data/configuration.json
Mar 28 20:39:39 x86 sce-ros-bringup.sh[2611]: [default] SWCompatibilityFile = /opt/ezw/firmwares/compatibility.json
Mar 28 20:39:39 x86 sce-ros-bringup.sh[2611]: [default] CANOpenEDSFile = /opt/ezw/usr/etc/ezw-canopen-dico/swd_core.eds
Mar 28 20:39:39 x86 sce-ros-bringup.sh[2611]: [default] targetVelocityTimeoutMs = 300
Mar 28 20:39:39 x86 sce-ros-bringup.sh[2611]: [default] maxMaintenanceSpeedValue = 400
Mar 28 20:39:39 x86 sce-ros-bringup.sh[2611]: [default] contextId = 12
Mar 28 20:39:39 x86 sce-ros-bringup.sh[2611]: [default] nodeId = 4
Mar 28 20:39:39 x86 sce-ros-bringup.sh[2611]: [default] canDevice = can0
Mar 28 20:39:39 x86 sce-ros-bringup.sh[2611]: [default] HWConfigurationEntry = SWD_LEFT
[... cut ...]
Mar 28 20:39:43 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:41;;SMCDS;INFO;enterInOperationEnabledState;535;Controller;Check FAULT;
Mar 28 20:39:43 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:41;;SMCDS;INFO;enterInOperationEnabledState;544;Controller;FAULT detected;
Mar 28 20:39:43 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:42;;SMCDS;INFO;enterInOperationEnabledState;562;Controller;Check SWITCH_ON_DISABLED;
Mar 28 20:39:43 x86 sce-ros-bringup.sh[2611]: 2022-3-28 20:39:42;;SMCDS;INFO;enterInOperationEnabledState;582;Controller;Check READY_TO_SWITCH_ON;

```

fig-shimooka, 2022-03-28:
Apparently only the condition of the right motor is inappropriate.

```
$ ./remote.py swd_right
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 721420300 .................................................................... [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : FAULT  ....................................................................... [OK]
NbError              : 2 ............................................................................ [OK]
LastError            : 800E : FAULT_ALIM_EXT ........................................................ [OK]
SystemError          : EZW_PROTECT_BRAKE_CC_PROTECT EZW_PROTECT_BRAKE_CC_PROTECT  ................... [OK]
SSW CAN1             : True, False, False, False, False, False, False, False  ....................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : False, False, False, False, False, False, True, True  ........................ [OK]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: False, position_polarity: False ........................... [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -1 ........................................................................... [OK]
OdometryValue        : 0 ............................................................................ [OK]

$ ./remote.py swd_left
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 722927685 .................................................................... [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : OPERATION_ENABLED  ........................................................... [OK]
NbError              : 3 ............................................................................ [OK]
LastError            : 8201 : SRDO_SCT_timeout ...................................................... [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
[... cut ...]
             └─1203 /opt/ezw/usr/bin/ezw-smc-service /opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini

$ cat /opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini
# SMC Drive service config file
contextId = 12
nodeId = 5
canDevice = can0

HWConfigurationEntry = SWD_RIGHT
HWConfigurationFile = /opt/ezw/data/configuration.json
SWCompatibilityFile = /opt/ezw/firmwares/compatibility.json

CANOpenEDSFile = /opt/ezw/usr/etc/ezw-canopen-dico/swd_core.eds

targetVelocityTimeoutMs = 300

# max value is 0.15m/s => 400
maxMaintenanceSpeedValue = 400

```

fig-shimooka, 2022-03-28:
The difference between the left and right side of the COMMISSION.
Left source code: right motor
Right source code: left motor

[image]
[image]

ez-Support, 2022-03-28:
It seems that you have the emergency button ON

fig-shimooka, 2022-03-28:
That was true. Thank you very much.

closed 2022-03-28 by fig-shimooka

ez-Support, 2022-03-28:
You can see it on the first value of the SSW CAN1