IDEC-ezWheel/swd_ros_controllers #13: Rapid acceleration when AMR is turning at low speeds.

fig-shimooka, 2022-04-25:
I previously took your advice and turned ramp to disable.
https://github.com/ezWheelSAS/swd_ros_controllers/issues/8

After that, when I turn the AMR at low speed, it may momentarily accelerate rapidly.

What is causing this
[commissioning.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/8550957/commissioning.zip)
?

ez-Support, 2022-04-25:
I'm not sure I really understand the behaviour you are telling me. 
The set-up of acceleration ramps, enables the robot to accelerate faster or slower. 
If it is set to '2000', this is the maximum acceleration configuration. Maybe you can try a bit smaller, like 1400 and see if it solves your problem.
Otherwise, make sure you do not have two inputs coming for the speed setpoint.

hiroyuki-s1, 2022-04-26:
I have modified the parameter to 1400.
But the result did not change.
If I keep sending cmd_vel messages with angular.z = about 0.1~0.2, I get movements like in the video.

https://user-images.githubusercontent.com/53794593/165287246-e15cdd88-5007-4e6f-8bf5-4f3b2fde37e4.mp4

ez-Support, 2022-04-26:
I would suggest you to increase the proportional value of the PID.

hiroyuki-s1, 2022-04-26:
Where can I change the proportional value of the PID controller?

ez-Support, 2022-04-27:
You are not using your own controller? 
Make sure that the targetVelocity sent to the SWD motor is bigger than 50 RPM. Otherwise, it is too low, and the motor will not turn smoothly.

fig-shimooka, 2022-04-28:
I had mistakenly thought that the SWD_Diff_Controller was changing the gain.
This was addressed by increasing the minimum angular velocity when turning.

ez-Support, 2022-04-28:
No actually, the PID control has value which are configured and do not change at runtime. 
Ok, happy to know you resolved it

closed 2022-05-02 by ez-Support