IDEC-ezWheel/swd_ros_controllers #14: I want to change the speed of SLS.

fig-shimooka, 2022-04-27:
I want to change the speed of SLS.
I assume that sls_vl_limit = 680 in swd_left_4_commissioning.py is the one. What are the units?
I want to change the maximum speed of SLS to 0.25 m/s

Also, there seems to be a lot of noise in the odometry twist.

https://user-images.githubusercontent.com/72783232/165419998-4b6385e7-541e-4ebb-8d38-5f86bcab278b.mp4

ez-Support, 2022-04-27:
For the odomometry, make sure that you do not have several sources of commands to the motors. 
It looks like there is a conflict between two orders sent to the SWD motor

ez-Support, 2022-04-28:
From another thought, to solve the odometry issue, you can try to change the sample frequency (`pub_freq_hz `) of `swd_ros_controllers/swd_diff_drive_controller node`, which stands at 50 Hz by default to 20 Hz.
You will find this ROS parameter in the `swd_diff_drive_controller.launch` file.

This might be cause, as Linux is not a real time OS, and do not manage time properly.

fig-shimooka, 2022-04-28:
No multiple sources were found.
Try changing the sample frequency and see if it improves.