IDEC-ezWheel/swd_ros_controllers #16: The acquisition of the status of the emergency switch connected to the right motor is sometimes slow.

fig-shimooka, 2022-05-10:
https://github.com/ezWheelSAS/swd_ros_controllers/issues/12
The status of the emergency switch connected to the left motor can now be acquired immediately.
However, acquiring the status of the emergency switch connected to the right motor takes time.
The trend is that it seems to be slower the first time it is pressed after startup.
After the second press, the status can be acquired relatively quickly.

```
fig@x86:/opt/ezw/usr/sbin$ ./remote.py swd_right
   1 : 	[8201] SRDO_SCT_timeout
   2 : 	[8201] SRDO_SCT_timeout
   3 : 	[8201] SRDO_SCT_timeout
   4 : 	[8201] SRDO_SCT_timeout
   5 : 	[8201] SRDO_SCT_timeout
   6 : 	[8201] SRDO_SCT_timeout
   7 : 	[8201] SRDO_SCT_timeout
   8 : 	[8201] SRDO_SCT_timeout
   9 : 	[8201] SRDO_SCT_timeout
  10 : 	[8201] SRDO_SCT_timeout
  11 : 	[8201] SRDO_SCT_timeout
  12 : 	[8201] SRDO_SCT_timeout
  13 : 	[8201] SRDO_SCT_timeout
  14 : 	[8201] SRDO_SCT_timeout
  15 : 	[8201] SRDO_SCT_timeout
  16 : 	[8201] SRDO_SCT_timeout
  17 : 	[8201] SRDO_SCT_timeout
  18 : 	[8201] SRDO_SCT_timeout
  19 : 	[8201] SRDO_SCT_timeout
  20 : 	[8201] SRDO_SCT_timeout
  21 : 	[8201] SRDO_SCT_timeout
  22 : 	[8201] SRDO_SCT_timeout
  23 : 	[8201] SRDO_SCT_timeout
  24 : 	[8201] SRDO_SCT_timeout
  25 : 	[8201] SRDO_SCT_timeout
  26 : 	[8201] SRDO_SCT_timeout
  27 : 	[8201] SRDO_SCT_timeout
  28 : 	[8201] SRDO_SCT_timeout
  29 : 	[8201] SRDO_SCT_timeout
  30 : 	[8201] SRDO_SCT_timeout
  31 : 	[8201] SRDO_SCT_timeout
  32 : 	[8201] SRDO_SCT_timeout
  33 : 	[8201] SRDO_SCT_timeout
  34 : 	[8201] SRDO_SCT_timeout
  35 : 	[8201] SRDO_SCT_timeout
  36 : 	[8201] SRDO_SCT_timeout
  37 : 	[8201] SRDO_SCT_timeout
  38 : 	[8201] SRDO_SCT_timeout
  39 : 	[8201] SRDO_SCT_timeout
  40 : 	[8201] SRDO_SCT_timeout
  41 : 	[8201] SRDO_SCT_timeout
  42 : 	[8201] SRDO_SCT_timeout
  43 : 	[8201] SRDO_SCT_timeout
  44 : 	[8201] SRDO_SCT_timeout
  45 : 	[8201] SRDO_SCT_timeout
  46 : 	[8201] SRDO_SCT_timeout
  47 : 	[8201] SRDO_SCT_timeout
  48 : 	[8201] SRDO_SCT_timeout
  49 : 	[8201] SRDO_SCT_timeout
  50 : 	[8201] SRDO_SCT_timeout
  51 : 	[8201] SRDO_SCT_timeout
  52 : 	[8201] SRDO_SCT_timeout
  53 : 	[8201] SRDO_SCT_timeout
  54 : 	[8201] SRDO_SCT_timeout
  55 : 	[8201] SRDO_SCT_timeout
  56 : 	[8201] SRDO_SCT_timeout
  57 : 	[8201] SRDO_SCT_timeout
  58 : 	[8201] SRDO_SCT_timeout
  59 : 	[8201] SRDO_SCT_timeout
  60 : 	[8201] SRDO_SCT_timeout
  61 : 	[8201] SRDO_SCT_timeout
  62 : 	[8201] SRDO_SCT_timeout
  63 : 	[8201] SRDO_SCT_timeout
  64 : 	[8201] SRDO_SCT_timeout
  65 : 	[8201] SRDO_SCT_timeout
  66 : 	[8201] SRDO_SCT_timeout
  67 : 	[8201] SRDO_SCT_timeout
  68 : 	[8201] SRDO_SCT_timeout
  69 : 	[8201] SRDO_SCT_timeout
  70 : 	[8201] SRDO_SCT_timeout
  71 : 	[8201] SRDO_SCT_timeout
  72 : 	[8201] SRDO_SCT_timeout
  73 : 	[8201] SRDO_SCT_timeout
  74 : 	[8201] SRDO_SCT_timeout
  75 : 	[8201] SRDO_SCT_timeout
  76 : 	[8201] SRDO_SCT_timeout
  77 : 	[8201] SRDO_SCT_timeout
  78 : 	[8201] SRDO_SCT_timeout
  79 : 	[8201] SRDO_SCT_timeout
  80 : 	[8201] SRDO_SCT_timeout
  81 : 	[8201] SRDO_SCT_timeout
  82 : 	[8201] SRDO_SCT_timeout
  83 : 	[8201] SRDO_SCT_timeout
  84 : 	[8201] SRDO_SCT_timeout
  85 : 	[8201] SRDO_SCT_timeout
  86 : 	[8201] SRDO_SCT_timeout
  87 : 	[8201] SRDO_SCT_timeout
  88 : 	[8201] SRDO_SCT_timeout
  89 : 	[8201] SRDO_SCT_timeout
  90 : 	[8201] SRDO_SCT_timeout
  91 : 	[8201] SRDO_SCT_timeout
  92 : 	[8201] SRDO_SCT_timeout
  93 : 	[8201] SRDO_SCT_timeout
  94 : 	[8201] SRDO_SCT_timeout
  95 : 	[8201] SRDO_SCT_timeout
  96 : 	[8201] SRDO_SCT_timeout
  97 : 	[8201] SRDO_SCT_timeout
  98 : 	[8201] SRDO_SCT_timeout
  99 : 	[8201] SRDO_SCT_timeout
 100 : 	[8201] SRDO_SCT_timeout
 101 : 	[8201] SRDO_SCT_timeout
 102 : 	[8201] SRDO_SCT_timeout
 103 : 	[8201] SRDO_SCT_timeout
 104 : 	[8201] SRDO_SCT_timeout
 105 : 	[8201] SRDO_SCT_timeout
 106 : 	[8201] SRDO_SCT_timeout
 107 : 	[8201] SRDO_SCT_timeout
 108 : 	[8201] SRDO_SCT_timeout
 109 : 	[8201] SRDO_SCT_timeout
 110 : 	[8201] SRDO_SCT_timeout
 111 : 	[8201] SRDO_SCT_timeout
 112 : 	[8201] SRDO_SCT_timeout
 113 : 	[8201] SRDO_SCT_timeout
 114 : 	[8201] SRDO_SCT_timeout
 115 : 	[8201] SRDO_SCT_timeout
 116 : 	[8201] SRDO_SCT_timeout
 117 : 	[8201] SRDO_SCT_timeout
 118 : 	[8201] SRDO_SCT_timeout
 119 : 	[8050] Generic_driver_error
 120 : 	[8201] SRDO_SCT_timeout
 121 : 	[800E] FAULT_ALIM_EXT

fig@x86:/opt/ezw/usr/sbin$ ./remote.py swd_right
   1 : 	[800E] FAULT_ALIM_EXT
   2 : 	[800E] FAULT_ALIM_EXT
   3 : 	[8201] SRDO_SCT_timeout
   4 : 	[800E] FAULT_ALIM_EXT
```

fig-shimooka, 2022-05-10:
Symptoms of slow acquisition of right motor status are sometimes slow even after the second acquisition (sometimes it takes about 20 seconds).

fig-shimooka, 2022-05-11:
Sorry. The right and left motors were reversed.
The emergency switch for the left motor gets the status immediately. There are also fewer timeouts.
The emergency switch for the right motor cannot get a state immediately. There are also many timeouts.

fig-shimooka, 2022-05-11:
The DiffController's publish_safety_functions was set to false, but the result did not change.

<rosparam param="publish_safety_functions">false</rosparam

fig-shimooka, 2022-05-11:
If the state cannot be taken immediately, but within approximately 2 seconds, that would still be fine.
Can you shorten the timeout and make the reaction a bit quicker?

ez-Support, 2022-05-13:
Can you send us the journal of the ros log:
`journalctl -f -u ezw-ros-bringup`

fig-shimooka, 2022-05-15:
```
May 15 15:49:37 x86 sce-ros-bringup.sh[3546]: [ WARN] [1652597377.614382386]: NMT state machine is not OK.
May 15 15:49:37 x86 sce-ros-bringup.sh[3546]: [ WARN] [1652597377.614507412]: PDS state machine is not OK.
May 15 15:49:38 x86 sce-ros-bringup.sh[3546]: [ WARN] [1652597378.624806236]: NMT state machine is not OK.
May 15 15:49:38 x86 sce-ros-bringup.sh[3546]: [ WARN] [1652597378.625198025]: PDS state machine is not OK.
May 15 15:49:39 x86 sce-ros-bringup.sh[3546]: [ WARN] [1652597379.620269694]: NMT state machine is not OK.
May 15 15:49:39 x86 sce-ros-bringup.sh[3546]: [ WARN] [1652597379.620802020]: PDS state machine is not OK.
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: ;544;Controller;FAULT detected;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:54;;SMCDS;INFO;enterInOperationEnabledState;562;Controller;Check SWITCH_ON_DISABLED;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:54;;SMCDS;INFO;enterInOperationEnabledState;582;Controller;Check READY_TO_SWITCH_ON;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:54;;SMCDS;INFO;enterInOperationEnabledState;603;Controller;Check SWITCHED_ON;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:54;;SMCDS;INFO;enterInOperationEnabledState;623;Controller;Check OPERATION_ENABLED;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:54;;SMCDS;ERROR;enterInOperationEnabledState;632;Controller;Not in OPERATION_ENABLED;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:55;;SMCDS;INFO;enterInOperationEnabledState;515;Controller;Check NOT_READY_TO_SWITCH_ON;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:55;;SMCDS;INFO;enterInOperationEnabledState;535;Controller;Check FAULT;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:55;;SMCDS;INFO;enterInOperationEnabledState;544;Controller;FAULT detected;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:55;;SMCDS;INFO;enterInOperationEnabledState;562;Controller;Check SWITCH_ON_DISABLED;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:55;;SMCDS;INFO;enterInOperationEnabledState;582;Controller;Check READY_TO_SWITCH_ON;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:55;;SMCDS;INFO;enterInOperationEnabledState;603;Controller;Check SWITCHED_ON;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:55;;SMCDS;INFO;enterInOperationEnabledState;623;Controller;Check OPERATION_ENABLED;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:55;;SMCDS;ERROR;enterInOperationEnabledState;632;Controller;Not in OPERATION_ENABLED;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:56;;SMCDS;INFO;enterInOperationEnabledState;515;Controller;Check NOT_READY_TO_SWITCH_ON;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:56;;SMCDS;INFO;enterInOperationEnabledState;535;Controller;Check FAULT;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:56;;SMCDS;INFO;enterInOperationEnabledState;544;Controller;FAULT detected;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:56;;SMCDS;INFO;enterInOperationEnabledState;562;Controller;Check SWITCH_ON_DISABLED;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:56;;SMCDS;INFO;enterInOperationEnabledState;582;Controller;Check READY_TO_SWITCH_ON;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:56;;SMCDS;INFO;enterInOperationEnabledState;603;Controller;Check SWITCHED_ON;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:56;;SMCDS;INFO;enterInOperationEnabledState;623;Controller;Check OPERATION_ENABLED;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:56;;SMCDS;ERROR;enterInOperationEnabledState;632;Controller;Not in OPERATION_ENABLED;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:57;;SMCDS;INFO;enterInOperationEnabledState;515;Controller;Check NOT_READY_TO_SWITCH_ON;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:57;;SMCDS;INFO;enterInOperationEnabledState;535;Controller;Check FAULT;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:57;;SMCDS;INFO;enterInOperationEnabledState;544;Controller;FAULT detected;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:57;;SMCDS;INFO;enterInOperationEnabledState;562;Controller;Check SWITCH_ON_DISABLED;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:57;;SMCDS;INFO;enterInOperationEnabledState;582;Controller;Check READY_TO_SWITCH_ON;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:57;;SMCDS;INFO;enterInOperationEnabledState;603;Controller;Check SWITCHED_ON;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:57;;SMCDS;INFO;enterInOperationEnabledState;623;Controller;Check OPERATION_ENABLED;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:57;;SMCDS;ERROR;enterInOperationEnabledState;632;Controller;Not in OPERATION_ENABLED;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:58;;SMCDS;INFO;enterInOperationEnabledState;515;Controller;Check NOT_READY_TO_SWITCH_ON;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:58;;SMCDS;INFO;enterInOperationEnabledState;535;Controller;Check FAULT;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:58;;SMCDS;INFO;enterInOperationEnabledState;544;Controller;FAULT detected;
May 15 15:49:40 x86 sce-ros-bringup.sh[3546]: 2022-5-15 12:21:58;;SMCDS;INFO;enterInOperationEnabledState;562;Controller;Check SWITCH_ON_DISABLED;
[... cut ...]
May 15 15:49:55 x86 sce-ros-bringup.sh[3546]: [ WARN] [1652597395.097485203]: PDS state machine is not OK.
May 15 15:49:55 x86 sce-ros-bringup.sh[3546]: [ERROR] [1652597395.930099630]: Failed to set PDS state for right motor, EZW_ERR: SMCService : Controller::enterInOperationEnabledState() return error code : 6
May 15 15:49:55 x86 sce-ros-bringup.sh[3546]: [ WARN] [1652597395.930173480]: PDS state machine is not OK.
May 15 15:49:57 x86 sce-ros-bringup.sh[3546]: [ WARN] [1652597397.658740483]: PDS state machine is not OK
```

ez-Support, 2022-05-16:
The fastest way to achieve if there was a problem/ an STO was triggered and stopped the motor state machine, is to look into the NMT state of the node/motor. If the node's NMT state has been set to pre-operational (7F), then you can consider that a problem/emergency happened. 
To get the NMT state, use the `NMTCommand `library and `getNmtState()` method

fig-shimooka, 2022-05-20:
I have modified cdTimerStateMachin as follows.
The result was no good.
It always outputs 1.

```
        void DiffDriveController::cbTimerStateMachine()
        {
            // NMT state machine
            smccore::Controller::NMTState nmt_state_l, nmt_state_r;
            smccore::Controller::PDSState pds_state_l, pds_state_r;
            ezw_error_t                   err_l, err_r;

            nmt_state_l = nmt_state_r = smccore::Controller::NMTState::UNKNOWN;
            pds_state_l = pds_state_r = smccore::Controller::PDSState::SWITCH_ON_DISABLED;
            err_l = m_left_controller.getNMTState(nmt_state_l);
            err_r = m_right_controller.getNMTState(nmt_state_r);

            ROS_WARN("err_l %d", (int)err_l);
            ROS_WARN("err_r %d", (int)err_r);
            
```

letis33, 2022-05-20:
The result is good => 1 means ERROR _NONE

```    
    ERROR_NOT_INITIALIZED = 0
    ERROR_NONE = 1
    ERROR_HARDWARE = 2
    ERROR_CONFIG = 3
    ERROR_ARGUMENT = 4
    ERROR_INTERNAL = 5
    ERROR_EXTERNAL = 6```

GMezWheel, 2022-05-20:
The result is good => 1 means ERROR _NONE

```    
    ERROR_NOT_INITIALIZED = 0
    ERROR_NONE = 1
    ERROR_HARDWARE = 2
    ERROR_CONFIG = 3
    ERROR_ARGUMENT = 4
    ERROR_INTERNAL = 5
    ERROR_EXTERNAL = 6```

ez-Support, 2022-05-20:
Can you try to set up a configuration and see if it solves your problem.
We will configure that the NMT states of the motor/nodes, will not go to PreOp, after an emergency button was pressed.

To do so, in your commissioning.py script, create a dbus client instance, e.g. `dbus_client = CANOpenServiceDBusClient()` after having imported the library CANOpenServiceDBusClient `from cosdbusclient.can_open_service import CANOpenServiceDBusClient`.

Then, in the commissionning of both motors (left and right), set this CAN Object to 1:
`commissioning.dbus_client.setValueUInt8(0x10, 0x10290108, 1)`
This permits to do not stop the node from sending SRDO data, after emergency button was pressed. I hope this wil solve mainly this issue

ez-Support, 2022-05-20:
Can you try to set up a configuration and see if it solves your problem.
We will configure that the NMT states of the motor/nodes, will not go to PreOp, after an emergency button was pressed.

To do so, in your commissioning.py script, create a dbus client instance, e.g. `dbus_client = CANOpenServiceDBusClient()` after having imported the library CANOpenServiceDBusClient `from cosdbusclient.can_open_service import CANOpenServiceDBusClient`.

Then, in the commissionning of both motors (left and right), set this CAN Object to 1:
`commissioning.dbus_client.setValueUInt8(0x10, 0x10290108, 1)`, after having updating the SLS parameters
This permits to do not stop the node from sending SRDO data, after emergency button was pressed. I hope this wil solve mainly this issue

fig-shimooka, 2022-05-23:
I have added that setting.
But the result was the same.

**commissioning.py add func**
```
def set_no_preop():
    from cosdbusclient.can_open_service import CANOpenServiceDBusClient
    dbus_client = CANOpenServiceDBusClient()
    dbus_client.setValueUInt8(0x10, 0x10290108, 1)
```

**swd_left_4_commissioning.py**
```
def main(argv):
    instance_id = "swd_left"
    node_id = 0x4
    polarity = True  # velocity demand value/motor revolution increments shall be multiplied by –1 if True
    vl_dec_delta_speed = 6000
    restart_acknowledge_behavior = False
    sls_vl_limit = 680
    sls_vl_time_monitoring = 1000
    

   ～～～～～～～～～～～～～～～～～～～～～～～
    #
    # Update STO parameters
    #
    commissioning.update_STO_parameters(restart_acknowledge_behavior)

    #
    # Update SLS parameters
    #
    commissioning.update_SLS_parameters(sls_vl_limit, sls_vl_time_monitoring)
    commissioning.set_no_preop()       ## **ADD**
    # Save modified parameters
    error = commissioning.communication_client.storeParameters(BlocId.ALL)
    commissioning.check("storeParameters", 1)  # error)
　
   ～～～～～～～～～～～～～～～～～～～～～～～
```

**swd_right_5_commissioning.py**
```

def main(argv):

    instance_id = "swd_right"
    node_id = 0x05
    polarity = False  # velocity demand value/motor revolution increments shall be multiplied by –1 if True
    vl_dec_delta_speed = 6000
    restart_acknowledge_behavior = False
    sls_vl_limit = 680
    sls_vl_time_monitoring = 1000

   ～～～～～～～～～～～～～～～～～～～～～～～
    commissioning.update_STO_parameters(restart_acknowledge_behavior)

    #
    # Update SLS parameters
    #
    commissioning.update_SLS_parameters(sls_vl_limit, sls_vl_time_monitoring)
    commissioning.set_no_preop()       ## **ADD**

    # Save modified parameters
    error = commissioning.communication_client.storeParameters(BlocId.ALL)
    commissioning.check("storeParameters", 1)  # error)

   ～～～～～～～～～～～～～～～～～～～～～～～
```

[... 79 lines cut ...]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: False, position_polarity: False ........................... [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 355 .......................................................................... [OK]
OdometryValue        : 341 .......................................................................... [OK]
<COM> Switch NodeID = i Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
EMCY list :
   1 :  [800E] FAULT_ALIM_EXT
   2 :  [8201] SRDO_SCT_timeout
   3 :  [800E] FAULT_ALIM_EXT
   4 :  [800E] FAULT_ALIM_EXT
   5 :  [800E] FAULT_ALIM_EXT
```

fig-shimooka, 2022-05-23:
```
May 23 10:21:22 x86 sce-ros-bringup.sh[3595]: [ WARN] [1653268882.519135503]: PDS state machine is not OK.
May 23 10:21:23 x86 sce-ros-bringup.sh[3595]: [ERROR] [1653268883.519838798]: Failed to set PDS state for right motor, EZW_ERR: SMCService : Controller::enterInOperationEnabledState() return error code : 6
May 23 10:21:23 x86 sce-ros-bringup.sh[3595]: [ WARN] [1653268883.519908396]: PDS state machine is not OK.
May 23 10:21:24 x86 sce-ros-bringup.sh[3595]: [ERROR] [1653268884.516278703]: Failed to set PDS state for right motor, EZW_ERR: SMCService : Controller::enterInOperationEnabledState() return error code : 6
May 23 10:21:24 x86 sce-ros-bringup.sh[3595]: [ WARN] [1653268884.516348180]: PDS state machine is not OK.
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: roller;Check SWITCHED_ON;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:20;;SMCDS;INFO;enterInOperationEnabledState;623;Controller;Check OPERATION_ENABLED;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:20;;SMCDS;ERROR;enterInOperationEnabledState;632;Controller;Not in OPERATION_ENABLED;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:21;;SMCDS;INFO;enterInOperationEnabledState;515;Controller;Check NOT_READY_TO_SWITCH_ON;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:21;;SMCDS;INFO;enterInOperationEnabledState;535;Controller;Check FAULT;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:21;;SMCDS;INFO;enterInOperationEnabledState;544;Controller;FAULT detected;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:21;;SMCDS;INFO;enterInOperationEnabledState;562;Controller;Check SWITCH_ON_DISABLED;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:21;;SMCDS;INFO;enterInOperationEnabledState;582;Controller;Check READY_TO_SWITCH_ON;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:21;;SMCDS;INFO;enterInOperationEnabledState;603;Controller;Check SWITCHED_ON;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:21;;SMCDS;INFO;enterInOperationEnabledState;623;Controller;Check OPERATION_ENABLED;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:21;;SMCDS;ERROR;enterInOperationEnabledState;632;Controller;Not in OPERATION_ENABLED;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:22;;SMCDS;INFO;enterInOperationEnabledState;515;Controller;Check NOT_READY_TO_SWITCH_ON;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:22;;SMCDS;INFO;enterInOperationEnabledState;535;Controller;Check FAULT;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:22;;SMCDS;INFO;enterInOperationEnabledState;544;Controller;FAULT detected;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:22;;SMCDS;INFO;enterInOperationEnabledState;562;Controller;Check SWITCH_ON_DISABLED;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:22;;SMCDS;INFO;enterInOperationEnabledState;582;Controller;Check READY_TO_SWITCH_ON;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:22;;SMCDS;INFO;enterInOperationEnabledState;603;Controller;Check SWITCHED_ON;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:22;;SMCDS;INFO;enterInOperationEnabledState;623;Controller;Check OPERATION_ENABLED;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:22;;SMCDS;ERROR;enterInOperationEnabledState;632;Controller;Not in OPERATION_ENABLED;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:23;;SMCDS;INFO;enterInOperationEnabledState;515;Controller;Check NOT_READY_TO_SWITCH_ON;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:23;;SMCDS;INFO;enterInOperationEnabledState;535;Controller;Check FAULT;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:23;;SMCDS;INFO;enterInOperationEnabledState;544;Controller;FAULT detected;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:23;;SMCDS;INFO;enterInOperationEnabledState;562;Controller;Check SWITCH_ON_DISABLED;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:23;;SMCDS;INFO;enterInOperationEnabledState;582;Controller;Check READY_TO_SWITCH_ON;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:23;;SMCDS;INFO;enterInOperationEnabledState;603;Controller;Check SWITCHED_ON;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:23;;SMCDS;INFO;enterInOperationEnabledState;623;Controller;Check OPERATION_ENABLED;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:23;;SMCDS;ERROR;enterInOperationEnabledState;632;Controller;Not in OPERATION_ENABLED;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:24;;SMCDS;INFO;enterInOperationEnabledState;515;Controller;Check NOT_READY_TO_SWITCH_ON;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:24;;SMCDS;INFO;enterInOperationEnabledState;535;Controller;Check FAULT;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:24;;SMCDS;INFO;enterInOperationEnabledState;544;Controller;FAULT detected;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:24;;SMCDS;INFO;enterInOperationEnabledState;562;Controller;Check SWITCH_ON_DISABLED;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:24;;SMCDS;INFO;enterInOperationEnabledState;582;Controller;Check READY_TO_SWITCH_ON;
May 23 10:21:25 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:24;;SMCDS;INFO;enterInOperationEnabledState;603;Controller;Check SWITCHED_ON;
[... cut ...]
May 23 10:21:36 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:31;;SMCDS;INFO;enterInOperationEnabledState;603;Controller;Check SWITCHED_ON;
May 23 10:21:36 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:31;;SMCDS;INFO;enterInOperationEnabledState;623;Controller;Check OPERATION_ENABLED;
May 23 10:21:36 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:31;;SMCDS;ERROR;enterInOperationEnabledState;632;Controller;Not in OPERATION_ENABLED;
May 23 10:21:36 x86 sce-ros-bringup.sh[3595]: 2022-5-23 10:21:32;;SMCDS;INFO;enterInOperationEnabledState;515;Controller;Check NOT_READY_TO_SWITCH_O
```

fig-shimooka, 2022-05-24:
Yes, it currently stops as soon as the emergency switch on the right is pressed.

That is not a problem, What I am concerned about is that pressing the emergency switch on the left side stops the update of "candamp can0 |grep 10A".
The emergency switch on the left side stops as soon as it is pressed, but sometimes the status is not updated for a long time.
The emergency switch on the right is not a problem, as it is updated immediately.

fig-shimooka, 2022-05-24:
https://github.com/ezWheelSAS/swd_ros_controllers/issues/12#issuecomment-1113891867
This is the current situation.

GMezWheel, 2022-05-24:
0x10A is a SRDO (TX) transmited by SWD Right motor.
If NMT state is PREOP, SRDO are no more transmit.
Can you verify that NMT state of the Right motor is not PREOP when 0x10A is not updated ?

fig-shimooka, 2022-05-25:
https://github.com/ezWheelSAS/swd_ros_controllers/issues/16#issuecomment-1132784596
After implementing this, the emergency state is no longer removed after a reset.
Therefore, I have reverted it back to its original state.
The following is the behaviour when no preop is not applied.

**During normal working hours**
```
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 724434949 .................................................................... [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : OPERATION_ENABLED  ........................................................... [OK]
NbError              : 45 ........................................................................... [OK]
LastError            : 8201 : SRDO_SCT_timeout ...................................................... [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, True, True, True, True, True, True  .............................. [OK]
Peripheral           : True, True, True, True, True, True  .......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 164 .......................................................................... [OK]
OdometryValue        : 157 .......................................................................... [OK]
<COM> Switch NodeID = i Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0

 ./remote.py swd_right
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 724434946 .................................................................... [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
[... cut ...]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -281 ......................................................................... [OK]
OdometryValue        : -270 ......................................................................... [OK]
```

fig-shimooka, 2022-05-25:
https://github.com/ezWheelSAS/swd_ros_controllers/issues/16#issuecomment-1132784596
The results after applying this change are as follows.

**During normal working hours**
```
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 724434946 .................................................................... [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : OPERATION_ENABLED  ........................................................... [OK]
NbError              : 8 ............................................................................ [OK]
LastError            : 800E : FAULT_ALIM_EXT ........................................................ [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, False, False, False, False, True, True  .......................... [OK]
Peripheral           : True, True, True, True, True, True  .......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: False, position_polarity: False ........................... [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 2 ............................................................................ [OK]
OdometryValue        : 1 ............................................................................ [OK]

SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 724434949 .................................................................... [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : OPERATION_ENABLED  ........................................................... [OK]
NbError              : 65 ........................................................................... [OK]
LastError            : 8201 : SRDO_SCT_timeout ...................................................... [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
[... cut ...]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 2 ............................................................................ [OK]
OdometryValue        : 1 ............................................................................ [OK]
```

GMezWheel, 2022-05-25:
Hi @fig-shimooka ,

Indeed, during "status not being updated", swd right is in NMT PREOP state; thus SRDO 0x10A is no longer transmitted.

Maybe the NMT state is not well emitted or received ?
`candump can0 | grep 0x705`
=> it can occur during heavy can bus loading for example.

`ip -details -statistics link show can0`
```
6: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state ERROR-ACTIVE restart-ms 100 
	  bitrate 1000000 sample-point 0.750 
	  tq 125 prop-seg 2 phase-seg1 3 phase-seg2 2 sjw 1
	  pcan_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 8000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 
    RX: bytes  packets  errors  dropped overrun mcast   
    417476193  104636785 0       52      0       0       
    TX: bytes  packets  errors  dropped carrier collsns 
    163447268  55977746 0       0       0       0   
```

There is also an other problem, the count error has not been incremented when you have pressed emergency right:
=> still 8 ???
Can you check it again ?

GMezWheel, 2022-05-25:
With the remote tool, check that the number error changed.

fig-shimooka, 2022-05-26:
https://github.com/ezWheelSAS/swd_ros_controllers/issues/12#issuecomment-1113891867
This setting may not be deactivated when Emergency Stop2 is deactivated (permanently).

fig-shimooka, 2022-05-26:
There is also an other problem, the count error has not been incremented when you have pressed emergency right:
=> still 8 ???
Yes, the Nberror value is 8, no matter how many times you try.

fig-shimooka, 2022-05-26:
"candump can0 | grep 0x705" results in nothing being displayed

fig-shimooka, 2022-05-26:
can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/can  promiscuity 0 minmtu 0 maxmtu 0
    can state ERROR-ACTIVE restart-ms 100
          bitrate 1000000 sample-point 0.750
          tq 62 prop-seg 5 phase-seg1 6 phase-seg2 4 sjw 1
          gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp-inc 1
          clock 48000000
          re-started bus-errors arbit-lost error-warn error-pass bus-off
          0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
    RX: bytes  packets  errors  dropped overrun mcast
    36788931   9837714  0       0       0       0
    TX: bytes  packets  errors  dropped carrier collsns
    20335856   7911935  0       0       0       0

fig-shimooka, 2022-05-26:
"clock 8000000" This value is different. Is this the cause?

GMezWheel, 2022-05-26:
Sorry for the mistake in the command. Try again with:
candump can0 | grep 705
This is the heartbeat send by the right motor (node 5) which contains the NMT state.

GMezWheel, 2022-05-26:
I think that swd right motor has not been re-commissioning with not enter in NMT state PREOP when FAULT mode (due to EMG for example).
Can you check the right commissioning ?

ez-Support, 2022-05-30:
dbus_client.setValueUInt8(0x10, 0x10290108, 1)
In your commissioning.py, the value ti set is 0x20290208. So, the right code is:
`    dbus_client.setValueUInt8(0x10, 0x10290208, 1)`

GMezWheel, 2022-05-30:
We confirm that sometimes, the both motor stay in NMT preop and the SWD diff drive controller need several retry and that can take a long time (10-20s).
We will send you a fix proposal tomorow.

GMezWheel, 2022-05-31:
The delay to enter in PDS OperationEnabled state at startup when NMT state is in PREOP or when PDS is no more in FAULT state (after EMCY released) has been improved with the following modifications:
* change srdo param SCT/SRVT to 50/20 in RX and 25/20 in TX (cf. commissioning files)
[commissioning.idec.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/8804409/commissioning.idec.zip)
* change cbTimerStateMachine in order to send NMT OPER command at the same time to the both motors
* reduce cbTimerStateMachine timer to 0.5
[DiffDriveController.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/8804455/DiffDriveController.zip)
* change libezw-canopen.so in order to allow nodeid 0x0 (broadcast)
[libezw-canopen.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/8804462/libezw-canopen.zip)

commit 6bbd9bed (2022-06-10): Merge pull request #24 from ezWheelSAS/issue-16

Fixed #16 : delay to enter in PDS OperationEnabled
commit 885d2321 (2022-06-10): Fixed #16 : delay to enter in PDS OperationEnabled
closed 2022-06-10 by GMezWheel