IDEC-ezWheel/swd_ros_controllers #17: During SLS, it meanders from left to right.

fig-shimooka, 2022-05-12:
During SLS, it meanders from left to right.
The sls_vl_limit is set to 535.
The cmd_vel at this time is 0.6 for linear.x and 0 for angular.z
Drop linear.x to about 0.3 and it will stabilise.

https://user-images.githubusercontent.com/72783232/168028775-51751968-b75b-40f5-9fdc-b7281eb18292.mp4

.

fig-shimooka, 2022-05-12:
```

DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 724434946 .................................................................... [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : OPERATION_ENABLED  ........................................................... [OK]
NbError              : 0 ............................................................................ [OK]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, False, False, False, False, True, True  .......................... [OK]
Peripheral           : True, True, True, True, True, True  .......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: False, position_polarity: False ........................... [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -602 ......................................................................... [OK]
OdometryValue        : -579 ......................................................................... [OK]

DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 724434949 .................................................................... [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : OPERATION_ENABLED  ........................................................... [OK]
NbError              : 0 ............................................................................ [OK]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
[... cut ...]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -616 ......................................................................... [OK]
OdometryValue        : -593 ......................................................................... [OK]

```

ez-Support, 2022-05-16:
This phenomenon happens when the command is slower than the error computed, so that you have got this oscillation.
You can try to play with the `m_pub_freq_hz` parameter. 
You can also try the ROS2 controller that was released https://github.com/ezWheelSAS/swd_ros2_controllers, and see if it helps.

fig-shimooka, 2022-05-20:
Changed m_pub_freq_hz (5~50).
However, the result did not change.
The CPU load is below 60% at this time.

fig-shimooka, 2022-05-26:
I think the problem is due to this: when the state updates to SLS, the aircraft bends.
It is as if the left and right motors are not synchronised.

https://user-images.githubusercontent.com/72783232/170421091-bdee9e1a-696e-4f47-889a-fbf794662cbf.mp4

GMezWheel, 2022-05-30:
Can you send me a full can log during amr bends (30s for example) ?
`candump can0 -tz > candump.log`

GMezWheel, 2022-05-30:
We have reproduce your problem with your SWD ROS Diff Drive Controller settings.
The parameter "wheel_safety_limited_speed_rpm" is incoherent with the SLS speed limit into your commissioning.py file : 
=> respectively 1120 (80*14) and 680 rpm motor.
We propose to reduce the "wheel_safety_limited_speed_rpm" parameter to 30 rpm wheel (30*14=420 rpm motor).
` <rosparam param="wheel_safety_limited_speed_rpm">30</rosparam>`

fig-shimooka, 2022-05-31:
I changed wheel_safety_limited_speed_rpm to 30 and restarted.
However, the SWD Core still meanders.

This is the log when travelling at 0.5 m/s during SLS conditions.
[candump.txt](https://github.com/ezWheelSAS/swd_ros_controllers/files/8801683/candump.txt)

fig-shimooka, 2022-05-31:
The sls_vl_limit at this time is 680

fig-shimooka, 2022-05-31:
https://github.com/ezWheelSAS/swd_ros_controllers/issues/18#issuecomment-1141378824
I have modified swd_diff_controller to apply this fix and the meandering has improved.

1. modified the setSpeeds function
2. changed rosparam's publish_safety_function from false to true
3. modified Duration in m_timer_safety
```
m_timer_safety = m_nh->createTimer(ros::Duration(1.0 / 5.0), boost::bind(&DiffDriveController::cbTimerSafety, this));
↓.
m_timer_safety = m_nh->createTimer(ros::Duration(1.0 / 1.0), boost::bind(&DiffDriveController::cbTimerSafety, this));
```

fig-shimooka, 2022-05-31:
https://github.com/ezWheelSAS/swd_ros_controllers/issues/12#issuecomment-1110963264

In the course of addressing this issue, changes were made to obtain the safety status from the CAN.
At that time, the publish_safety_function was set to false.
However, we realised that we were referencing the value of m_safety_msg in the setSpeeds function.
I assume this is the cause of the meandering.

commit da11d297 (2022-06-10): Fixed #17 : avoid to be link with safety message function into the set_speed function
closed 2022-06-10 by GMezWheel