IDEC-ezWheel/swd_ros_controllers #19: How do I recover the starter-kit "ARM iMX6 pre-installed with Linux, ROS and SWD drivers"

stakakusu, 2022-05-18:
I'm dealing with starter-kit.
but I don't want to modify default settings since I need to show the factory settings to business partners as released product image sometime. In this case, if I  can recover the factory image by any steps, I can modify the default settings. Is there any way?

stakakusu, 2022-05-19:
If I get "TEK3-IMX6" and finish setting up "IMX6/IMX7 Platforms for Ubuntu 20.04 with X11 Framework", what else do I need?
"TEK3-IMX6": https://www.technexion.com/shop/embedded-computing/fanless-computing/tek3-series/tek3-imx6/
"IMX6/IMX7 Platforms for Ubuntu 20.04 with X11 Framework": https://github.com/TechNexion-customization/ubuntu-tn-imx/blob/master/README-imx6.md#imx6imx7-platforms-for-ubuntu-2004-with-x11-framework

ez-Support, 2022-05-19:
You will find in [http://packages.ez-wheel.com:8081/ubuntu/images/](url) the image of the SD card to restore the Stater Kit [arm32: tek3_starter_kit_arm32.gz](url).

1. Flash the 16GB-SD Card (e.g. `/dev/sdc`): 
`dd if=tek3_starter_kit_arm32.gz | gunzip -c | sudo dd bs=16M of=/dev/sdc status=progress`.
2. Insert SD card in the IPC.
3. Connect the ethernet cable of the IPC, to the internet
4. Turn on the IPC and keep pressing the S1 button for 10 seconds.
5. During the installation PC's led will keep blinking. Wait for the 4 leds to stay in solid green.
6. Restart the IPC and edit host name in `/etc/hosts`, by replacing technexion by StarterKit's default name.

stakakusu, 2022-06-14:
I was trying to buy "TEK3-IMX6" from Japanese supplier but they said "delivery takes  2 month"...
So I haven't checked that this restoring steps works or not, yet.

What I did was checking the uploaded ".gz" file by MacBook like below image
[image]

stakakusu, 2023-02-14:
@HirokiShikazaki

HirokiShikazaki, 2023-02-14:
> You will find in http://packages.ez-wheel.com:8081/ubuntu/images/ the image of the SD card to restore the Stater Kit [ tek3_starter_kit_arm32.gz](http://packages.ez-wheel.com:8081/ubuntu/images/tek3_starter_kit_arm32.gz).
> 
> 1. Flash the 16GB-SD Card (e.g. `/dev/sdc`):
>    `dd if=tek3_starter_kit_arm32.gz | gunzip -c | sudo dd bs=16M of=/dev/sdc status=progress`.
> 2. Insert SD card in the IPC.
> 3. Connect the ethernet cable of the IPC, to the internet
> 4. Turn on the IPC and keep pressing the S1 button for 10 seconds.
> 5. During the installation PC's led will keep blinking. Wait for the 4 leds to stay in solid green.
> 6. Restart the IPC and edit host name in `/etc/hosts`, by replacing technexion by StarterKit's default name.

I was trying write the image of the SD card to the ROM of the IPC.
However, after turning on the power of the IPC, I keep pressing the S1 button for 10 seconds, but it still does not write to the ROM.
Is my procedure wrong?

GMezWheel, 2023-03-10:
Be sure to press the S1 button before turning on the power of the IPC; otherwise the IPC will not boot on the SD card.
You can connect a serial cable in order to see what wrong at boot in PuTTY for example.

closed 2023-04-03 by GMezWheel