IDEC-ezWheel/swd_ros_controllers #20: Build of source code of swd_ros_controllers fails

suguruhamada1003910, 2022-05-20:
I could not build the source code.
Error message is below.

> [ 60%] Built target swd_ros_controllers_generate_messages_eus
> [ 70%] Built target swd_ros_controllers_generate_messages_cpp
> [ 70%] Built target swd_ros_controllers_generate_messages
> [ 80%] Building CXX object swd_ros_controllers/CMakeFiles/swd_diff_drive_controller.dir/src/diff_drive_controller/DiffDriveController.cpp.o
> /home/suguru/ros_ws/src/swd_ros_controllers/src/diff_drive_controller/DiffDriveController.cpp:11:10: fatal error: ezw-canopen-service/DBusClient.hpp: No such file or directory
>    11 | #include "ezw-canopen-service/DBusClient.hpp"
>       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> compilation terminated.
> make[2]: *** [swd_ros_controllers/CMakeFiles/swd_diff_drive_controller.dir/build.make:63: swd_ros_controllers/CMakeFiles/swd_diff_drive_controller.dir/src/diff_drive_controller/DiffDriveController.cpp.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:1032: swd_ros_controllers/CMakeFiles/swd_diff_drive_controller.dir/all] Error 2
> make: *** [Makefile:141: all] Error 2
> Invoking "make install -j4 -l4" failed

Then, I noticed that "ezw-canopen-service" folder does not exist in my PC.
And 1 package is not installed when I execute installing of swd-services.

> suguru@C-Latitude-7280:~$ sudo apt-get update && sudo apt install swd-services
> Hit:1 http://archive.ubuntulinux.jp/ubuntu focal InRelease
> Hit:2 http://jp.archive.ubuntu.com/ubuntu focal InRelease                                                                                      
> Hit:3 http://archive.ubuntulinux.jp/ubuntu-ja-non-free focal InRelease                                                                         
> Hit:4 http://jp.archive.ubuntu.com/ubuntu focal-updates InRelease                                                                              
> Hit:5 http://jp.archive.ubuntu.com/ubuntu focal-backports InRelease                                                                            
> Hit:6 https://dl.google.com/linux/chrome/deb stable InRelease                                                                        
> Hit:7 http://security.ubuntu.com/ubuntu focal-security InRelease                                                              
> Hit:8 http://packages.ez-wheel.com:8081/ubuntu focal InRelease                                                               
> Hit:9 http://packages.ros.org/ros/ubuntu focal InRelease
> Reading package lists... Done
> N: Skipping acquire of configured file 'main/binary-i386/Packages' as repository 'http://packages.ez-wheel.com:8081/ubuntu focal InRelease' doesn't support architecture 'i386'
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> swd-services is already the newest version (0.2.3).
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

I am thinking the cause of this problem is the architecture of CPU of my PC is not correct. Is it correct?
Is there a solution instead of changing my CPU?

GMezWheel, 2022-05-20:
Hi @suguruhamada1003910 ,

The ROS1 swd controller is only compatible with swd-service version 0.1.x
The ROS2 swd controller is only compatible with swd-service version 0.2.x

Thus, downgrade swd-service to 0.1.2 in order to compile ROS1 swd_controller.

NOTA : swd-service package exists only for armhf and amd64 architecture.

suguruhamada1003910, 2022-05-23:
@GMezWheel 

Thank you for reply. I changed version of swd-service from 0.2.3 to 0.1.2 and clone source code of "Tag 2.0.0". Then, I could compile source code!

closed 2022-05-23 by suguruhamada1003910