IDEC-ezWheel/swd_ros_controllers #21: "configuration.json" is not found

suguruhamada1003910, 2022-05-25:
When I build source code and launch swd_ros_controllers, error is displayed as "Can't load Json configuration file".

> suguru@C-Latitude-7280:~/ros_ws$ roslaunch swd_ros_controllers swd_diff_drive_controller.launch
> ... logging to /home/suguru/.ros/log/55da996a-dbca-11ec-a3b4-17b4142a270b/roslaunch-C-Latitude-7280-43261.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://C-Latitude-7280:43631/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /rosdistro: noetic
>  * /rosversion: 1.15.14
>  * /swd_diff_drive_controller/base_frame: base_link
>  * /swd_diff_drive_controller/baseline_m: 0.485
>  * /swd_diff_drive_controller/command_timeout_ms: 500
>  * /swd_diff_drive_controller/control_mode: Twist
>  * /swd_diff_drive_controller/have_backward_sls: False
>  * /swd_diff_drive_controller/left_swd_config_file: /opt/ezw/usr/etc/...
>  * /swd_diff_drive_controller/odom_frame: odom
>  * /swd_diff_drive_controller/positive_polarity_wheel: Right
>  * /swd_diff_drive_controller/pub_freq_hz: 50
>  * /swd_diff_drive_controller/publish_odom: True
>  * /swd_diff_drive_controller/publish_safety_functions: True
>  * /swd_diff_drive_controller/publish_tf: True
>  * /swd_diff_drive_controller/right_swd_config_file: /opt/ezw/usr/etc/...
>  * /swd_diff_drive_controller/wheel_max_speed_rpm: 95
>  * /swd_diff_drive_controller/wheel_safety_limited_speed_rpm: 40
> 
> NODES
>   /
>     swd_diff_drive_controller (swd_ros_controllers/swd_diff_drive_controller)
> 
> auto-starting new master
> process[master]: started with pid [43269]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to 55da996a-dbca-11ec-a3b4-17b4142a270b
> process[rosout-1]: started with pid [43279]
> started core service [/rosout]
> process[swd_diff_drive_controller-2]: started with pid [43286]
> [ INFO] [1653442261.852667393]: Ready !
> [ INFO] [1653442261.855669168]: Initializing swd_diff_drive_controller, node name : /swd_diff_drive_controller
> [ INFO] [1653442261.870771027]: Motors config files, right : /opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini, left : /opt/ezw/usr/etc/ezw-smc-core/swd_left_config.ini
> [default] contextId = 12
> [default] nodeId = 5
> [default] canDevice = can0
> [default] HWConfigurationEntry = SWD_RIGHT
> [default] HWConfigurationFile = /opt/ezw/data/configuration.json
> [default] SWCompatibilityFile = /opt/ezw/firmwares/compatibility.json
> [default] CANOpenEDSFile = /opt/ezw/usr/etc/ezw-canopen-dico/swd_core.eds
> [default] targetVelocityTimeoutMs = 300
> [default] maxMaintenanceSpeedValue = 400
> [default] contextId = 12
> [default] nodeId = 5
> [default] canDevice = can0
> [default] HWConfigurationEntry = SWD_RIGHT
> [default] HWConfigurationFile = /opt/ezw/data/configuration.json
> [default] SWCompatibilityFile = /opt/ezw/firmwares/compatibility.json
> [default] CANOpenEDSFile = /opt/ezw/usr/etc/ezw-canopen-dico/swd_core.eds
> [default] targetVelocityTimeoutMs = 300
> [default] maxMaintenanceSpeedValue = 400
> 2022-5-25 10:31:1;;SMCDS;ERROR;getHWConfigurationMember;81;HardwareHelper;Can't load Json configuration file : /opt/ezw/data/configuration.json;
> [ERROR] [1653442261.871418207]: Failed loading right motor's config file </opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini>, CONTEXT_ID: 0, EZW_ERR: SMCService : Config.init() return error code : 3

And then, I found the "configuration.json" from IPC of Starter Kit.
Usually, how can I get "configuration.jos"? Can I get it with apt command or git?

GMezWheel, 2022-05-25:
Hi @suguruhamada1003910 

Indeed, a Starter Kit is composed of the following ez-Wheel debian packages:
* [swd-starter-kit](http://packages.ez-wheel.com:8081/ubuntu/pool/main/swd-starter-kit_0.1.2_armhf.deb)
* [swd-services](http://packages.ez-wheel.com:8081/ubuntu/pool/main/swd-services_0.1.2_armhf.deb)
* [ros-noetic-swd-ros-controllers](http://packages.ez-wheel.com:8081/ubuntu/pool/main/ros-noetic-swd-ros-controllers_2.0.3-0focal_armhf.deb)
* [ros-noetic-swd-starter-kit-bringup](http://packages.ez-wheel.com:8081/ubuntu/pool/main/ros-noetic-swd-starter-kit-bringup_1.0.0-0focal_armhf.deb)
* [ros-noetic-swd-starter-kit-description](http://packages.ez-wheel.com:8081/ubuntu/pool/main/ros-noetic-swd-starter-kit-description_0.1.0-0focal_armhf.deb)
* [ros-noetic-swd-robot-manager](http://packages.ez-wheel.com:8081/ubuntu/pool/main/ros-noetic-swd-robot-manager_0.1.2-0focal_armhf.deb)

The "configuration.json" is in the **swd-starter-kit** package.

suguruhamada1003910, 2022-05-25:
@GMezWheel 

Thank you for reply.
I tried to install it but the package is not found.

> suguru@C-Latitude-7280:~$ sudo apt install swd-starter-kit
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> E: Unable to locate package swd-starter-kit

suguruhamada1003910, 2022-05-26:
@GMezWheel 

Thank you for reply. I understand.

closed 2022-05-26 by suguruhamada1003910

stakakusu, 2022-06-14:
[configuration.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/8899363/configuration.zip)
Since github doesn't support ".json" extension, I attached zipped "configuration.json" copied from "swd-starter-kit package".