IDEC-ezWheel/swd_ros_controllers #25: Cannot run ros2 controller.

fig-shimooka, 2022-07-14:
I got the latest swd_ros2_controllers.
After building and running it, it fails to start the node.
It seems to be failing to load the configuration file.

```
[INFO] [launch]: All log files can be found below /home/fig/.ros/log/2022-07-14-13-06-41-167843-x86-66118
[INFO] [launch]: Default logging verbosity is set to INFO
SYSTEMCTL_dbus.id detected
[INFO] [swd_diff_drive_controller-1]: process started with pid [66135]
[swd_diff_drive_controller-1] [INFO] [1657771601.654043967] [swd_diff_drive_controller]: publish_odom : 1
[swd_diff_drive_controller-1] [INFO] [1657771601.669690412] [swd_diff_drive_controller]: publish_safety_functions : 1
[swd_diff_drive_controller-1] [INFO] [1657771601.675937567] [swd_diff_drive_controller]: right_config_file : /opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini
[swd_diff_drive_controller-1] [ERROR] [1657771601.677838560] [swd_diff_drive_controller]: Failed loading right motor's config file </opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini>, CONTEXT_ID: 0, EZW_ERR: SMCService : Config.init() return error code : 4
[swd_diff_drive_controller-1] terminate called after throwing an instance of 'std::runtime_error'
[swd_diff_drive_controller-1]   what():  Failed loading right motor's config file
[ERROR] [swd_diff_drive_controller-1]: process has died [pid 66135, exit code -6, cmd '/home/fig/swd_ros2/install/swd_ros2_controllers/lib/swd_ros2_controllers/swd_diff_drive_controller --ros-args -r __node:=swd_diff_drive_controller --params-file /tmp/launch_params_9idso2sl --params-file /tmp/launch_params_mepokmtl --params-file /tmp/launch_params_ochubis0 --params-file /tmp/launch_params_1y0pid0c --params-file /tmp/launch_params__kuu3em7 --params-file /tmp/launch_params_pvcbnk5a --params-file /tmp/launch_params_biy6bxc7 --params-file /tmp/launch_params__s74lrqk --params-file /tmp/launch_params_a65dcu66 --params-file /tmp/launch_params_zweeixdv --params-file /tmp/launch_params_7q630o6w --params-file /tmp/launch_params_0cfzp4_1 --params-file /tmp/launch_params_4g6_bxoy --params-file /tmp/launch_params_d8rugosg --params-file /tmp/launch_params_q6yyf3uw --params-file /tmp/launch_params_86hiba3z --params-file /tmp/launch_params_gdpxalk1'].

```
[ezw.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/9108353/ezw.zip)

fig-shimooka, 2022-07-15:
I gave up trying to run it with ros2-controller and tried again with ros1-controller.
As a result, the build failed.
I have reinstalled swd-service.
Is the latest swd-service incompatible with ros1 and ros2?

```
$ sudo apt-show-versions swd-services -a
swd-services:amd64 0.2.6 install ok installed
swd-services:amd64 0.2.6 focal packages.ez-wheel.com:8081
No stable version
swd-services:amd64/focal 0.2.6 uptodate

[ 81%] Building CXX object swd_ros_controllers/CMakeFiles/swd_diff_drive_controller.dir/src/diff_drive_controller/DiffDriveController_sls.cpp.o
In file included from /opt/ezw/usr/include/ezw-canopen-service/DBusClient.hpp:12,
                 from /home/fig/ros1_swd/src/swd_ros_controllers/src/diff_drive_controller/DiffDriveController_sls.cpp:11:
/opt/ezw/usr/include/ezw-tools/DBusClientDelegate.hpp:17:10: fatal error: CommonAPI/CommonAPI.hpp: No such file or directory
   17 | #include <CommonAPI/CommonAPI.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /opt/ezw/usr/include/ezw-canopen-service/DBusClient.hpp:12,
                 from /home/fig/ros1_swd/src/swd_ros_controllers/src/diff_drive_controller/DiffDriveController.cpp:11:
/opt/ezw/usr/include/ezw-tools/DBusClientDelegate.hpp:17:10: fatal error: CommonAPI/CommonAPI.hpp: No such file or directory
   17 | #include <CommonAPI/CommonAPI.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [swd_ros_controllers/CMakeFiles/swd_diff_drive_controller.dir/build.make:63: swd_ros_controllers/CMakeFiles/swd_diff_drive_controller.dir/src/diff_drive_controller/DiffDriveController.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [swd_ros_controllers/CMakeFiles/swd_diff_drive_controller.dir/build.make:76: swd_ros_controllers/CMakeFiles/swd_diff_drive_controller.dir/src/diff_drive_controller/DiffDriveController_sls.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2107: swd_ros_controllers/CMakeFiles/swd_diff_drive_controller.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

```

GMezWheel, 2022-07-15:
Hi @fig-shimooka ,
Indeed, a new major version of swd-services (0.2.x) is released.
Both ros1 and ros2 swd_diff_drive_controllers have been updated.

Thus :
* ros1 version (<3.0.0) are not compatible with swd-services 0.2.x => downgrade to swd-services 0.1.x
* ros1 version (>=3.0.0) are not compatbile with swd-services 0.1.x => upgrade swd-services to 0.2.x
* ros2 version is not compatbile with swd-services 0.1.x => upgrade swd-services to 0.2.x

GMezWheel, 2022-07-15:
Concerning the ros2 node, the configuration.json file is invalid. 
=> /opt/ezw/data/configuration.json

The new entry SWD_CORE is missing.
cf. /opt/ezw/usr/etc/ezw-smc-core/swd_left_config.ini

An example is given into /opt/ezw/usr/etc/ezw-smc-core/configuration.json.
In your case, add the following entry:
```
   {
      "name": "SWD_CORE",
      "nbStepRevolutionElec": 6,
      "nbPolePair": 5,
      "reduction": 14.0,
      "diameter": 125.0 or 128.7
   }
```
NOTA : The log files of ezw-smc-service are write into /tmp/sce-swd-left.log and  /tmp/sce-swd-right.log files.