IDEC-ezWheel/swd_ros_controllers #27: can't commission

fig-shimooka, 2022-07-21:
I have put in the latest swd-service and ros2-controller and set the firmware to 1.0.1.
The last time I updated the firmware, commissioning was initialized.

I have written the configuration again with this as a reference, but it fails.
What could be the cause.

https://github.com/ezWheelSAS/swd_ros_controllers/issues/7#issuecomment-1073877410

```
SYSTEMCTL_dbus.id detected
create_dbus_clients(swd_left) : OK
Restore factory parameters : OK
Reset to apply parameters : OK
setNetworkParameters(node_id=4) : Failed

Commissioning failed !

```
Same for right.

```
 ./remote.py swd_left
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x0 .......................................................................... [KO]
RT activated         : False ........................................................................ [KO]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x0 .......................................................................... [KO]
Product code         : 0x0 .......................................................................... [KO]
Serial number        : XXXXXXXXXXXX ................................................................. [KO]
Revision number      : 0 ............................................................................ [KO]
SWVersion            :  ............................................................................. [KO]
HWVersion            :  ............................................................................. [KO]
Calibrated           : False ........................................................................ [KO]
SRDO validity        : False ........................................................................ [KO]
NMT State            : UNKNOWN  ..................................................................... [OK]
PDS State            : NOT_READY_TO_SWITCH_ON  ...................................................... [KO]
NbError              : 0 ............................................................................ [KO]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [KO]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [KO]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [KO]
SCW SAFEIN_1         : False, False, False, False, False, False, False, False  ...................... [KO]
Peripheral           : False, False, False, False, False, False  .................................... [KO]
VelocityModeSW       : internal_limit_active: False ................................................. [KO]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [KO]
Polarity             : velocity_polarity: False, position_polarity: False ........................... [KO]
TargetVelocity       : 0 ............................................................................ [KO]
VelocityDemand       : 0 ............................................................................ [KO]
VelocityActualValue  : 0 ............................................................................ [KO]
PositionValue        : 0 ............................................................................ [KO]
OdometryValue        : 0 ............................................................................ [KO]
HallEncoder          : 0 ............................................................................ [KO]
FineOdometryValue    : 0 ............................................................................ [KO]
```

GMezWheel, 2022-07-22:
Hi @fig-shimooka,

WARNING : during commissioning, swd-diff-driver controller has to be stopped. Otherwise, an NMT OPER command is sent and commissioning failed !

Moreover, during commissioning, node id is initialized and is changed to default value (0x10). If commissioning failed, node id stay to default node id. Thus, if the second motor is commissioned too, a conflict wil occured as both motors have the same node id. I think, its your case when commissioning failed at step setNetworkParameter.

Procedure:
* stop swd-diff-drive controller
* with candump, check the node id of the motors : candump can0 | grep '710\|704\|705'
* if both motors have node id 0x10 (710). Power off right motor
* launch remote.py on left motor and connect it to the motor (key 'i' : 16 or 4)
* commission left motor
* power off left motor (not necessary) and power on right motor
* launch remote.py on right motor and connect it to the motor (key 'i' : 16 or 5)
* commission right motor
* power on left motor
* start swd-diff-drive controller

suguruhamada1003910, 2022-07-27:
/opt/ezw/usr/sbin/remote.py

fig-shimooka, 2022-08-16:
The node_id has changed to 6 and 7. Why?
I have disconnected the power to the right motor and run Commission on the left motor.
However, it fails.
I have tried changing the node_id in swd_left_4_commissioning.py to 4, 6 and 7, but it fails.
```

$ candump can0 |grep 7
  can0  706   [1]  05
  can0  707   [1]  05
  can0  706   [1]  05
  can0  707   [1]  05
```

GMezWheel, 2022-08-16:
Hi @fig-shimooka,
The node_id haven't changed to 6 and 7.
In fact, it works in pairs of node id.
Thus, for left motor : 4 is the node_id of the motor and 6 is the node id of the DBUS canopen client.
For right, it is 5 for the motor and 7 for the DBUS canopen client.
Thus node id (**6 and 7**) are also configurable into **/opt/ezw/usr/etc/ezw-smc-core/swd_left_config.ini** 
```
nodeId = 4
coreNodeId = 6
```
and **opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini**
```
nodeId = 5
coreNodeId = 7
```
In your case, as you only have 706 and 707 nodes, it seems that the both motors are off (704 and 705 not visible) (default 710 either).

fig-shimooka, 2022-08-16:
I checked the motor wiring and it seems to be powered on (LED lights up).
Is there any cause other than the wiring?

GMezWheel, 2022-08-16:
When the firmware into the motor start after boot on, it sends automatically its hearbeat (7xx) over the can bus.
Thus either the firmware has failed to start or the can wiring is broken.
You can open the USB motor console to check the firmware boot sequence:
`minicom -D /dev/ttyUSB0 -b 1000000`

fig-shimooka, 2022-08-17:
It seems to be failing to start.
Is this due to the wiring?

```
0x800D Fault alim ext error RAISED
0x800Eapp_init() : CAN_STBY UNLATCHED
app_init() : IGN latch inhibed
app_init() : Safe TimeOut duration set to :7
app_init() : Safe TimeOut activated
ezw_tempsfty_process failed
Unable to ezw_adc_get_voltage on ADC_V_ALIM_EXT
ezw_adc_get_current ADC_MOTOR_U_MEAS failed
ezw_system_error_manager failed
<COMM-Event> Tr Queue Empty

```

fig-shimooka, 2022-08-17:
I tried updating the firmware again, but the result was the same.

```
$ sudo ./ezw-bootloader-tms /dev/ttyUSB0 1000000 GM_05_SWD_v1.0.1_gm.bin  --active_bootloader --active_application
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
lsof: WARNING: can't stat() fuse file system /run/user/1000/doc
      Output information may be incomplete.
 CRC recu du receiver 
PACKET 0 :
316296...........GEN2_V1.........................GEN2-V1-APPLICATION.............VER 1.0........................................ packet 0 EST envoye 
 Ymodem_Transmit : ACK apres packet 0
 Ymodem_Transmit : CRC apres ACK
.....................................................................................................................................................................................................................................................................................................................
SendDataPacket n°0 : EOT packet send
```

GMezWheel, 2022-08-17:
This errors are not fatal. The firmware has started correctly. 
Hereafter my boot sequence:
```
ezw_watchdogtps_init: PMIC configuration skipped because FSM state is: SAFE
ezw_watchdogtps_init: Failed to enter in ACTIVE state
Safe boot OK
CDG6_DATA : 
CDG6_DATA : 37
ezw_tempsfty_process failed
Unable to ezw_adc_get_voltage on ADC_V_ALIM_EXT
ezw_adc_get_current ADC_MOTOR_U_MEAS failed
ezw_system_error_manager failed
<COMM-Event> Tr Queue Empty
GMGM-V1ezw_command_sci_init done
Safe boot OK
CDG6_DATA : 
CDG6_DATA : 37
ezw_tempsfty_process failed
Unable to ezw_adc_get_voltage on ADC_V_ALIM_EXT
ezw_adc_get_current ADC_MOTOR_U_MEAS failed
ezw_system_error_manager failed
<COMM-Event> Tr Queue Empty
```

To ensure the firmware has started correctly, you can use ezw-bootloader-tms to ping applicative software:
```
ezw-bootloader-tms /dev/ttyUSB0 1000000 --command 52 --verbose

Entering boot loader utility for TMS

Send APP_PING ... [OK]

Exiting boot loader utility for TMS
```

Then, check 0x710 HB message:
```
candump can0 | grep 71
  can0  710   [1]  7F
  can0  710   [1]  7F
  can0  710   [1]  7F
  can0  710   [1]  7F
  can0  710   [1]  7F
```

GMezWheel, 2022-08-17:
Maybe the error is this one: 
`app_init() : CAN_STBY UNLATCHED`

I am looking for information about it next week.

fig-shimooka, 2022-08-17:
It seems no response is coming.
There was no 710 in the CAN data.

```
Entering boot loader utility for TMS
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
lsof: WARNING: can't stat() fuse file system /run/user/1000/doc
      Output information may be incomplete.

Send APP_PING ............................................................... [KO]
Application doesn't respond. Exiting

Exception :  Exiting boot loader utility for TMS
```

GMezWheel, 2022-08-17:
It is normal as APP_PING failed (KO) :
`Send APP_PING ............................................................... [KO]`

It confirms that firmware has not started correctly.

Try to start it manually.

Check if bootloader is running:
```
ezw-bootloader-tms /dev/ttyUSB0 1000000 --command 30 --verbose
Send BL_CMD_PING (30)... [OK]
```

Then, run applicative firmware:
```
ezw-bootloader-tms /dev/ttyUSB0 1000000 --command 36 --verbose
Send BL_CMD_EXEC_APP ... [OK]
```

Then, check applicative firmware is running:
```
ezw-bootloader-tms /dev/ttyUSB0 1000000 --command 52 --verbose
Send APP_PING ... [OK]
```

Then, check 0x710 HB message:
```
candump can0 | grep 71
  can0  710   [1]  7F
  can0  710   [1]  7F
  can0  710   [1]  7F
  can0  710   [1]  7F
  can0  710   [1]  7F
```

fig-shimooka, 2022-08-17:
All failed.

```

$ sudo ./ezw-bootloader-tms /dev/ttyUSB0 1000000 --command 30 --verbose
Entering boot loader utility for TMS
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
lsof: WARNING: can't stat() fuse file system /run/user/1000/doc
      Output information may be incomplete.

Send BL_CMD_PING (30)............ [KO]
Bootloader doesn't respond. Exiting  

Exception :  Exiting boot loader utility for TMS

$ sudo ./ezw-bootloader-tms /dev/ttyUSB0 1000000 --command 36 --verbose
Entering boot loader utility for TMS
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
lsof: WARNING: can't stat() fuse file system /run/user/1000/doc
      Output information may be incomplete.

Send BL_CMD_EXEC_APP ............................................................... [KO]
Application doesn't respond. Exiting

Exception :  Exiting boot loader utility for TMS

$ sudo ./ezw-bootloader-tms /dev/ttyUSB0 1000000 --command 52 --verbose
Entering boot loader utility for TMS
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
lsof: WARNING: can't stat() fuse file system /run/user/1000/doc
      Output information may be incomplete.

Send APP_PING ............................................................... [KO]
Application doesn't respond. Exiting

Exception :  Exiting boot loader utility for TMS

```

GMezWheel, 2022-08-17:
No other tools use /dev/ttyUSB0 ? 
Close minicom, ...

fig-shimooka, 2022-08-17:
I discovered something new.
If only the motor is turned on and the ROS-PC is turned off, it responds.
However, command 30 fails.

If I then turn on the ROS-PC, all commands will fail
The ezw-ros-bringup is set to disable, so it is stopped.

```
$ sudo ./ezw-bootloader-tms /dev/ttyUSB0 1000000 --command 30 --verbose
Entering boot loader utility for TMS
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
lsof: WARNING: can't stat() fuse file system /run/user/1000/doc
      Output information may be incomplete.

Send BL_CMD_PING (30)...e.z.w._.c.o.m.m.a. [KO]
Bootloader doesn't respond. Exiting  

Exception :  Exiting boot loader utility for TMS

fig@MOUSE:/opt/ezw/usr/bin$ sudo ./ezw-bootloader-tms /dev/ttyUSB0 1000000 --command 36 --verbose
Entering boot loader utility for TMS
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
lsof: WARNING: can't stat() fuse file system /run/user/1000/doc
      Output information may be incomplete.

Send BL_CMD_EXEC_APP .... [OK]

Exiting boot loader utility for TMS
fig@MOUSE:/opt/ezw/usr/bin$ sudo ./ezw-bootloader-tms /dev/ttyUSB0 1000000 --command 52 --verbose
Entering boot loader utility for TMS
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
lsof: WARNING: can't stat() fuse file system /run/user/1000/doc
      Output information may be incomplete.

Send APP_PING .... [OK]

Exiting boot loader utility for TMS
```

GMezWheel, 2022-08-17:
I will try to reproduce this behavior with the firmware GM_05_SWD_v1.0.1_gm.bin next week.

GMezWheel, 2022-08-17:
Concerning command 30, it is normal as applicative firmware is running (not bootlader firmware).

GMezWheel, 2022-08-22:
Hi @fig-shimooka,
An other way to check the firmware is to press the emergency button. If led (top left) is blinking red thus, the firmware is running.
=> it should emits its HB on can bus.

NOTA : CAN_STBY UNLATCHED log is normal

GMezWheel, 2022-08-26:
@fig-shimooka 
For information, I didn't reproduce your behavior with :
* firmware : 1.0.1 (GM_05_SWD_v1.0.1_gm.bin)
* swd-services : 0.2.6 (package ubuntu x86_64)
* swd-starter-kit-config/commissioning :  v2.0.0 (tag github)

Are you still stuck ?

[image]

[image]

[image]

fig-shimooka, 2022-08-29:
Thank you.
Yes, still not working.

GMezWheel, 2022-08-29:
Have you pressed the emergency button in order to check that the firmware is alive.
=> the top left led shall blinking red
Thus, as the firmware is alive,it should emits its HB on can bus.

GMezWheel, 2022-09-06:
Is the problem closed with the official v1.0.1 firmware ?

fig-shimooka, 2022-09-12:
We have successfully started the motor by applying the official firmware v1.0.1.
Thank you very much.

Now I am trying to run swd-ros2-controller.
But it fails to start because of the left motor setting.
What is the cause?

```
$ ros2 launch swd_ros2_controllers swd_diff_drive_controller.launch.py
[INFO] [launch]: All log files can be found below /home/fig/.ros/log/2022-09-12-15-59-59-225704-x86-49917
[INFO] [launch]: Default logging verbosity is set to INFO
SYSTEMCTL_dbus.id detected
[INFO] [swd_diff_drive_controller-1]: process started with pid [49958]
[swd_diff_drive_controller-1] [INFO] [1662965999.914690081] [swd_diff_drive_controller]: publish_odom : 1
[swd_diff_drive_controller-1] [INFO] [1662965999.921215597] [swd_diff_drive_controller]: publish_safety_functions : 1
[swd_diff_drive_controller-1] [INFO] [1662965999.929884445] [swd_diff_drive_controller]: right_config_file : /opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini
[swd_diff_drive_controller-1] [INFO] [1662965999.948378078] [swd_diff_drive_controller]: Initializing right dbus client !
[swd_diff_drive_controller-1] [INFO] [1662966001.107756692] [swd_diff_drive_controller]: left_config_file : /opt/ezw/usr/etc/ezw-smc-core/swd_left_config.ini
[swd_diff_drive_controller-1] [INFO] [1662966001.108570890] [swd_diff_drive_controller]: Initializing left dbus client !
[swd_diff_drive_controller-1] [INFO] [1662966002.224046110] [swd_diff_drive_controller]: fine_odometry : 0
[swd_diff_drive_controller-1] [ERROR] [1662966002.296459839] [swd_diff_drive_controller]: Failed initial reading from left motor, EZW_ERR: SMCService : Controller::getOdometryValue() return error code : 6
[swd_diff_drive_controller-1] terminate called after throwing an instance of 'std::runtime_error'
[swd_diff_drive_controller-1]   what():  Initial reading from left motor failed
[ERROR] [swd_diff_drive_controller-1]: process has died [pid 49958, exit code -6, cmd '/home/fig/ros2_swd/install/swd_ros2_controllers/lib/swd_ros2_controllers/swd_diff_drive_controller --ros-args -r __node:=swd_diff_drive_controller --params-file /tmp/launch_params_a7cnvfer --params-file /tmp/launch_params_x16ltgh7 --params-file /tmp/launch_params_pn4ktt62 --params-file /tmp/launch_params_cksh0a9k --params-file /tmp/launch_params_j7t50fa4 --params-file /tmp/launch_params_uck3fld5 --params-file /tmp/launch_params_9xmfm4ak --params-file /tmp/launch_params_3r32erq3 --params-file /tmp/launch_params_60ur3x_7 --params-file /tmp/launch_params__lv6llm_ --params-file /tmp/launch_params_h5cift03 --params-file /tmp/launch_params_ycwm6nc0 --params-file /tmp/launch_params_1nq1qx_6 --params-file /tmp/launch_params_1zxa6_sb --params-file /tmp/launch_params_yoc7g4f_ --params-file /tmp/launch_params_8kkxvdf9 --params-file /tmp/launch_params_v9wwuccf'].
```

[ezw_20220912.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/9545540/ezw_20220912.zip)

fig-shimooka, 2022-09-12:
After rebooting, the content changed.

```

[swd_diff_drive_controller-1] [INFO] [1662970470.615029681] [swd_diff_drive_controller]: publish_odom : 1
[swd_diff_drive_controller-1] [INFO] [1662970470.618171721] [swd_diff_drive_controller]: publish_safety_functions : 1
[swd_diff_drive_controller-1] [INFO] [1662970470.625811289] [swd_diff_drive_controller]: right_config_file : /opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini
[swd_diff_drive_controller-1] [INFO] [1662970470.638242471] [swd_diff_drive_controller]: Initializing right dbus client !
[swd_diff_drive_controller-1] [INFO] [1662970471.041958378] [swd_diff_drive_controller]: left_config_file : /opt/ezw/usr/etc/ezw-smc-core/swd_left_config.ini
[swd_diff_drive_controller-1] [INFO] [1662970471.047685059] [swd_diff_drive_controller]: Initializing left dbus client !
[swd_diff_drive_controller-1] [INFO] [1662970472.165448718] [swd_diff_drive_controller]: fine_odometry : 0
[swd_diff_drive_controller-1] [INFO] [1662970472.297729482] [swd_diff_drive_controller]: left motor polarity : True
[swd_diff_drive_controller-1] [INFO] [1662970472.297924243] [swd_diff_drive_controller]: watchdog_receive_ms : 500
[swd_diff_drive_controller-1] [INFO] [1662970472.298053150] [swd_diff_drive_controller]: pub_freq_hz : 20
[swd_diff_drive_controller-1] [INFO] [1662970472.298116993] [swd_diff_drive_controller]: swd_diff_drive_controller initialized successfully!
[swd_diff_drive_controller-1] [ERROR] [1662970472.465604552] [swd_diff_drive_controller]: Failed reading from left motor, EZW_ERR: SMCService : Controller::getOdometryValue() return error code : 6
[swd_diff_drive_controller-1] [INFO] [1662970472.579007530] [swd_diff_drive_controller]: left_encoder_relative_error : 0.200000
[swd_diff_drive_controller-1] [INFO] [1662970472.579154345] [swd_diff_drive_controller]: right_encoder_relative_error : 0.200000
[swd_diff_drive_controller-1] [INFO] [1662970472.579206933] [swd_diff_drive_controller]: baseline_m : 0.485000
[swd_diff_drive_controller-1] [INFO] [1662970472.579298753] [swd_diff_drive_controller]: odom_frame : odom
[swd_diff_drive_controller-1] [INFO] [1662970472.579356344] [swd_diff_drive_controller]: base_frame : base_link
[swd_diff_drive_controller-1] [INFO] [1662970472.579626579] [swd_diff_drive_controller]: publish_tf : 1
[swd_diff_drive_controller-1] [ERROR] [1662970472.682913114] [swd_diff_drive_controller]: Failed reading from right motor, EZW_ERR: SMCService : Controller::getOdometryValue() return error code : 6
[swd_diff_drive_controller-1] [INFO] [1662970472.874751132] [swd_diff_drive_controller]: motor_max_speed_rpm : 1050
[swd_diff_drive_controller-1] [INFO] [1662970472.874866026] [swd_diff_drive_controller]: have_backward_sls : 0
[swd_diff_drive_controller-1] [ERROR] [1662970473.363697730] [swd_diff_drive_controller]: Failed reading from left motor, EZW_ERR: SMCService : Controller::getOdometryValue() return error code : 6
[swd_diff_drive_controller-1] [INFO] [1662970473.509609413] [swd_diff_drive_controller]: NMT state machine is not OK.
[swd_diff_drive_controller-1] [INFO] [1662970473.525596045] [swd_diff_drive_controller]: PDS state machine is not OK.
[swd_diff_drive_controller-1] [ERROR] [1662970474.599773104] [swd_diff_drive_controller]: Error reading SLS from left motor, EZW_ERR: SMCService : Controller::getSafetyFunctionCommand() return error code : 6
[swd_diff_drive_controller-1] [ERROR] [1662970474.865633453] [swd_diff_drive_controller]: Failed setting velocity of left motor, EZW_ERR: SMCService : Controller::setTargetVelocity() return error code : 6
[swd_diff_drive_controller-1] [ERROR] [1662970475.067825448] [swd_diff_drive_controller]: Failed reading from left motor, EZW_ERR: SMCService : Controller::getOdometryValue() return error code : 6
[swd_diff_drive_controller-1] [ERROR] [1662970490.080728134] [swd_diff_drive_controller]: Error reading SLS from left motor, EZW_ERR: SMCService : Controller::getSafetyFunctionCommand() return error code : 5
[swd_diff_drive_controller-1] [INFO] [1662970490.258991315] [swd_diff_drive_controller]: motor_max_safety_limited_speed_rpm : 490
[swd_diff_drive_controller-1] [ERROR] [1662970505.272329531] [swd_diff_drive_controller]: Failed setting velocity of left motor, EZW_ERR: SMCService : Controller::setTargetVelocity() return error code : 5

[swd_diff_drive_controller-1] [ERROR] [1662970520.285522659] [swd_diff_drive_controller]: Failed reading from left motor, EZW_ERR: SMCService : Controller::getOdometryValue() return error code : 5
[swd_diff_drive_controller-1] [ERROR] [1662970535.300097677] [swd_diff_drive_controller]: Error reading SLS from left motor, EZW_ERR: SMCService : Controller::getSafetyFunctionCommand() return error code : 5
[swd_diff_drive_controller-1] [ERROR] [1662970550.312012733] [swd_diff_drive_controller]: Error reading SLS from right motor, EZW_ERR: SMCService : Controller::getSafetyFunctionCommand() return error code : 5

```

fig-shimooka, 2022-09-12:
After several reboots, the system was successfully activated.

However, it seems to be stuck due to SAFETY being applied.
The LED on the scanner is green. The emergency switch is not on either.
Is my commision wrong?

header:
  stamp:
    sec: 1641791029
    nanosec: 95690074
  frame_id: base_link
safe_torque_off: true
safe_brake_control: true
safety_limited_speed: true
safe_direction_indication_forward: false
safe_direction_indication_backward: false

GMezWheel, 2022-09-12:
Hi @fig-shimooka ,

Your SRDO commissioning seems OK.

However, the following timings are recommended :
- SRDO RX : sct=50 and srvt = 20
- SRDO TX : sct = 25 and srvt = 20

You can disable some PDO:
- TPDO1, TPDO2, TPDO_4, RPDO_2 and RPDO_3

And let enabled :
- TPDO_3, RPDO_1 and RPDO_4

There is a mistake into this function :
```def set_no_preop():
dbus_client.setValueUInt8(0x10,  0x10290208, 1)
=> dbus_client.setValueUInt8(0x1029_02_00, 1)  # error_behavior_syserr_for_error => 1 (no change of the NMT state)
```

fig-shimooka, 2022-09-13:
cosdbusclient.can_open_serviceimport is no longer available.
```
  File "/opt/ezw/commissioning/commissioning.py", line 32, in set_no_preop
    from cosdbusclient.can_open_serviceimport import CANOpenServiceDBusClient
ModuleNotFoundError: No module named 'cosdbusclient'
```

dbus_client.setValueUInt8(0x1029_02_00, 1)
Is this necessary?
Is PDO disabled and enabled with or without commenting out setTPDOCommunicationParameters?

fig-shimooka, 2022-09-13:
Since I didn't know, I commented out the dbus_client.setValueUInt8() function and ran it.
However, it does not work even if I give cmd_vel
ros2 topic echo safety topic is also no longer published

```
def set_no_preop():
    pass
    #from cosdbusclient.can_open_serviceimport import CANOpenServiceDBusClient
    #dbus_client = CANOpenServiceDBusClient()
    #dbus_client.setValueUInt8(0x10, 0x10290108, 1)
    # dbus_client.setValueUInt8(0x10,  0x10290208, 1)
    #dbus_client.setValueUInt8(0x1029_02_00, 1)
```

```

[INFO] [swd_diff_drive_controller-1]: process started with pid [13831]
[swd_diff_drive_controller-1] [INFO] [1641790951.043712368] [swd_diff_drive_controller]: publish_odom : 1
[swd_diff_drive_controller-1] [INFO] [1641790951.048934413] [swd_diff_drive_controller]: publish_safety_functions : 1
[swd_diff_drive_controller-1] [INFO] [1641790951.059838757] [swd_diff_drive_controller]: right_config_file : /opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini
[swd_diff_drive_controller-1] [INFO] [1641790951.060834615] [swd_diff_drive_controller]: Initializing right dbus client !
[swd_diff_drive_controller-1] [INFO] [1641790952.197550929] [swd_diff_drive_controller]: left_config_file : /opt/ezw/usr/etc/ezw-smc-core/swd_left_config.ini
[swd_diff_drive_controller-1] [INFO] [1641790952.198280229] [swd_diff_drive_controller]: Initializing left dbus client !
[swd_diff_drive_controller-1] [INFO] [1641790953.309996410] [swd_diff_drive_controller]: fine_odometry : 0
[swd_diff_drive_controller-1] [INFO] [1641790953.353162145] [swd_diff_drive_controller]: left motor polarity : True
[swd_diff_drive_controller-1] [INFO] [1641790953.353281808] [swd_diff_drive_controller]: watchdog_receive_ms : 500
[swd_diff_drive_controller-1] [INFO] [1641790953.353395885] [swd_diff_drive_controller]: pub_freq_hz : 20
[swd_diff_drive_controller-1] [INFO] [1641790953.353437734] [swd_diff_drive_controller]: swd_diff_drive_controller initialized successfully!
[swd_diff_drive_controller-1] [INFO] [1641790953.353902002] [swd_diff_drive_controller]: baseline_m : 0.485000
[swd_diff_drive_controller-1] [INFO] [1641790953.353971017] [swd_diff_drive_controller]: motor_max_speed_rpm : 1050
[swd_diff_drive_controller-1] [INFO] [1641790953.354000635] [swd_diff_drive_controller]: have_backward_sls : 0
[swd_diff_drive_controller-1] [INFO] [1641790953.383761413] [swd_diff_drive_controller]: motor_max_safety_limited_speed_rpm : 490
[swd_diff_drive_controller-1] [WARN] [1641790953.396514946] [swd_diff_drive_controller]: New subscription discovered on topic '/odom', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[swd_diff_drive_controller-1] [INFO] [1641790953.474797960] [swd_diff_drive_controller]: left_encoder_relative_error : 0.200000
[swd_diff_drive_controller-1] [INFO] [1641790953.474917211] [swd_diff_drive_controller]: right_encoder_relative_error : 0.200000
[swd_diff_drive_controller-1] [INFO] [1641790953.474975562] [swd_diff_drive_controller]: odom_frame : odom
[swd_diff_drive_controller-1] [INFO] [1641790953.475003334] [swd_diff_drive_controller]: base_frame : base_link
[swd_diff_drive_controller-1] [INFO] [1641790953.475390426] [swd_diff_drive_controller]: publish_tf : 1
[swd_diff_drive_controller-1] [INFO] [1641790954.362430635] [swd_diff_drive_controller]: NMT state machine is not OK.
[swd_diff_drive_controller-1] [INFO] [1641790954.373920404] [swd_diff_drive_controller]: PDS state machine is not OK.
[swd_diff_drive_controller-1] [INFO] [1641791025.547546702] [swd_diff_drive_controller]: The target speed exceeds the SLS maximum speed limit (490 rpm). Set speed to (left, right) (490, 490) rpm
[swd_diff_drive_controller-1] [INFO] [1641791025.624415833] [swd_diff_drive_controller]: The target speed exceeds the SLS maximum speed limit (490 rpm). Set speed to (left, right) (490, 490) rpm
[swd_diff_drive_controller-1] [INFO] [1641791026.697330480] [swd_diff_drive_controller]: The target speed exceeds the SLS maximum speed limit (490 rpm). Set speed to (left, right) (490, 490) rpm
[swd_diff_drive_controller-1] [INFO] [1641791026.753031245] [swd_diff_drive_controller]: The target speed exceeds the SLS maximum speed limit (490 rpm). Set speed to (left, right) (490, 490) rpm
[swd_diff_drive_controller-1] [INFO] [1641791027.404995358] [swd_diff_drive_controller]: The target speed exceeds the SLS maximum speed limit (490 rpm). Set speed to (left, right) (490, 490) rpm
[swd_diff_drive_controller-1] [INFO] [1641791027.495780950] [swd_diff_drive_controller]: The target speed exceeds the SLS maximum speed limit (490 rpm). Set speed to (left, right) (490, 490) rpm
[swd_diff_drive_controller-1] [INFO] [1641791027.732471547] [swd_diff_drive_controller]: The target speed exceeds the SLS maximum speed limit (490 rpm). Set speed to (left, right) (490, 490) rpm
[swd_diff_drive_controller-1] [INFO] [1641791028.847892674] [swd_diff_drive_controller]: The target speed exceeds the maximum delta speed limit (245 rpm). Speed set to (left, right) (122, -122) rpm
[swd_diff_drive_controller-1] [INFO] [1641791028.923609985] [swd_diff_drive_controller]: The target speed exceeds the maximum delta speed limit (245 rpm). Speed set to (left, right) (122, -122) rpm
[swd_diff_drive_controller-1] [INFO] [1641791029.007948268] [swd_diff_drive_controller]: The target speed exceeds the maximum delta speed limit (245 rpm). Speed set to (left, right) (122, -122) rpm

```

fig-shimooka, 2022-09-13:
```
$ ./remote.py swd_left
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21302SC001@0 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : PREOP  ....................................................................... [OK]
PDS State            : SWITCH_ON_DISABLED  .......................................................... [OK]
NbError              : 254 .......................................................................... [OK]
LastError            : 8201 : SRDO_SCT_timeout ...................................................... [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : True, False, False, False, False, False, False, False  ....................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, False, False, False, False, True, True  .......................... [OK]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 0 ............................................................................ [OK]
OdometryValue        : 0 ............................................................................ [OK]
HallEncoder          : 0 ............................................................................ [KO]
FineOdometryValue    : 0 ............................................................................ [KO]

$ ./remote.py swd_right
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21257SC002@0 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
[... cut ...]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -2 ........................................................................... [OK]
OdometryValue        : -1 ........................................................................... [OK]
HallEncoder          : 0 ............................................................................ [KO]
FineOdometryValue    : 0 ............................................................................ [KO]

```

GMezWheel, 2022-09-13:
```
from smcdbusclient.can_open import CANOpenDBusClient
can_open_client = CANOpenDBusClient(instance_id)

def update_error_behavior():
    error = can_open_client.setValueUInt8(0x1029_02_00, 1)  # error_behavior_syserr_for_error => 1 (no change of the NMT state)
    check("update_error_behavior()", error)
```

GMezWheel, 2022-09-13:
It seems that an emergency button is pressed. First bit of SSW_CAN1 is True
=>` SSW CAN1             : True, False, False, False, False, False, False, False  ....................... [OK]`

Moreover, it seems that SRDO are bad emmitted:
`NbError              : 254 .......................................................................... [OK]`
`LastError            : 8201 : SRDO_SCT_timeout ...................................................... [OK]`

Thus, the NMT state stay in PREOP

GMezWheel, 2022-09-13:
When you press 'm', check that :
- can-id 0x160 is sent by swd_left motor 
``` shell
candump can0 | grep 160
 can0  160   [1]  FF
 can0  160   [1]  FF
 can0  160   [1]  FF
 ```
- can-id 0x109 is sent by swd_right motor 
```
candump can0 | grep 109
  can0  109   [1]  C3
  can0  109   [1]  C3
  can0  109   [1]  C3
```

fig-shimooka, 2022-09-14:
I have checked them all and the beginning of SSW CAN1 remains True.
I predict that the hardware may be the cause.
If SSW CAN1 is True, where should I look at the electrical wiring?

Also, when I set some of the valid to False, I get an error from SE2L. (code:A9 reason:reset input)
I know it is most likely due to the hardware, but is there any possibility of an error in SE2L when I change the settings?

now settings:
[commissioning_0914.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/9563164/commissioning_0914.zip)

$ candump can0 | grep 160
  can0  160   [1]  C3
  can0  160   [1]  C3
  can0  160   [1]  C3

$ candump can0 | grep 109
  can0  109   [1]  C3
  can0  109   [1]  C3
  can0  109   [1]  C3
  can0  109   [1]  C3

fig-shimooka, 2022-09-14:
Enabling all PDOs, etc. did not change the SE2L error.
So I expect that the SE2L is caused by something else.

GMezWheel, 2022-09-14:
According to your commissioning.py, set_no_preop does nothing. => Call this new function instead :
```
from smcdbusclient.can_open import CANOpenDBusClient
can_open_client = CANOpenDBusClient(instance_id)

def update_error_behavior():
    error = can_open_client.setValueUInt8(0x1029_02_00, 1)  # error_behavior_syserr_for_error => 1 (no change of the NMT state)
    check("update_error_behavior()", error)
```

fig-shimooka, 2022-09-15:
sorry!
The zip file was out of date.
I have now implemented the following

[candump.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/9574154/candump.zip)

```
from smcdbusclient.can_open import CANOpenDBusClient

def main(argv):

    instance_id = "swd_left"
    node_id = 0x4
    can_open_client = CANOpenDBusClient(instance_id)

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

    commissioning.update_SLS_parameters(sls_vl_limit, sls_vl_time_monitoring)

    error = can_open_client.setValueUInt8(0x1029_02_00, 1)  # error_behavior_syserr_for_error => 1 (no change of the NMT state)
    commissioning.check("update_error_behavior()", error)

```

fig-shimooka, 2022-09-15:
When the first bit of SSW CAN1 is True, the motor is blinking green (both left and right)

GMezWheel, 2022-09-15:
As the both motors are blinking green, it indicates that no emergency button STO are detected.
Thus, the STO flag in SSW CAN1 bit 0 indicates that an other STO issued of the safety functions is raised.
=> probably the SE2L SDI output in your case

For information, I will try your commissioning today in order to reproduce the problem.

GMezWheel, 2022-09-15:
Hereafter your commissioning files tested on my platform and used to enter left motor in PDS state OPER with the remote tool ('m' then 'o'):
[commissioning.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/9576090/commissioning.zip)

### Thus on left motor

When **no SLS, SDI and emergency** enabled :

NMT State            : OPER
PDS State            : OPERATION_ENABLED
SSW CAN1             : False, False, False, False, False, False, False, False
SCW SAFEIN_1         : True, True, True, True, True, True, True, True

When **SLS** enabled :

NMT State            : OPER
PDS State            : OPERATION_ENABLED
SSW CAN1             : False, False, False, False, False, False, False, False
SCW SAFEIN_1         : True, True, True, True, **False**, **False**, True, True

When **SLS+SDI** enabled :

NMT State            : OPER
PDS State            : OPERATION_ENABLED
SSW CAN1             : **True**, False, False, False, False, False, False, False
SCW SAFEIN_1         : True, True, **False**, **False**, **False**, **False**, True, True

When only **left Emergency** button pressed :

NMT State            : OPER
PDS State            : **FAULT**
SSW CAN1             : **True**, False, False, False, False, False, False, False 
SCW SAFEIN_1         : **False**, **False**, True, True, True, True, True, True

When only **right Emergency** button pressed :

NMT State            : OPER
PDS State            : OPERATION_ENABLED
SSW CAN1             : **True**, False, False, False, False, False, False, False 
SCW SAFEIN_1         : True, True, True, True, True, True, True, True

fig-shimooka, 2022-09-16:
I have executed your commissioning and I see the following.
Also, the LED on the right motor has changed to light up red.

```
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21302SC001@0 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : OPERATION_ENABLED  ........................................................... [OK]
NbError              : 0 ............................................................................ [OK]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : True, False, False, False, False, False, False, False  ....................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, False, False, False, False, True, True  .......................... [OK]
Peripheral           : True, True, True, True, True, True  .......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 0 ............................................................................ [OK]
OdometryValue        : 0 ............................................................................ [OK]
HallEncoder          : 0 ............................................................................ [KO]
FineOdometryValue    : 0 ............................................................................ [KO]

DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21257SC002@0 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
[... cut ...]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 1 ............................................................................ [OK]
OdometryValue        : 0 ............................................................................ [OK]
HallEncoder          : 0 ............................................................................ [KO]
FineOdometryValue    : 0 ............................................................................ [KO]
```

fig-shimooka, 2022-09-16:
When only right emergency is pressed

```
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21257SC002@0 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : FAULT  ....................................................................... [OK]
NbError              : 11 ........................................................................... [OK]
LastError            : 800E : EMERGENCY ............................................................. [OK]
SystemError          : EZW_PROTECT_BRAKE_CC_PROTECT EZW_PROTECT_BRAKE_CC_PROTECT  ................... [OK]
SSW CAN1             : True, False, False, False, False, False, False, False  ....................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : False, False, True, True, True, True, True, True  ............................ [OK]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: False, position_polarity: False ........................... [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -95 .......................................................................... [OK]
OdometryValue        : -91 .......................................................................... [OK]
HallEncoder          : 0 ............................................................................ [KO]
FineOdometryValue    : 0 ............................................................................ [KO]
```

When only left emergency is pressed

```
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21302SC001@0 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
[... cut ...]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -23 .......................................................................... [OK]
OdometryValue        : -22 .......................................................................... [OK]
HallEncoder          : 0 ............................................................................ [KO]
FineOdometryValue    : 0 ............................................................................ [KO]
```

fig-shimooka, 2022-09-16:
I ran the commissioning I got and the
The REST input error in SE2L has disappeared.

GMezWheel, 2022-09-16:
Great !
To sum up, the STO in SSW CAN1 bit 0 is due to SAFEIN_1 functions 3/4 (SDIp->STO):

SSW CAN1             : **True**, False, False, False, False, False, False, False  ....................... [OK]
SCW SAFEIN_1         : True, True, **False**, **False**, False, False, True, True  .......................... [OK]

Indeed, when no emergency is pressed, SAFEIN_1 functions 1/2 (STO) is True (disabled):
SCW SAFEIN_1         : **True**, **True**, False, False, False, False, True, True  .......................... [OK]

Moreover, an SLS is raised (functions 5/6) :
SCW SAFEIN_1         : True, True, False, False, **False**, **False**, True, True  .......................... [OK]

fig-shimooka, 2022-09-16:
STO is currently not responding (as far as I can see on the SE2L display)
Why is the current state of the STO

GMezWheel, 2022-09-16:
If SCW SAFEIN_1 functions 3/4/5/6 are False, it means that SE2L doesn't send +24v on pins 3/4 (from OSSD1/2) and on pins 5/6 (from OSSD 3/4) of the SWD 12points connector.
Can you check again the wiring ?

fig-shimooka, 2022-09-16:
left ? or right?

GMezWheel, 2022-09-16:
Normally, the SE2L is connected to the left motor.

fig-shimooka, 2022-09-16:
The SE2L input was in the right motor, so it was reconnected to the left motor.
However, I think the early swd starter kit had the SE2L input on the right side.

```
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21302SC001@0 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : OPERATION_ENABLED  ........................................................... [OK]
NbError              : 0 ............................................................................ [OK]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : True, False, False, False, False, False, False, False  ....................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, False, False, False, False, True, True  .......................... [OK]
Peripheral           : True, True, True, True, True, True  .......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 0 ............................................................................ [OK]
OdometryValue        : 0 ............................................................................ [OK]
HallEncoder          : 0 ............................................................................ [KO]
FineOdometryValue    : 0 ............................................................................ [KO]
```

```
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21257SC002@0 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
[... cut ...]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -4 ........................................................................... [OK]
OdometryValue        : -3 ........................................................................... [OK]
HallEncoder          : 0 ............................................................................ [KO]
FineOdometryValue    : 0 ............................................................................ [KO]
```

fig-shimooka, 2022-09-16:
When STO and SLS of SE2L are enabled, there is no change in the CAN1 bit.
Is this due to wiring?

```
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21257SC002@0 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : OPERATION_ENABLED  ........................................................... [OK]
NbError              : 0 ............................................................................ [OK]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : True, False, False, False, False, False, False, False  ....................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, False, False, False, False, True, True  .......................... [OK]
Peripheral           : True, True, True, True, True, True  .......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: False, position_polarity: False ........................... [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -4 ........................................................................... [OK]
OdometryValue        : -3 ........................................................................... [OK]
HallEncoder          : 0 ............................................................................ [KO]
FineOdometryValue    : 0 ............................................................................ [KO]

DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21302SC001@0 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
[... cut ...]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 0 ............................................................................ [OK]
OdometryValue        : 0 ............................................................................ [OK]
HallEncoder          : 0 ............................................................................ [KO]
FineOdometryValue    : 0 ............................................................................ [KO]
```

GMezWheel, 2022-09-16:
**_Rigth motor behavior seems OK:_**
=> when there is no STO/SLS from SE2L , SCW SAFEIN_1 is OK : all functions to true
SCW SAFEIN_1         : True, True, True, True, True, True, True, True
=> when there is STO/SLS from SE2L,  SCW SAFEIN_1 is OK : functions 3/4/5/6 to false
SCW SAFEIN_1         : True, True, False, False, False, False, True, True

**_Left motor behavior seems KO:_**
=> when there is no STO/SLS from SE2L , SCW SAFEIN_1 is OK : all functions are not true
SCW SAFEIN_1         : True, True, **False, False, False, False**, True, True
=> when there is STO/SLS from SE2L,  SCW SAFEIN_1 is OK : functions 3/4/5/6 to false
SCW SAFEIN_1         : True, True, False, False, False, False, True, True

STO from SSW CAN1 bit 1 will be to false when SCW SAFEIN_1 will have all its values to True
=> it's not actually the case for left motor
It seems to be a wiring pb between left 12 points connector and SE2L.

**_Left motor wiring_**
SE2L        ->         M12A
OSSD1/Red -> pin 3 
OSSD2/Yellow -> pin 4
OSSD3/Red-black -> pin 5 
OSSD4/Yellow-black -> pin 6

GMezWheel, 2022-09-19:
Hi @fig-shimooka,

I noticed that when the SE2L was connected to the right motor, all the SCW SAFEIN_1 was True
=> it means that all input are well connected to SE2L and that no SDI/SLS are activated : +24v

Thus, if you reconnect SE2L to right motor and change the SRDO commissioning, it will works !

Left :
```
    scwMapping[0] = SafetyFunctionId.STO
    scwMapping[1] = SafetyFunctionId.STO
```
Right :
```
    scwMapping[0] = SafetyFunctionId.STO
    scwMapping[1] = SafetyFunctionId.STO
    scwMapping[2] = SafetyFunctionId.STO
    scwMapping[3] = SafetyFunctionId.STO
    scwMapping[4] = SafetyFunctionId.SLS_1
    scwMapping[5] = SafetyFunctionId.SLS_1
```

fig-shimooka, 2022-09-20:
I connected the SE2L signal to the right motor and tested again.
The result didn't change.

```

DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21302SC001@0 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT State            : PREOP  ....................................................................... [OK]
PDS State            : NOT_READY_TO_SWITCH_ON  ...................................................... [OK]
NbError              : 0 ............................................................................ [OK]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ..............0........ [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False 0 ...................... [OK]
SCW SAFEIN_1         : True, True, True, True, True, True, True, True  .............................. [OK]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 0 ............................................................................ [OK]
OdometryValue        : 0 ............................................................................ [OK]
HallEncoder          : 0 ............................................................................ [KO]
FineOdometryValue    : 0 ............................................................................ [KO]

DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21257SC002@0 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
[... cut ...]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -2 ........................................................................... [OK]
OdometryValue        : -1 ........................................................................... [OK]
HallEncoder          : 0 ............................................................................ [KO]
FineOdometryValue    : 0 ............................................................................ [KO]

```

GMezWheel, 2022-09-20:
As SAFEIN_1 is True, True, True, True, True, True, True, True on left motor, I think that the SE2L is connected to the left motor (not to the right motor).
Thus, in this case, you have to commission motors with the commissioning i sent.

**Left:**
TX

    scwMapping[0] = SafetyFunctionId.STO
    scwMapping[1] = SafetyFunctionId.STO
    scwMapping[2] = SafetyFunctionId.STO
    scwMapping[3] = SafetyFunctionId.STO
    scwMapping[4] = SafetyFunctionId.SLS_1
    scwMapping[5] = SafetyFunctionId.SLS_1
RX

    scwMapping[0] = SafetyFunctionId.STO
    scwMapping[1] = SafetyFunctionId.STO

**Right :**
TX

    scwMapping[0] = SafetyFunctionId.STO
    scwMapping[1] = SafetyFunctionId.STO
RX

    scwMapping[0] = SafetyFunctionId.STO
    scwMapping[1] = SafetyFunctionId.STO
    scwMapping[2] = SafetyFunctionId.STO
    scwMapping[3] = SafetyFunctionId.STO
    scwMapping[4] = SafetyFunctionId.SLS_1
    scwMapping[5] = SafetyFunctionId.SLS_1

fig-shimooka, 2022-09-21:
The motor now runs with swd-ros2-controller!
The cause was the following two points.
1. improper Commissioning setting.  -> https://github.com/ezWheelSAS/swd_ros_controllers/issues/27#issuecomment-1250829923
2. CANOpenIO was connected in the opposite direction. ->https://github.com/ezWheelSAS/swd_ros_controllers/issues/27#issuecomment-1252229546

I used the same wiring as when I received the SWD-Starterkit.
By updating swd-service this time, it was necessary to connect the motor to the opposite side.
Has the CANOpenIO connection been reversed with the recent update?

fig-shimooka, 2022-09-22:
Is it possible to swap the CANOpenIO connection between right and left in the Commissioning settings?
I have tried various ways to get the motor to work this time.
I am assuming that the CANOpenIO connections were reversed in the process.

GMezWheel, 2022-09-22:
> This issue has been resolved. the motor now runs with swd-ros2-controller! The cause was the following two points.
> 
> 1. improper Commissioning setting.  -> [can't commission #27 (comment)](https://github.com/ezWheelSAS/swd_ros_controllers/issues/27#issuecomment-1250829923)
> 2. CANOpenIO was connected in the opposite direction. ->[can't commission #27 (comment)](https://github.com/ezWheelSAS/swd_ros_controllers/issues/27#issuecomment-1252229546)
> 
> I used the same wiring as when I received the SWD-Starterkit. By updating swd-service this time, it was necessary to connect the CANOpenIO to the opposite side. Has the CANOpenIO connection been reversed with the recent update?

The new release of swd-services doesn't impact CANOpenIO connection. 
I think that the misunderstanding is about the definition of the left motor. 
The left motor is at the left side when looking the robot from behind (in the driving direction).

GMezWheel, 2022-09-22:
> Is it possible to swap the CANOpenIO connection between right and left in the Commissioning settings? I have tried various ways to get the motor to work this time. I am assuming that the CANOpenIO connections settings were reversed in the process.
> 
> I have not changed the hardware connections, but depending on the settings, the CANOpenIO connections are swapped between left and right. Is this possible to happen

Indeed, the commissioning is not the same for the both motors. Thus be carefull to not invert it.

[image]

fig-shimooka, 2022-09-22:
Is it correct to say that the left motor and the right motor are strictly defined?

Is there a way to identify the left motor from the right motor?
They look the same on the outside.

ez-Support, 2022-09-22:
The hardware is strictly the same, however the commissioning is different has they are monted top to tail. Thus, to move forward, one is turning in one direction, and the other in the opposite way. Moreover, only one motor is connected to the laser scanner, so it has to be defined in the commissioning which motor receives the signals from the laser.

GMezWheel, 2022-09-22:
I agree with you. 
The left commissioning corresponds in fact to the motor which receives the signals from the laser.

[32 more replies]

closed 2022-10-14 by GMezWheel