IDEC-ezWheel/swd_ros_controllers #28: CAN ID is not written with Commissioning of ROS1

suguruhamada1003910, 2022-07-26:
When I write command as below, I can see CAN ID with 710.

```
swd_sk@C-Latitude-7280:~$ candump can0 | grep 7
  can0  710   [1]  7F
  can0  701   [1]  05
  can0  710   [1]  7F
  can0  701   [1]  05
  can0  710   [1]  7F
  can0  701   [1]  05
  can0  710   [1]  7F
  can0  701   [1]  05
  can0  710   [1]  7F
  can0  701   [1]  05
```
Therefore, I think I need to do commissioning. However, CAN ID is not changed after commissioning.
Is the following result of commissioning is success or fail?

```
swd_sk@C-Latitude-7280:/opt/ezw/commissioning$ python3 swd_left_4_commissioning.py 
SYSTEMCTL_dbus.id detected
create_dbus_clients(swd_left) : OK
SRDOParameters(valid=True, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=287, can_id2=288)
SRDOParameters(valid=True, sct=25, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=259, can_id2=260)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=261, can_id2=262)
SRDOParameters(valid=False, sct=25, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=263, can_id2=264)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=265, can_id2=266)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=267, can_id2=268)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=269, can_id2=270)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=271, can_id2=272)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=273, can_id2=274)
SRDOParameters(valid=False, sct=50, srvt=20, transmission_type=<SRDOTransmissionType.PDO_ASYNC_FE: 254>, can_id1=275, can_id2=276)

```

Files of commissioning I'm using is below.
[commissioning.zip](https://github.com/ezWheelSAS/swd_ros_controllers/files/9186425/commissioning.zip)

ez-Support, 2022-07-26:
See https://github.com/ezWheelSAS/swd_ros_controllers/issues/27#issuecomment-1192225799

suguruhamada1003910, 2022-07-26:
> See [#27 (comment)](https://github.com/ezWheelSAS/swd_ros_controllers/issues/27#issuecomment-1192225799)

Thank you for reply.
I already checked [#27 ](https://github.com/ezWheelSAS/swd_ros_controllers/issues/27#issuecomment-1192225799) and do that.
However, CAN ID is not changed.
When commissioning succeeds, what is displayed?

ez-Support, 2022-07-26:
I don't understand why line 111 you have `return 0`, before all updates. Thus, your are not doing any commissioning.

ez-Support, 2022-07-26:
It displays `OK`, after each successfull step

suguruhamada1003910, 2022-07-26:
> It displays `OK`, after each successfull step

Thank you for reply!
I changed files to [v1.0.0](https://github.com/ezWheelSAS/swd-starter-kit-config/tree/v1.0.0/commissioning ) and commissioning was success! ("Commissioning succeeded !" is displayed.)

Please close this.

closed 2022-07-26 by suguruhamada1003910