IDEC-ezWheel/swd_ros_controllers #30: Can I get commissioning files for Starter Kit?

suguruhamada1003910, 2022-08-01:
I did commissioning with [v1.0.0](https://github.com/ezWheelSAS/swd-starter-kit-config/tree/v1.0.0/commissioning) because node ID was initialized as 0x10.
Then, node ID is modified to 0x04 and 0x05 but the behavior is strange now.

SLS and STO is not detected properly. Specifically, when I execute starter-kit.launch, firstly, SLS is 0. Therefore, SWD Core is not running. After that, when I hold my hand in front of the SE2L, SLS becomes 1 and SWD Core is running.
And a few minutes later, STO repeats 0 and 1.

I do not know whether this problem depends commissioning or not. However, this problem occurs after commissioning.

Can I get commissioning files for Starter Kit?

ez-Support, 2022-08-01:
You can find default commissioning files in this repository:
https://github.com/ezWheelSAS/swd-starter-kit-config/tree/main/commissioning

suguruhamada1003910, 2022-08-02:
I did commissioning with it. Result is below.

```
swd_sk@C-Latitude-7280:~/ros_ws/src/swd-starter-kit-config/commissioning$ python3 swd_left_4_commissioning.py 
Traceback (most recent call last):
  File "swd_left_4_commissioning.py", line 7, in <module>
    import commissioning
  File "/home/swd_sk/ros_ws/src/swd-starter-kit-config/commissioning/commissioning.py", line 22, in <module>
    from smcdbusclient.can_open import CANOpenDBusClient
ModuleNotFoundError: No module named 'smcdbusclient.can_open'

```

There are some *.py files in /opt/ezw/usr/lib/smcdbusclient but can_open.py does not exist.
I am using swd-services v0.1.2 because I am using ROS1. [Explanation of swd-services version](https://github.com/ezWheelSAS/swd_ros_controllers/issues/25#issuecomment-1185272762)

Is it causing problem?

ez-Support, 2022-08-02:
You can use V1.0.0 version of commissioning in your case.

Otherwise, using ROS1 you can upgrade swd-services to 0.2.x and ROS controller to version > 3.0.0 (https://github.com/ezWheelSAS/swd_ros_controllers/releases)

suguruhamada1003910, 2022-08-03:
>You can use V1.0.0 version of commissioning in your case.
OK. I understand that files I used have no problem.

>Otherwise, using ROS1 you can upgrade swd-services to 0.2.x and ROS controller to version > 3.0.0 (https://github.com/ezWheelSAS/swd_ros_controllers/releases)

I have two questions.
1. In v3.2.0, "Sync with ROS2" is written as explanation of source code. What is the meaning? I thought version greater than v2.0.4 is created for ROS2 and version greater than v3.0.0 cannot be work on ROS1.
2. What is difference between the latest commissioning files and v1.0.0?

ez-Support, 2022-08-03:
1. ros1 version (>=3.0.0) are still compliant with ROS1. The update of the controller v3.2.0 enables to use ROS1 and ROS2

1. The previous commissioning file use the old swd-service version (0.1.x). The new version of the commissioning use the new version of swd-service (0.2.x)

suguruhamada1003910, 2022-08-04:
I understand. Thank you so much.

suguruhamada1003910, 2022-09-28:
> 1. ros1 version (>=3.0.0) are still compliant with ROS1. The update of the controller v3.2.0 enables to use ROS1 and ROS2

Does it mean that this source code can be built with catkin comand and colcon command. Is it write?
Then, what is the difference between "https://github.com/ezWheelSAS/swd_ros_controllers" and "https://github.com/ezWheelSAS/swd_ros2_controllers"?

GMezWheel, 2022-09-28:
Hi @suguruhamada1003910 ,

The answer is no. 

The both version of the controllers do the same thing. But as the ROS1 and ROS2 C++ API are différents, we have to maintain two source code separately.

For now, ROS1 3.3.0 controller is equals in term of capability to ROS2 1.4.1 controller.
* catkin command to build ROS1 controller
* colcon command to build ROS2 controller

suguruhamada1003910, 2022-10-03:
@GMezWheel 

Thank you for reply.
In my understanding, if I communicate between "https://github.com/ezWheelSAS/swd_ros_controllers" and another ROS2 node, I needs [ros1_bridge](https://github.com/ros2/ros1_bridge). It is right?
If it is wrong, how can I use "https://github.com/ezWheelSAS/swd_ros_controllers" in ROS2 system?

GMezWheel, 2022-10-03:
You're right, you have to use ros1_bridge in order to communicate whith ROS2 system.

closed 2022-10-03 by suguruhamada1003910