IDEC-ezWheel/swd_ros_controllers #31: How can I check the SWD firmware version?

suguruhamada1003910, 2022-08-04:
[image]

I would like to use the latest source code. Then, I noticed that firmware version must be greater than v1.0.1.
How can I check the version?

ez-Support, 2022-08-04:
You can have access to this information through the remote diagnostic, it corresponds to the line "SWVersion" :
`~/remote.py swd_right
~/remote.py swd_left`

suguruhamada1003910, 2022-08-04:
I got it. Then, SWD version is V1.0.1. Should I upgrade firmware version? "SWD firmware (> 1.0.1)" means it must be 1.0.2 or bigger?

ez-Support, 2022-08-04:
I think '=' is missing. SWD firmware version V1.0.1 should be fine 😉!

suguruhamada1003910, 2022-08-05:
Thanks. I created pull request.

closed 2022-08-05 by suguruhamada1003910