IDEC-ezWheel/swd_ros_controllers #34: where is the code on the starterkit computer ?

JacquesRoubaud, 2022-09-02:
Hi
I have received my starterkit last week.
I am able to connect to the robot PC via vscode using an additional wifi key
When I am in the home directory, I don't see the code that is used and launched by the robot. 
I could locate some code in opt/ezw, but if I make a small change in the index.html file in /opt/ezw/html,  I don't see it reflected on the web page when I type 10.10.0.1 in Chrome. So the "real" code must be somewhere else ? where ? 
I would expect to see some code to compile with catkin_make, some urdf, some python code, ...

can you help ? thank you
Jacques

ez-Support, 2022-09-02:
Oui c'est volvo de Venissieux

Ils ont pris 2 starterkit dans le but de créer leur propre  AGV.

JacquesRoubaud, 2022-09-08:
thank you for this, I am making some progress!

can you tell me what service I should kill and restart to update the webpage (help with the right commands is appreciated). I guess there is a better option than rebooting the computer every time :)

thanks!

GMezWheel, 2022-09-08:
Hi,

There is no service to restart.
You only have to force a refresh of your page into your broswer.
ie : Ctrl+F5 into Chrome

JacquesRoubaud, 2022-09-09:
I still have some difficulties to find everything. For instance, where is the code / node that publishes the laser data ? where is the SWD code that is **actually** run on the robot, eg the code for differential drive.

closed 2022-09-09 by JacquesRoubaud