IDEC-ezWheel/swd_ros_controllers #36: CAn't access webcam from my windows PC using SSH

JacquesRoubaud, 2022-09-08:
Hi
I am connected to the starterkit via SSH
I want to to run a simple opencv program to access a camera with cap = cv2.VideoCapture(0)
I have run all possible indexes. This is what ls /dev/vid* gives:
swd_sk@SWDSK880937:~$ ls /dev/vide*
/dev/video0  /dev/video1  /dev/video16  /dev/video17  /dev/video2

I have this error message 
 [ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1758) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Could not open device '/dev/video2' for reading and writing.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (887) open VIDEOIO(V4L2:/dev/video2): can't open camera by index
[]

is there something missing in my configuration ? 
Many thanks

GMezWheel, 2022-09-08:
It seems to be a permission problem.
Try this :
`sudo chmod 777 /dev/video0 and restart the camera `

JacquesRoubaud, 2022-09-09:
it works
thanks!

closed 2022-09-09 by JacquesRoubaud