IDEC-ezWheel/swd_ros_controllers #38: Unable to locate linux-kernel

JacquesRoubaud, 2022-09-21:
Hi
I am trying to use a second wifi dongle (TP link Archer T3U AC1300), and the driver is not recognized.
I am following this procedure : https://community.tp-link.com/en/home/forum/topic/208022
git clone https://github.com/cilynx/rtl88x2bu.git
cd rtl88x2bu
VER=$(sed -n 's/\PACKAGE_VERSION="\(.*\)"/\1/p' dkms.conf)
sudo rsync -rvhP ./ /usr/src/rtl88x2bu-${VER}
sudo dkms add -m rtl88x2bu -v ${VER}
sudo dkms build -m rtl88x2bu -v ${VER}
sudo dkms install -m rtl88x2bu -v ${VER}
sudo modprobe 88x2bu

Note that VER = 5.8.7.1

when I run sudo dkms build -m rtl88x2bu -v ${VER}, I have an error message that says Error! your kernel headers for kernel 5.4.70-g78d1314be027-dirty cannot be found. Please install the linux-headers-5.4.70-g78d1314be027-dirty package
I have spent a few hours googling possible solutions, but every time, in a way or another , I am blocked by the fact that the linux header cannot be found.

Is there anything special in the linux installation of the PC ? Any clue ? 

thank you
Jacques

GMezWheel, 2022-09-22:
Hi @JacquesRoubaud,

You're right. In order to install a new driver, you need to compile it and have linux headers installed.
I confirm that the package linux-headers-5.4.70-g78d1314be027-dirty is not available.

I will check with the manufacturer (TechNexion) if the package exists or if a kernel upgrade is necessary.

GMezWheel, 2022-09-22:
Acutually, only the 8192eu driver is available.
```
$ dmesg
...
[ 8783.107978] usb 1-1.5: new high-speed USB device number 6 using ci_hdrc
[ 8783.259141] usb 1-1.5: New USB device found, idVendor=2357, idProduct=0109, bcdDevice= 2.00
[ 8783.259152] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8783.259159] usb 1-1.5: Product: 802.11n NIC 
[ 8783.259164] usb 1-1.5: Manufacturer: Realtek 
[ 8783.259169] usb 1-1.5: SerialNumber: 00e04c000001
[ 8783.606821] usbcore: registered new interface driver rtl8192eu
```
```
$ lsmod
Module                  Size  Used by
8192eu               1396736  0
mxc_v4l2_capture       49152  0
ipu_bg_overlay_sdc     16384  1 mxc_v4l2_capture
ipu_still              16384  1 mxc_v4l2_capture
v4l2_int_device        16384  1 mxc_v4l2_capture
ipu_prp_enc            16384  1 mxc_v4l2_capture
ipu_csi_enc            16384  1 mxc_v4l2_capture
ipu_fg_overlay_sdc     16384  1 mxc_v4l2_capture
imx_vdoa               16384  0
imx_sdma               36864  2
binfmt_misc            20480  1
```

closed 2022-11-02 by JacquesRoubaud