IDEC-ezWheel/swd_ros_controllers #39: How can I measure the distance SWD has traveled?

suguruhamada1003910, 2022-09-27:
I would like to create program which SWD repeats forward and backward. Then, I would like to specify the distance of forward and backward.
I thought I can calculate the distance from the elapsed time and the speed but SWD increments speed gradually. Therefore, I thought I need the value of encoder.

Is it possible that I get the value of encoder? And if there is easy way to calculate the distance, please tell me.

GMezWheel, 2022-09-27:
Hi @suguruhamada1003910 ,

Have a look at cbTimerOdom function of the DiffDriveController ROS node. 
This function gets the odometry encoder value of each motors and compute new distance/theta in order to send odom topic.

suguruhamada1003910, 2022-09-27:
Hi @GMezWheel 

I think `d_dist_left` and `d_dist_left` are the distance (m). Is it right?

GMezWheel, 2022-09-27:
Yes. 
* _mm means millimeters
* _m means meters

Thus d_dist_left_m et d_dist_right_m are the distance in meter.

closed 2022-09-28 by suguruhamada1003910