IDEC-ezWheel/swd_ros_controllers #41: Distances of forward and backward are different.

suguruhamada1003910, 2022-10-03:
I would like to run SWD to forward and backward repeatedly. Thefore, I created 3 seconds timer and send "cmd_vel_msg.linear.x = 1" and "cmd_vel_msg.linear.x = -1" to swd_ros_controller node repeatedly from teleop_twist_joy node.

Then, the number of messages for forward and backward is same but distances of forwrd and back ward are different at the first repeat only. (The distance of forward is longer than backward.)

Do you know any problem in swd_ros_controller?

GMezWheel, 2022-10-03:
Hi @suguruhamada1003910,

By default, velocity in backward direction is limited to max SLS speed => ros parameter 'have_backward_sls' is false by default
You have to set this parameter to True if you want to disable this limitation.

Moreover, have you deployed the latest version ? Indeed, a latency problem due to safety message computation has been resolved. 

New packages :
- debian package swd-services 0.2.7 
- swd_ros_controllers 3.3.0 : use new TPDO1 with SAFEIN_1 data
- swd-starter-kit-config 2.1.0 : new commissioning of the both SWD is needed (active TPDO1 with SAFEIN_1 data into SafetyHub12pts commissioning)

suguruhamada1003910, 2022-10-04:
@GMezWheel 

Thank you for reply.

>By default, velocity in backward direction is limited to max SLS speed => ros parameter 'have_backward_sls' is false by default
>You have to set this parameter to True if you want to disable this limitation.
I already changed it to "True". Therefore, the distance is same at the second repeat or later. This problem is occurred at the first repeat only.

>Moreover, have you deployed the latest version ? Indeed, a latency problem due to safety message computation has been resolved.
I am using v2.0.0. I try to update version.

suguruhamada1003910, 2022-10-04:
@GMezWheel 

>debian package swd-services 0.2.7
>swd_ros_controllers 3.3.0 : use new TPDO1 with SAFEIN_1 data
>swd-starter-kit-config 2.1.0 : new commissioning of the both SWD is needed (active TPDO1 with SAFEIN_1 data into SafetyHub12pts commissioning)

I upgraded version of data as you said. As a result, the behavior is improved. The difference of the distance becomes small. However, there is a small difference. The distance of forward is greater than backward. SWD runs forward for about 4 seconds and backward for about 3 seconds. Is there a way to improve more?

suguruhamada1003910, 2022-10-05:
I checked the number of received twist data from teleop_twist_joy node for forward and backward is same.

GMezWheel, 2022-10-05:
Hi @suguruhamada1003910,

I will try to reproduce your issue today.

GMezWheel, 2022-10-05:
I think I have reproduced your problem.

[image]

Indeed, the first cycle start from a null velocity value while the other cycles start from a min/max velocity value.

In order to control your move, I suggest you to :
* wait for reaching the null velocity before requesting the new one 
=> linear.x 1m/s -> 0m/s -> -1m/s -> 0m/s ...
* take into account the slopes
=> for example, about 1 more second to continue to go forward while a negative velocity is requested

suguruhamada1003910, 2022-10-06:
@GMezWheel 

Thank you for idea!! I added sleep for 2 seconds when changing direction, problem is resovled!!

I would like to know the method to create the graph of velocity. Is it possible?

GMezWheel, 2022-10-07:
1) At the end of DiffDriveController::setSpeeds() method, add the following lines:
 ```
       int32_t _left_speed, _right_speed;

        err = m_left_controller.getVelocityActualValue(_left_speed);
        if (ERROR_NONE != err) {
            RCLCPP_ERROR(get_logger(),
                         "Failed get velocity of left motor, EZW_ERR: SMCService : "
                         "Controller::getTargetVelocity() return error code : %d",
                         (int)err);
            return;
        }

        err = m_right_controller.getVelocityActualValue(_right_speed);
        if (ERROR_NONE != err) {
            RCLCPP_ERROR(get_logger(),
                         "Failed get velocity of right motor, EZW_ERR: SMCService : "
                         "Controller::getTargetVelocity() return error code : %d",
                         (int)err);
            return;
        }

        RCLCPP_INFO(get_logger(), "monitor_velocity;%d;%d;%d;%d", p_left_speed, p_right_speed, _left_speed, _right_speed);
```
2) Then launch ros node, grep instrum_velocity tag and save result into a .csv file.
`ros2 launch swd_ros2_controllers swd_diff_drive_controller.launch.py motor_max_speed_rpm:=2000 | grep monitor_velocity > log.csv`
3) Use Excel to draw graph

suguruhamada1003910, 2022-10-14:
Thank you so much!

closed 2022-10-14 by suguruhamada1003910