IDEC-ezWheel/swd_ros_controllers #42: Ajusting speed of acceleration and deceleration

suguruhamada1003910, 2022-10-04:
[Adjusting the PID gain of the SWD Core. #8 ](https://github.com/ezWheelSAS/swd_ros_controllers/issues/8#issuecomment-1077475302)

As written here, I think we can change speed of acceleration and deceleration with commissioning. However, I do not know the meaning of the value of "vl_velocity_deceleration_delta_speed" and "vl_velocity_acceleration_delta_speed". What is the meaning of 1500?

GMezWheel, 2022-10-04:
The documentation is provided into include files of ezw-smc-core library.
=>IVelocityModeService.hpp

```
                uint32_t vl_velocity_acceleration_delta_speed = uint32_t{}; /*!< The configured delta speed of the slope of the acceleration ramp. The values shall be given in rotations per minute (r/min) or in user-defined velocity units. Default : 500 r/min. */
                uint16_t vl_velocity_acceleration_delta_time  = uint16_t{}; /*!< The configured delta time of the slope of the acceleration ramp. The values shall be given in second or in user-defined time units. Default : 1s. */
                uint32_t vl_velocity_deceleration_delta_speed = uint32_t{}; /*!< The configured delta speed of the slope of the deceleration ramp. The values shall be given in rotations per minute (r/min) or in user-defined velocity units. Default : 500 r/min. */
                uint16_t vl_velocity_deceleration_delta_time  = uint16_t{}; /*!< The configured delta time of the slope of the deceleration ramp. The values shall be given in second or in user-defined time units. Default : 1s. */
```

Thus, in your case, 1500rpm means that the ramp acc/dec is 1500 rpm/s

suguruhamada1003910, 2022-10-05:
Thanks a lot!!

closed 2023-08-29 by suguruhamada1003910