IDEC-ezWheel/swd_ros_controllers #43: ezw-canopen.service is not started

suguruhamada1003910, 2022-10-17:
```
komatsu@komatsu-Latitude-7280:/opt/ezw/usr/etc/ezw-smc-core$ systemctl -a | grep ezw
  ezw-canopen.service                                                                      loaded    activating auto-restart EZW CANOPEN SERVICE                                                                
  ezw-dbus-user-session.service                                                            loaded    active     running      EZW DBUS USER SESSION SERVICE                                                      
  ezw-ros-bringup.service                                                                  loaded    active     running      EZW ROS BRINGUP SERVICE                                                            
  ezw-stack.service                                                                        loaded    active     running      EZW APP STARTUP - PARENT SERVICE                                                   
  ezw-swd-left.service                                                                     loaded    active     running      EZW SWD LEFT SERVICE                                                               
  ezw-swd-right.service                                                                    loaded    active     running      EZW SWD RIGHT SERVICE      
```
```

komatsu@komatsu-Latitude-7280:/opt/ezw/usr/etc/ezw-smc-core$ systemctl -a | grep ezw
  ezw-canopen.service                                                                      loaded    activating auto-restart EZW CANOPEN SERVICE                                                                
  ezw-dbus-user-session.service                                                            loaded    active     running      EZW DBUS USER SESSION SERVICE                                                      
  ezw-ros-bringup.service                                                                  loaded    active     running      EZW ROS BRINGUP SERVICE                                                            
  ezw-stack.service                                                                        loaded    active     running      EZW APP STARTUP - PARENT SERVICE                                                   
  ezw-swd-left.service                                                                     loaded    active     running      EZW SWD LEFT SERVICE                                                               
  ezw-swd-right.service                                                                    loaded    active     running      EZW SWD RIGHT SERVICE      
```

I am using swd_ros_controller V3.3.0. I cannot start ezw-canopen.service.
LED of SWD Core is flashing and color is green. Emergency switch is OFF.
And remote.py fails.

How should I do?

GMezWheel, 2022-10-17:
Hi @suguruhamada1003910,

For information, swd_ros_controller V3.3.0 needs ezw-smc-service (>= 0.2.7).

Moreover, in the new version of ezw-smc-service (> 0.2.x), ezw-canopen-service is no more used.

GMezWheel, 2022-10-18:
Hi @suguruhamada1003910,

Hereafter the procedure to update your Starter Kit with the latest ez-Wheel packages :
-	swd-services -> 0.2.7
-	ros-noetic-swd-ros-controllers -> 3.3.0
-	swd_starter_kit -> 0.1.5

Either by updating all the system :
```
sudo apt update
sudo apt upgrade
```

Or by updating only the ez-Wheel packages:
```
sudo apt update
sudo apt install swd-services ros-noetic-swd-ros-controllers swd_starter_kit
```

suguruhamada1003910, 2022-10-18:
> Hi @suguruhamada1003910,
> 
> For information, swd_ros_controller V3.3.0 needs ezw-smc-service (>= 0.2.7).
> 
> Moreover, in the new version of ezw-smc-service (> 0.2.x), ezw-canopen-service is no more used.

@GMezWheel 

How can I get ezw-smc-service?

suguruhamada1003910, 2022-10-18:
I already did the procedure in URL you are sent except commissioning. Therefore, I would like to do commissioning but remote.py is failed and I cannot do commissining.

GMezWheel, 2022-10-18:
You have also installed swd_starter_kit ?
https://github.com/ezWheelSAS/swd_ros_controllers/issues/43#issuecomment-1281881036

suguruhamada1003910, 2022-10-18:
I tried install of swd_starter_kit but the package was not found. I think it is prepared for armhf only. Is it right?
My PC is x86_x64.

GMezWheel, 2022-10-18:
It is right.
In this case you must installed a new package instead : **swd-system-config-2wheels**
[https://github.com/ezWheelSAS/swd_ros_controllers#usage-on-your-own-ipc](https://github.com/ezWheelSAS/swd_ros_controllers#usage-on-your-own-ipc)

GMezWheel, 2022-10-18:
For information, commissioning files have changed:
=> `/opt/ezw/commissioning/SafetyHub12pts`

Ensure you restart with this new files for your commissioning in order to have STO/SLS/SDI detection improve into swd_ros_controllers.

Indeed TPDO1 has been activated with safety_controlword_safein_1 into commissioning.py
```
    # Set the COB ID of the TPDO_1
    params.cob_id.valid = True
    params.cob_id.can_id = 0x180 + node_id  # 0x180 + $NODE_ID
```
...
 ```
    # TPDO1 communication mapping

    # - safety_controlword_safein_1

    mapping.nb = 1
    mapping.items.clear()
    mapping.items.append(0x2620_02_08)

    communication_client.setTPDOMappingParameters(PDOId.PDO_1, mapping)
    check("setTPDOMappingParameters(PDOId.PDO_1)", error)
 ```

suguruhamada1003910, 2022-10-18:
@GMezWheel 

Hi. I am checking my PC and I noticed that ros-noetic-swd-ros-controllers is not installed.
Error message is below.

```
ros-noetic-swd-ros-controllers : Depends: swd-service (>= 0.2.7) but it is not installable
E: Unable to correct problems, you have held broken packages
```

I know `swd-services` but I do not know `swd-service`. How can I get it?

GMezWheel, 2022-10-18:
Oups ! 
I think it is a mistake. swd-service doesn't exist.
Try to force installation. 
I have to fix the issue and rebuild ros-noetic-swd-ros-controllers.

GMezWheel, 2022-10-18:
For information, the package ros-noetic-swd-ros-controllers_3.3.0-0focal_amd64.deb has been rebuilding.

suguruhamada1003910, 2022-10-18:
> For information, the package ros-noetic-swd-ros-controllers_3.3.0-0focal_amd64.deb has been rebuilding.
Thank you for support! I could install it.

> It is right. In this case you must installed a new package instead : **swd-system-config-2wheels** https://github.com/ezWheelSAS/swd_ros_controllers#usage-on-your-own-ipc

I installed the package but I cannot communicate with SWD...
remote.py fails and candump is below.

[image]

Can you have any advice?

suguruhamada1003910, 2022-10-18:
Yes. I know it but I do not know the meaning of 707 and 706 and how should I do.

GMezWheel, 2022-10-18:
The old ezw-canopen-service (nodeId 0x701) has been included into ezw-smc-service :
* 0x706 is the left ezw-smc-service 
* 0x707 is the right ezw-smc-service

GMezWheel, 2022-10-18:
If you pressed an Emergency button and no LEDS turn to red on the SWD drive, it means that the firmware drive is not started. Thus it doesn't send HB on CANopen bus.

suguruhamada1003910, 2022-10-18:
@GMezWheel 

I pressed an Emergency button. Then, LEDs are green. I think it is strange...
I may be hardware problem such as wiring or etc...

GMezWheel, 2022-10-18:
You will receive by mail a procedure to update the firmware in the both SWD drives.
I think you have the pre-released version which has this behavior with new ezw-smc-service package.

suguruhamada1003910, 2022-10-20:
In V3.3.0, ezw-canopen.service is not needed. This issue is closed.

closed 2022-10-20 by suguruhamada1003910