IDEC-ezWheel/swd_ros_controllers #47: LEDs of SWD is flashing when roslaunch is executed

suguruhamada1003910, 2022-10-27:
It looks like roslaunch is success but SWD is not running.
[roslaunch Log.docx](https://github.com/ezWheelSAS/swd_ros_controllers/files/9876551/roslaunch.Log.docx)

LEDs of SWD is flashing. (When SWD is connected with PC properly, LEDs is turned ON with green.)

```
komatsu@komatsu-Latitude-7280:~$ python3 /opt/ezw/usr/sbin/remote.py swd_left
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21302SC003A1 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT state on error   : 0 ............................................................................ [OK]
NMT State            : PREOP  ....................................................................... [OK]
PDS State            : NOT_READY_TO_SWITCH_ON  ...................................................... [OK]
NbError              : 1 ............................................................................ [OK]
LastError            : 6020 : EZW_SYSTEM_ERROR_CODE_CANOPEN_PARAMETER_ERROR ......................... [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, True, True, True, True, True, True  .............................. [OK]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 1 ............................................................................ [OK]
OdometryValue        : 0 ............................................................................ [OK]
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
```

```
komatsu@komatsu-Latitude-7280:~$ python3 /opt/ezw/usr/sbin/remote.py swd_right
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21302SC006A1 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
[... 13 lines cut ...]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, False, False, False, False, True, True  .......................... [OK]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: False, position_polarity: False ........................... [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -4 ........................................................................... [OK]
OdometryValue        : -3 ........................................................................... [OK]
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
```

GMezWheel, 2022-10-27:
An error has occured during SWD commissioning :
=> LastError            : 6020 : **EZW_SYSTEM_ERROR_CODE_CANOPEN_PARAMETER_ERROR** ......................... [OK]

1) Restart IPC : OFF/ON
2) Start ezw services
3) Stop ROS bringup (swd diff drive controllers node at least)
3) Applied 12pts commissioning in both drives
4) Launch remote.py to check that error has disappeared

suguruhamada1003910, 2022-10-27:
@GMezWheel 

Thank you for prompt reply. I will try to do commissioning again!

suguruhamada1003910, 2022-10-28:
@GMezWheel 

I tried commissioning but LEDs are still flashing...

```
komatsu@komatsu-Latitude-7280:~$ python3 /opt/ezw/usr/sbin/remote.py swd_right
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21302SC006A1 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT state on error   : 1 ............................................................................ [OK]
NMT State            : PREOP  ....................................................................... [OK]
PDS State            : NOT_READY_TO_SWITCH_ON  ...................................................... [OK]
NbError              : 0 ............................................................................ [OK]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, False, False, False, False, True, True  .......................... [OK]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: False, position_polarity: False ........................... [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -5 ........................................................................... [OK]
OdometryValue        : -4 ........................................................................... [OK]
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
```

```
komatsu@komatsu-Latitude-7280:~$ python3 /opt/ezw/usr/sbin/remote.py swd_left
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21302SC003A1 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
[... cut ...]
[ INFO] [1666933735.859519387]: NMT state machine is not OK.
[ INFO] [1666933735.870216891]: PDS state machine is not OK.
^[[B^[[B^C[robot_manager-10] killing on exit
[teleop_twist_joy-9] killing on exit
[map_odom_broadcaster-5] killing on exit
[tf2_web_republisher-7] killing on exit
[xbox_joystick-8] killing on exit
[urg_node-2] killing on exit
[hector_mapping-4] killing on exit
[rosbridge_websocket-6] killing on exit
[laser_broadcaster-3] killing on exit
[swd_diff_drive_controller-1] killing on exit
```

GMezWheel, 2022-10-28:
As SDIp/SDIn are enabled into ros log and safety signals to True into left SAFEIN_1, it seems that the current commissioning is 17pts and not 12pts.
Make sure that commissioning is 12pts.

Moreover, it seems that swd_diff_drive_controller is unable to change NMT state to OPER.
You can do it manually with key 'm' into left remote.py.
If it fix the issue, I will try to reproduce it and make a fix into swd_diff_drive_controller.

suguruhamada1003910, 2022-11-01:
@GMezWheel 

Thank you for support! This problem was resolved!!

suguruhamada1003910, 2022-11-01:
Commissioning file was not correct.

suguruhamada1003910, 2022-12-12:
I noticed that commissioning fails even if "Commissioning succeeded!" is displayed.
I did commissioning with the same file in several times and roslaunch and it seems that roslaunch was succeeded but connection between SWD and IPC failed and "NMT state machine is not OK" was displayed.

Could you look into this problem? What files should I send? e.g. candump.

GMezWheel, 2022-12-12:
Could you send me the remote logs of the both drives ?
It seems that something prevents to changing NMT state to OPER.

suguruhamada1003910, 2022-12-13:
komatsu@komatsu-Latitude-7280:/opt/ezw/usr/sbin$ python3 remote.py swd_left
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21302SC003A1 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.3 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT state on error   : 1 ............................................................................ [OK]
NMT State            : PREOP  ....................................................................... [OK]
PDS State            : NOT_READY_TO_SWITCH_ON  ...................................................... [OK]
NbError              : 1 ............................................................................ [OK]
LastError            : 6020 : EZW_SYSTEM_ERROR_CODE_CANOPEN_PARAMETER_ERROR ......................... [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, True, True, True, True, True, True  .............................. [OK]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 0 ............................................................................ [OK]
OdometryValue        : 0 ............................................................................ [OK]
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
 

[10:32] Sakai, Keisuke
komatsu@komatsu-Latitude-7280:/opt/ezw/usr/sbin$ python3 remote.py swd_right
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 21302SC006A1 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
[... cut ...]
PositionValue        : -1 ........................................................................... [OK]
OdometryValue        : 0 ............................................................................ [OK]
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0

suguruhamada1003910, 2022-12-13:
This behavior is not always occurred. Sometimes commissioning fails, sometimes it succeeds in spite of the same file.
I would like to know the cause of fail.

GMezWheel, 2022-12-13:
Hi @suguruhamada1003910,

Indeed, the commissioning of the left drive has failed :
`LastError            : 6020 : EZW_SYSTEM_ERROR_CODE_CANOPEN_PARAMETER_ERROR ......................... [OK]`
=> this error appears when safety signatures (CRC) are mismatched (maybe an error during commissioning).

Can you send me both smc-services log ?
```
/tmp/logs/sce-swd-left.log
/tmp/logs/sce-swd-right.log
```
I suspect that an SDO request error has not been catched during commissioning.

GMezWheel, 2022-12-14:
@suguruhamada1003910 ,

Both DBus clients can't connect correctly to the drives :
`2022-12-14 9:15:3;SMCS;CAN;ERROR;operator();248;Initialize PDO failed;` 
Can you stop/start both smc-services DBus clients and check smc-services log again ?
If PDO initializing failed again, can you stop both smc-services DBus clients and start them manually and send me corresponding logs ?
```
export LD_LIBRARY_PATH=/opt/ezw/usr/lib
export $(cat /tmp/SYS*.id)
/opt/ezw/usr/bin/ezw-smc-service /opt/ezw/usr/etc/ezw-smc-core/swd_left_config.ini
/opt/ezw/usr/bin/ezw-smc-service /opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini
```

suguruhamada1003910, 2022-12-16:
>Can you stop/start both smc-services DBus clients and check smc-services log again ?
You means I need to restart ezw-swd-left.service and ezw-swd-right.service?
In other words, I need to do the following commands?

`sudo systemctl restart ezw-swd-left.service ezw-swd-right.service`

GMezWheel, 2022-12-16:
> @GMezWheel
> 
> > Can you stop/start both smc-services DBus clients and check smc-services log again ?
> 
> You mean I need to restart ezw-swd-left.service and ezw-swd-right.service. Is it right? In other words, should I do the following commands?
> 
> `sudo systemctl restart ezw-swd-left.service ezw-swd-right.service`

yes

suguruhamada1003910, 2022-12-21:
I'm trying to reproduce problem but it doesn't occur...

suguruhamada1003910, 2023-02-10:
I cannot reproduce problem. I close this.

suguruhamada1003910, 2023-02-27:
This problem is reproduced. When this problem is occurred and execute launch, the following messages are displayed.

NMT state machine is not OK.
PDS state machine is not OK.

And log file is below.

[sce-swd-left.log](https://github.com/ezWheelSAS/swd_ros_controllers/files/10836280/sce-swd-left.log)
[sce-swd-right.log](https://github.com/ezWheelSAS/swd_ros_controllers/files/10836281/sce-swd-right.log)

I did stop and start both smc-services DBus clients but this problem is not resolved.
The results of remote.py are below.

```
suguru@c-lati7280-076:/opt/ezw/usr/sbin$ python3 remote.py swd_left
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 22307SC002A1 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT state on error   : 1 ............................................................................ [OK]
NMT State            : PREOP  ....................................................................... [OK]
PDS State            : NOT_READY_TO_SWITCH_ON  ...................................................... [OK]
NbError              : 1 ............................................................................ [OK]
LastError            : 6020 : EZW_SYSTEM_ERROR_CODE_CANOPEN_PARAMETER_ERROR ......................... [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, True, True, True, True, True, True  .............................. [OK]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -2 ........................................................................... [OK]
OdometryValue        : -6 ........................................................................... [OK]
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
```

```
suguru@c-lati7280-076:/opt/ezw/usr/sbin$ python3 remote.py swd_right
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
[... 21 lines cut ...]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
[... cut ...]
PositionValue        : 2 ............................................................................ [OK]
OdometryValue        : 6 ............................................................................ [OK]
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
```

suguruhamada1003910, 2023-02-27:
After that, I did restart PC and commissioning with 2 SWD Cores. Then, this problem was resolved.

closed 2023-02-27 by suguruhamada1003910