IDEC-ezWheel/swd_ros_controllers #48: swd_diff_drive_controller.launch fails with swd-services v0.2.7 and swd-ros-controllers v3.3.0

suguruhamada1003910, 2022-10-28:
I bought SWD Starterkit and it ran properly. Then, I installed swd-services v0.2.7 and swd-ros-controllers v3.3.0 with apt command.
SWD do not run properly now.

Information is below.

```
atmobi@atmobi-box:~$ systemctl -a | grep ezw
  ezw-dbus-user-session.service                                                                                    loaded    active     running   EZW DBUS USER SESSION SERVICE                                                                   
  ezw-ros-bringup.service                                                                                          loaded    active     running   EZW ROS BRINGUP SERVICE                                                                         
  ezw-stack.service                                                                                                loaded    active     running   EZW APP STARTUP - PARENT SERVICE                                                                
  ezw-swd-left.service                                                                                             loaded    active     running   EZW SWD LEFT SERVICE                                                                            
  ezw-swd-right.service                                                                                            loaded    active     running   EZW SWD RIGHT SERVICE        
```

```
atmobi@atmobi-box:~$ python3 /opt/ezw/usr/sbin/remote.py swd_left
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 22150SC038A1 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT state on error   : 1 ............................................................................ [OK]
NMT State            : PREOP  ....................................................................... [OK]
PDS State            : NOT_READY_TO_SWITCH_ON  ...................................................... [OK]
NbError              : 0 ............................................................................ [OK]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, True, True, True, True, True, True  .............................. [OK]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -3 ........................................................................... [OK]
OdometryValue        : -2 ........................................................................... [OK]
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
Traceback (most recent call last):
  File "/opt/ezw/usr/sbin/remote.py", line 549, in <module>
    main(sys.argv[1:])
  File "/opt/ezw/usr/sbin/remote.py", line 544, in main
    menu_diagnostic()
  File "/opt/ezw/usr/sbin/remote.py", line 410, in menu_diagnostic
[... 117 lines cut ...]
[default] dbusNamespace = swd_right
[default] HWConfigurationEntry = SWD_CORE
[default] HWConfigurationFile = /opt/ezw/data/configuration.json
[default] CANOpenEDSFile = /opt/ezw/usr/etc/ezw-canopen-dico/swd_core.eds
2022-10-28 11:38:58;;SMCDS;WARN;load;44;Config;Invalid type for SWCompatibilityFile;
2022-10-28 11:38:58;;SMCDS;INFO;getHWConfigurationMember;90;HardwareHelper;Search for entry SWD_CORE / nbStepRevolutionElectype Integer into /opt/ezw/data/configuration.json;
2022-10-28 11:38:58;;SMCDS;INFO;getHWConfigurationMember;111;HardwareHelper;nbStepRevolutionElec : 6;
2022-10-28 11:38:58;;SMCDS;INFO;getHWConfigurationMember;90;HardwareHelper;Search for entry SWD_CORE / nbPolePairtype Integer into /opt/ezw/data/configuration.json;
2022-10-28 11:38:58;;SMCDS;INFO;getHWConfigurationMember;111;HardwareHelper;nbPolePair : 5;
2022-10-28 11:38:58;;SMCDS;INFO;getHWConfigurationMember;145;HardwareHelper;Search for entry SWD_CORE / reduction type Double into /opt/ezw/data/configuration.json;
2022-10-28 11:38:58;;SMCDS;INFO;getHWConfigurationMember;166;HardwareHelper;reduction : 14.000000;
2022-10-28 11:38:58;;SMCDS;INFO;getHWConfigurationMember;145;HardwareHelper;Search for entry SWD_CORE / diameter type Double into /opt/ezw/data/configuration.json;
2022-10-28 11:38:58;;SMCDS;INFO;getHWConfigurationMember;166;HardwareHelper;diameter : 125.000000;
2022-10-28 11:38:58;;SMCDS;INFO;init;76;DBusClientDelegate;Checking DBUS proxy availability : commonapi.ezw.smcservice.swd_right;
2022-10-28 11:40:58;;SMCDS;ERROR;init;83;DBusClientDelegate;Checking DBus proxy availability timeout (120s) for proxy commonapi.ezw.smcservice.swd_right;
[ERROR] [1666924858.471717588]: Failed initializing right motor, EZW_ERR: SMCService : Controller::init() return error code : 5
[swd_diff_drive_controller-1] process has died [pid 6031, exit code 1, cmd /opt/ros/noetic/lib/swd_ros_controllers/swd_diff_drive_controller __name:=swd_diff_drive_controller __log:=/home/atmobi/.ros/log/81b0d2b6-5668-11ed-8d17-03aa7ac47ebf/swd_diff_drive_controller-1.log].
log file: /home/atmobi/.ros/log/81b0d2b6-5668-11ed-8d17-03aa7ac47ebf/swd_diff_drive_controller-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

[candump.txt](https://github.com/ezWheelSAS/swd_ros_controllers/files/9884927/candump.txt)

suguruhamada1003910, 2022-10-28:
I also installed swd-system-config-2wheels.

GMezWheel, 2022-10-28:
Hi @suguruhamada1003910 ,

If you start manually swd_diff_drive_controller, you have to export DBus session environment variables:
[https://github.com/ezWheelSAS/swd_ros_controllers/#usage-on-a-swd-starter-kit](https://github.com/ezWheelSAS/swd_ros_controllers/#usage-on-a-swd-starter-kit)
`export $(cat /tmp/SYSTEMCTL_dbus.id)`

suguruhamada1003910, 2022-10-28:
@GMezWheel 

Thank you for reply. I will try it.
By the way, how does SWD Starterkit export it? I finally connect IPC and SWD automatically when power is ON.

GMezWheel, 2022-10-28:
The automatic export of DBus session (if exists) is made by ezw-ros-bringup.service.

NOTA : if the IPC has already a DBus window manager session, the ezw-dbus-user-session.service can be disable.
Thus, by default, swd_diff_drive_controller wil use DBus window manager session (no need to export DBus session environment variables).

suguruhamada1003910, 2022-11-01:
```
process 39532: arguments to dbus_set_error_from_message() were incorrect, assertion "(error) == NULL || !dbus_error_is_set ((error))" failed in file dbus-message.c line 3938.
This is normally a bug in some application using the D-Bus library.
  D-Bus not built with -rdynamic so unable to print a backtrace
```

I got new error message. Do you have any solution?

GMezWheel, 2022-11-01:
### Solution one
The dbus library is not the right one. Set "**export LD_LIBRARY_PATH=/opt/ezw/usr/lib**" before running application.
### Solution two
If the IPC has a window manager installed, there is already a DBus session for it. Thus, remove and don't create /tmp/SYSTEMCTL_dbus.id file. Then :
- restart the both ezw-smc-service. 
- check with remote.py
- launch swd_diff_drive_controller

suguruhamada1003910, 2022-11-02:
```
process 39532: arguments to dbus_set_error_from_message() were incorrect, assertion "(error) == NULL || !dbus_error_is_set ((error))" failed in file dbus-message.c line 3938.
This is normally a bug in some application using the D-Bus library.
  D-Bus not built with -rdynamic so unable to print a backtrace
```

I did not know the cause of this problem. I cannot resolve it. However, swd_diff_drive_controller.launch succeeded with other PC. So I close this.

closed 2022-11-02 by suguruhamada1003910