IDEC-ezWheel/swd_ros_controllers #49: SWD moves slightly to the left.

suguruhamada1003910, 2022-11-02:
I created program which SWD Starterkit runs forward and backward repeatedly. Then, SWD moves slightly to the left.
The motor on the left side appears to rotate slightly faster than the motor on the right side.

https://1drv.ms/v/s!AqHxzXmtIbtrqF1JNW5qfiJQfFip?e=e7wgB1

Do you have any idea to resolve this problem?

suguruhamada1003910, 2022-11-11:
I recorded the behavior of the running SWD Starterkit.

https://1drv.ms/u/s!AqHxzXmtIbtrqGKyXn12ck5eUW00?e=n31Go3

suguruhamada1003910, 2022-11-11:
> I recorded the behavior of the running SWD Starterkit.
> 
> https://1drv.ms/u/s!AqHxzXmtIbtrqGKyXn12ck5eUW00?e=n31Go3

I also logged the value of Twist to `/swd_diff_drive_controller/cmd_vel`.

[cmd_vel.txt](https://github.com/ezWheelSAS/swd_ros_controllers/files/9986197/cmd_vel.txt)

GMezWheel, 2022-11-13:
Thanks for the video.
I will try to reproduce the behavior this week and analyse your log.

GMezWheel, 2022-11-14:
To better understand the problem, can you send me a candump log ?

suguruhamada1003910, 2022-11-14:
@GMezWheel 

Please see the attached file.

[candump.txt](https://github.com/ezWheelSAS/swd_ros_controllers/files/10001240/candump.txt)

closed 2022-12-16 by suguruhamada1003910