IDEC-ezWheel/swd_ros_controllers #50: access to GPIO in user space

JacquesRoubaud, 2022-11-02:
Hi
I have been able to work with GPIO with python on the TEK3 computer, but I always need to work with "sudo"
Running a ros node to access gpio in sudo mode is not ideal

back to the basic, I can not simply call "gpioinfo" without starting with sudo. 
I have been googling a lot, and could not find anything helping. Also, in the linux installation,  I could not see a "gpio" group that I could authorize for user swd_sk.

Do you have a suggestion so that user swd_sk can run "gpioinfo" without sudo ?

Many thanks
Jacques

abougouffa, 2022-11-21:
Another solution is to set the `SUID` bit of the command you want to run as root.

For example, to set the `SUID` of `gpioinfo`, you can run this command:

```shell
sudo chmod u+s $(which gpioinfo)
```

After this, you should be able to call `gpioinfo` with root permissions and without typing the password.

JacquesRoubaud, 2022-12-01:
None of above worked, and I can't run a "rosrun" with sudo. So I really needed a solution. 

I could finally make it using this:
1. Create GPIO group and add swd_sk
`sudo groupadd gpio`
`sudo usermod -a -G gpio swd_sk`
`newgrp gpio`
check that gpuoi is part of the output of this command: 
`newgrp gpio`
here is my output : swd_sk sudo www-data video gpio

2. In /etc/udev/rules.d/10-imx.rules add a last line 
SUBSYSTEM=="gpio", KERNEL=="gpiochip[0-6]", GROUP="gpio", MODE="0660"

reboot

now it's possible to access GPIO in a ROS node

I don't know yet if this is valid for writing to output, but at least this sample program works:
import gpiod
import time

chip=gpiod.Chip('gpiochip0')
line = gpiod.find_line("GPIO_P239")
print("line")
print(line)
lines = chip.get_lines([line.offset()])
print(lines)
lines.request(consumer='foobar', type=gpiod.LINE_REQ_DIR_IN, default_vals=[0])
print('--------')
while True:
    vals = lines.get_values()
    print("=> ", lines.get_values()[0])
    time.sleep(1)
    print()

GMezWheel, 2022-12-02:
Hi @JacquesRoubaud,
I think your solution is the good one.
I also used a udev rule to enable permissions for read/write to usb tty (to avoid sudo).
```
SUBSYSTEM=="usb", ATTRS{idVendor}=="04d8", MODE="0660"
SUBSYSTEM=="usb_device", ATTRS{idVendor}=="04d8", MODE="0666"
```
NOTA : permissions 660 allow also write operation

closed 2022-12-02 by abougouffa