IDEC-ezWheel/swd_ros_controllers #52: Linear speed goes above 0.3m/s in SLS ?

JacquesRoubaud, 2022-11-21:
Hi
my system shows that I am in SLS mode, but the topic /swd_diff_drive_controller/odom shows linear speed above that speed, up to 0.5m/s, when I drive with the remote control. Is that normal ? It is oscillating a lot between 0.25 and 0.5m/s, whereas the wheels seem to spin regularly.

the objective is to use the real-time velocity to do line-following

abougouffa, 2022-11-21:
Hello @JacquesRoubaud,

As you can see in the `swd_diff_drive_controller` source code, the velocity is calculated from the difference of the wheel encoders' odometry. 
 https://github.com/ezWheelSAS/swd_ros_controllers/blob/d07357c886ba52d8c0404c62ce3aa218514c2f71/src/diff_drive_controller/DiffDriveController.cpp#L515-L516

There are two levels of synchronization in the API, in the first stage, `swd-services` samples the odometry at a fixed rate, and serves the `swd_diff_drive_controller` requests (made at a different rate). With that in mind, measured velocity may vary between two time instances (see the schema below).

```
  SERVICE       SAMPLING TIME
                t0   t1   t2   t3   t4   t5   t6   t7   t8
swd-services    |----|----|----|----|----|----|----|----|-->
swd-diff-drive   |----------|----------|----------|-------->
                 t0'        t1'        t2'        t3'
```

To use the velocity for control, you can either:

- Use a lower value for the `pub_freq_hz` parameter (20Hz should work fine);
- Or filter the velocity using, for example, a mean filter over a small window (3 to 5 should be fine).

ez-Support, 2022-11-22:
I tried to repdroduce the behaviour with a SLS activated, but I did not find such difference, see below:
[image]
[odom.csv](https://github.com/ezWheelSAS/swd_ros_controllers/files/10065867/odom.csv)

JacquesRoubaud, 2022-11-22:
Hi
thanks for the replies
setting the publish rate to 10Hz helps

closed 2022-11-22 by JacquesRoubaud