IDEC-ezWheel/swd_ros_controllers #54: to have steady low speed is difficult

JacquesRoubaud, 2022-11-22:
Hi
I am setting the speed to low values, like 0.01m/s up to 0.04m/s
I can see that the movement is not fully linear . At 0.01m/s one of the motor stops (or accelerate?) randomly for a very short period time, so short it's hard to see if it stops or accelerates.
Overall, I can see that at very low speed the direction fluctuates
with speed = 0.08m/s, everything is ok
Did you notice that and do you have some suggestion ?

GMezWheel, 2022-11-22:
Hi @JacquesRoubaud,
At this time, the swd motor velocity range is [40-2000] rpm by default. 
Thus, a target velocity value in the range [1-40] is forced to 40rpm by the diff_drive_controller node.
You can set the min value of the VelocityModeParameters.vl_velocity_min_amount during commissioning.

JacquesRoubaud, 2022-11-22:
Hi !
Good news ! is there anything particular to think about when setting the lowest velocity value ? can I set 1 rpm ?
Can you indicate me where to find the file with the parameter VelocityModeParameters.vl_velocity_min_amount ?
thanks

ez-Support, 2022-11-22:
It is not suitable to set-up the velocity speed under 40 rpm. The motor will not be able to follow the a target velocity below this value.

closed 2022-11-24 by ez-Support