IDEC-ezWheel/swd_ros_controllers #55: Additional request at closed Issues #51.

HirokiShikazaki, 2022-11-25:
Thank you for response to Issues #51.
Is it possible to add to the feature, that a ROS program can option enable/disable the brakes for 20 minutes after the SWD is turned off?

suguruhamada1003910, 2022-12-05:
My customer would like to delete this behavior (the brake of 20 minutes). Is there any solution?
If it is not possible, is there any reason?

Koma2yu, 2022-12-22:
Hello,

In the description of Braking systen in ISO3691-4 standard, the braking should be activated when the power supply was shutdown. So, we prefer that the enable/disable of the brakes for 20 minutes be configuarable when turn off SWD..
Could you consider the update it in next firemware release?

4.2 Braking system
The truck shall be equipped with a braking system that is designed to do the following:
a) operate on the interruption of the power supply;

HirokiShikazaki, 2022-12-22:
Reopen this request.
Please respond.

ez-Support, 2023-01-04:
What best suited this requirement is an external brake, which is turned on when the robot is parked and the power supply is off.

closed 2023-01-04 by ez-Support