IDEC-ezWheel/swd_ros_controllers #60: restart_acknowledge_behavior is not expected behavior

suguruhamada1003910, 2022-12-16:
I am using FW V1.0.3 of SWD Core and running SWD Startar Kit.

### Actual Behavior

- "restart_acknowledge_behavior" is "False"

When I turn ON emergency switch, SWD stops. After that, if I turn OFF emergency switch, `SWD runs again.`
When SDI is enabled, SWD stops. After that, if SDI is disabled, SWD runs again.

- "restart_acknowledge_behavior" is "True"

When I turn ON emergency switch, SWD stops. After that, if I turn OFF emergency switch, SWD keeps stopping.
When SDI is enabled, SWD stops. After that, if SDI is disabled, `SWD keeps stopping.`

### Expected Behavior

- "restart_acknowledge_behavior" is "False"

When I turn ON emergency switch, SWD stops. After that, if I turn OFF emergency switch, `SWD keeps stopping.`
When SDI is enabled, SWD is stopped. After that, if SDI is disabled, SWD runs again.

- "restart_acknowledge_behavior" is "True"

When I turn ON emergency switch, SWD stops. After that, if I turn OFF emergency switch, SWD keeps stopping.
When SDI is enabled, SWD is stopped. After that, if SDI is disabled, `SWD runs again.`

I am thinking that if emergency switch is ON and STO is enabled, AGV should not restored the behavior without power OFF and ON or reset switch.
And the behaivor of SDI should not be same as STO.

ez-Support, 2022-12-19:
Indeed, SDI should not be affected by the `restart_acknowledge_behavior` value. We will check that, and come back to you.
SDI function does not require any acknowledge to be released. Thus, when SDI is released, the SWD should run in any cases.

However, if "restart_acknowledge_behavior" is "False", after that the STO has been released, then the SWD should run again.

ez-Support, 2022-12-19:
Can you please send us the diagnostics from remote.py ? 
Thanks !

suguruhamada1003910, 2022-12-21:
suguru@c-latitude-7280-075:~$ python3 /opt/ezw/usr/sbin/remote.py swd_left
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 22150SC051A1 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.3 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT state on error   : 1 ............................................................................ [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : OPERATION_ENABLED  ........................................................... [OK]
NbError              : 0 ............................................................................ [OK]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : True, False, False, False, False, False, True, False  ........................ [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, True, True, True, True, True, True  .............................. [OK]
Peripheral           : True, True, True, True, True, True  .......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 1617 ......................................................................... [OK]
OdometryValue        : 1511 ......................................................................... [OK]
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0

suguru@c-latitude-7280-075:~$ python3 /opt/ezw/usr/sbin/remote.py swd_right
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 22150SC043A1 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.3 ........................................................................ [OK]
[... cut ...]
PositionValue        : 1663 ......................................................................... [OK]
OdometryValue        : 1564 ......................................................................... [OK]
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0

@ez-Support

ez-Support, 2023-01-04:
@suguruhamada1003910, I reproduced the behaviour. 

If "restart_acknowledge_behavior" is "False" :
- If you turn ON emergency switch, SWD stops. Then, if you turn OFF emergency switch, SWD runs again.
- If you turn ON SDI, SWD stops. Then, if you turn OFF SDI, SWD runs again.

If "restart_acknowledge_behavior" is "True" :
- If you turn ON emergency switch, SWD stops. Then, if you turn OFF emergency switch, SWD keeps stopping
- If you turn ON SDI, SWD stops. Then, if you turn OFF SDI, SWD runs again.

However, if you turn on SDI while the robot was moving, it triggers a SDI violation, and thus a STO signal is raised. Consequently, an STO acknowledge has to be made, before restarting.

suguruhamada1003910, 2023-01-13:
OK. I understood this is specification.

closed 2023-01-13 by suguruhamada1003910