IDEC-ezWheel/swd_ros_controllers #61: SWD Starter Kit always tilts to the left even if diameter is changed

suguruhamada1003910, 2022-12-22:
When SWD runs forward and backward repeatedly, it  tilts to the left even though The rear wheel has been changed to one that can only be front and rear.
Therefore, I adjusted diameter by following the #53 .

https://github.com/ezWheelSAS/swd_ros_controllers/issues/53

### Pattern 1

I changed diameter as below.
SWD_LEFT: 125.0
SWD_RIGHT: 120.0

As a result, SWD tilts left.

### Pattern 2

I changed diameter as below.
SWD_LEFT: 125.0
SWD_RIGHT: 130.0

As a result, SWD tilts left.

I recorded movies. Please confirm them.
https://1drv.ms/u/s!AqHxzXmtIbtrqQOxIv_tHgUkEvl4?e=GTpjal

How should I do to remove or decrease the tilting.

ez-Support, 2023-01-10:
It seems that the tilting occurs only when the kit slows down, i.e. velocity command migh be null at this moement.
The control of the wheel is made in velocity, thus it can explain that when stopping, both wheels do not travel the same distance.

Does the problem occur when going for a long distance straight ?

PS: 
Please send the diagnostic from remote.py for both wheels.

suguruhamada1003910, 2023-01-23:
@ez-Support 

Thank you for reply.

How long is "a long distance"?
SWD Starter Kit always tilts to left when stopping. Is it specification? Do you have any solution?

>PS:
>Please send the diagnostic from remote.py for both wheels.
I am sorry. I could not get data of remote.py now.

ez-Support, 2023-01-24:
The robot is controlled by speed with velocity commands, thus it does not monitor the distance travelled by each wheel.
Consequently, it is possible that on side of the robot goes further than the other when stopping.

suguruhamada1003910, 2023-01-26:
I understand that the change of diameter setting is not effective for tilting when SWD stops.

closed 2023-01-26 by suguruhamada1003910