IDEC-ezWheel/swd_ros_controllers #64: What is "coreNodeIsMaster"?

suguruhamada1003910, 2023-03-07:
`coreNodeIsMaster` exists in `swd_right_config.ini` and `swd_left_config.ini`.
Could you tell me the meaning of the parameter?

GMezWheel, 2023-03-08:
This parameter is used to configure the core node (left/right linux dbus client) as a master or as a slave CANopen node.
A master node can request to change the NMT state of an other CANopen node for example.
Indeed, the left/right SWD motors as configured as slave CANopen node and start in NMT PREOP state.
Thus the both core nodes (if configured as a maste node) can request the left/right SWD to go to NMT OPER state.

suguruhamada1003910, 2023-03-09:
@GMezWheel 

Thank you for reply. I understand the functionality.
Currently, the default values of `coreNodeIsMaster` in swd_right_config.ini and swd_left_config.ini are `True`. It means SWDs are Master.
And if I change it to `False`, SWD cannot run.

I think if I use `coreNodeIsMaster`, I should change source code. Is it right?

GMezWheel, 2023-03-10:
@suguruhamada1003910,

You're right. You have to change swd_ros_controller code in order to not try to set the both SWD to NMT OPER mode.
```
 if (!m_nmt_ok) {
                // Broadcast NMT command PREOP to all canopen nodes
                ezw_error_t err = m_left_controller.broadcastNMTState(smccore::INMTService::NMTCommand::PREOP);
                if (ERROR_NONE != err) {
                    ROS_ERROR(
                        "Failed to broadcast NMT command PREOP"
                        "Controller::broadcastNMTState() return error code : %d",
                        (int)err);
                }
                else {
                    usleep((10) * 1000);
                    // Broadcast NMT command OPER to all canopen nodes
                    err = m_left_controller.broadcastNMTState(smccore::INMTService::NMTCommand::OPER);
                    if (ERROR_NONE != err) {
                        ROS_ERROR(
                            "Failed to broadcast NMT state OPER"
                            "Controller::broadcastNMTState() return error code : %d",
                            (int)err);
                    }
                }
            }
```
And you must have an other master CANOpen node that pass all nodes into NMT OPER; in order to have swd_ros_controller work again !

suguruhamada1003910, 2023-03-10:
@GMezWheel 

Thank you so much! I understand.

closed 2023-03-10 by suguruhamada1003910