IDEC-ezWheel/swd_ros_controllers #65: Control two starter kits from one PC.

Koma2yu, 2023-03-19:
I would like to control two Starter kits separately via CAN communication from one PC.
In that case, the CAN IDs are covered by the default settings, so I think I need to set the CAN IDs for the first and second Starter kit to different IDs. How should I change the CAN ID? Also, are there any other settings that need to be changed?

ez-Support, 2023-03-20:
To hange the Node-ID of the SWD, you can use the commissioning file the SWD:
[https://github.com/ezWheelSAS/swd-starter-kit-config/tree/main/commissioning/SafetyHub12pts](url)

In the python file you can set the value here:
`node_id = 0x4`

NOTA: This requires to execute the commissioning to each wheel where Node-ID has to be changed, one after the other

Koma2yu, 2023-03-29:
Could you give us an example of how to assign Node-IDs when trying to connect two starter kits together to make a 4-wheel drive?

ez-Support, 2023-03-30:
Please follow this procedure to change the Node-ID:

1. `systemctlstop ezw-ros-bringup`
2. `cd ~/commissioning/SafetyHub12pts/`
3. In the file ./swd_left_4_commissioning.py change `node_id = 0x4` to your value, e.g. `0x14`
4. `python3 ./swd_left_4_commissioning.py`, wait until you see "Commissioning succeeded!» at the end of the script execution.
5. In the file ./swd_right_5_commissioning.py change `node_id = 0x5 ` to your value, e.g. `0x15`
6. `python3 ./swd_right_5_commissioning.py`, wait until you see “Commissioning succeeded!» at the end of the script execution.

Come back to me if you need more help

Koma2yu, 2023-03-30:
Thank you for the information on how to write the Node-ID for SWD Core.
But what I want to know is how to control 4 wheels (4 Node-IDs) with swd_ros_controller. The swd_ros_controller program only supports control of 2 wheels, so I do not know how to control 4 wheels.

ez-Support, 2023-03-30:
Yes indeed, `swd_ros_controllers` provided by ez-Wheel, is a controller that works for a 2 wheels differential drive.
The code is open source and free to be downloaded and modified. It works as an example of integration for SWD Safety Wheel Drives.
In your case, you should have a look for a ROS package that can run a 4 wheel drive mobile robot architecture, and adpat the `swd_ros_controllers` to it.

Koma2yu, 2023-03-31:
Thank you for your answer.

closed 2023-03-31 by Koma2yu