IDEC-ezWheel/swd_ros_controllers #66: When SWD runs in low speed, running is not stable

suguruhamada1003910, 2023-04-06:
As you can see in the following video, running is not stable whe SWD runs in low speed.
https://1drv.ms/v/s!AqHxzXmtIbtrqgKw63yjUhrH9J-X?e=OoX872

Speed is changed without periodicity.
I am thinking it is occurred with minimum speed. 

`[swd_diff_drive_controller-1] [INFO] [1680766698.369958939] [swd_diff_drive_controller]: The target speed falls behind the minimum speed limit (left, right) (26, 26 rpm).Set speed to (left, right) (40, 40) rpm`

How can I change the minimum speed?

ez-Support, 2023-04-06:
Hi Suguru, which version of Firmware are you using ? 

Be aware, RPM speed at motor shaft, should not be lower than 5 RPM.

The configuration is made in SWD's CANopen dictionnary in object `UINT32: 0x6046:0x01 `

You can set it, in the commissioning file for example, using:
`can_open_client.setValueUInt32(0x6046_01_20, 5)`

ez-Support, 2023-04-06:
Was the process of Fine calibration made on the SWD motors ?
The procedure was sent to IDEC before

suguruhamada1003910, 2023-04-07:
`Hi Suguru, which version of Firmware are you using ?`

I will check it but I am using prototype of SWD125 which sent from EZWheel. I did not update firmware.

```
Be aware, RPM speed at motor shaft, should not be lower than 5 RPM.
The configuration is made in SWD's CANopen dictionnary in object UINT32: 0x6046:0x01 
You can set it, in the commissioning file for example, using:
can_open_client.setValueUInt32(0x6046_01_20, 5)
```
OK. I will try it.

```
Was the process of Fine calibration made on the SWD motors ?
The procedure was sent to IDEC before
```
Is the firmware for calibration also available with SWD125?

suguruhamada1003910, 2023-04-07:
I paste information of remote.py to check firmware version.

```
python3 /opt/ezw/usr/sbin/remote.py swd_left
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 22307SC002A1 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT state on error   : 1 ............................................................................ [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : OPERATION_ENABLED  ........................................................... [OK]
NbError              : 2 ............................................................................ [OK]
LastError            : 800E : FAULT_ALIM_EXT ........................................................ [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, True, True, True, True, True, True  .............................. [OK]
Peripheral           : True, True, True, True, True, True  .......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 1 ............................................................................ [OK]
OdometryValue        : 3 ............................................................................ [OK]
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
```

```
python3 /opt/ezw/usr/sbin/remote.py swd_right
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_right .................................................................... [OK]
Node ID              : 0x5 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 22307SC001A1 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.1 ........................................................................ [OK]
[... cut ...]
PositionValue        : 5 ............................................................................ [OK]
OdometryValue        : 16 ........................................................................... [OK]
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
```

ez-Support, 2023-04-07:
`Is the firmware for calibration also available with SWD125?`
Yes, it works for both. I really recommand you to use the process of fine calibration that was sent to you for improving low speed behaviour.

ez-Support, 2023-04-07:
For your information, some debian packages for the StarterKit have been released, you can update them using apt install, if you want to :

`sudo apt search 'swd[_-]’`
`sudo apt install "swd_package"`

suguruhamada1003910, 2023-04-07:
Thank you so much!! I will try them!!

closed 2023-04-07 by suguruhamada1003910