IDEC-ezWheel/swd_ros_controllers #67: How can I four SWDs with one PC?

suguruhamada1003910, 2023-04-06:
I read the following link but I cannot resolve my issue.

[Control two starter kits from one PC](https://github.com/ezWheelSAS/swd_ros_controllers/issues/65)

When I control 4 SWDs with one PC, I think there are two method.

1. Controlling two `swd_ros_controllers` nodes in two PC from one PC
[image]

2. Controlling two `swd_ros_controllers` nodes in one PC
[image]

I think I can create method 1.
Can I create method 2?

I am thinking I can create method 2 by creating two `ezw-swd-left.service` and two `ezw-swd-right.service` in one PC.

ez-Support, 2023-04-06:
Warning CAN IDs 6 and 7, are used by the swd services instances. Thus i recommand using IDs 0x14 and 0x15 for the 2 additional wheels ;)

ez-Support, 2023-04-06:
What is the set-up of your wheels on the robot ?

suguruhamada1003910, 2023-04-07:
`Warning CAN-IDs 0x6 and 0x7 are used by the swd services instances (in the IPC). Thus I recommand using IDs 0x14 and 0x15 for the 2 additional wheels ;) As a reminder, CAN-IDs must be unique on the CANopen bus field.`
Thank you for advice!!

`What is the set-up of your wheels on the robot ?`
I am not sure the meaning of your question but I try to explain the details.
My robot needs to carry very heavy object. The power of SWD is not enough. Therefore, I am thinking I equip four wheels in my robot.

ez-Support, 2023-04-07:
Ok! What is the load to move and the maximum speed required ?

suguruhamada1003910, 2023-04-07:
> Ok! What is the load to move and the maximum speed required ?
I am checking information.

By the way, what do you think method 2? Do you have another the best way?

suguruhamada1003910, 2023-04-07:
> Ok! What is the load to move and the maximum speed required ?

Please check below.

```
maximum speed：1m/s
for one starter kit, load will be around 120kg
for one starter kit, load will be around 240kg
vertical load: max. 30kg per axis
```

ez-Support, 2023-04-07:
So the total load will be < 500kg?

ez-Support, 2023-04-07:
Yes method2 looks better.

suguruhamada1003910, 2023-04-10:
I correct information of AMR.

```
maximum speed：1m/s
for one starter kit, load will be around 120kg
for two starter kit, load will be around 240kg
vertical load: max. 30kg per axis
Size of Wheel: 200mm

User want to keep 1m/s when loading 360kg.
```

>I really do think using 2xSWD-125 or 2xSWD-150 would be much easier to implement in that case.
>Why using a 4x wheel drive set-up here ?
SWD125 is not selling in Japan for now.

suguruhamada1003910, 2023-04-11:
I don't have any query for now. I close this.

suguruhamada1003910, 2023-04-14:
Hi Support, I am creating the following configuration.

[image]

Then, I can communicate with four SWDs by using remote.py.
However, I could not control four SWDs. Specifically, only 2 SWDs are running by joy.

I am thinking dbus session is not specified correctly. In the following code, only one DBUS_SESSION_BUS_ADDRESS can be specified.
https://github.com/ezWheelSAS/swd_ros2_controllers/blob/main/launch/swd_diff_drive_controller.launch.py#L17-L25

If I write code as below, DBUS_SESSION_BUS_ADDRESS is overwritten and PC can communicate with only one session.

```
    # Load specific dbus user session if exists
    session = "/tmp/SYSTEMCTL_dbus.id"
    if os.path.isfile(session):
        print("SYSTEMCTL_dbus.id detected")

        with open(session) as f:
            lines = f.readlines()
        env = dict(line.strip().split("=", 1) for line in lines)
        os.environ.update(env)

    # Load specific dbus user session if exists
    session = "/tmp/SYSTEMCTL_dbus2.id"
    if os.path.isfile(session):
        print("SYSTEMCTL_dbus2.id detected")

        with open(session) as f:
            lines = f.readlines()
        env = dict(line.strip().split("=", 1) for line in lines)
        os.environ.update(env)
```

I want to specify SYSTEMCTL_dbus.id for CAN ID 4 and5 and SYSTEMCTL_dbus2.id for CAN ID 14 and 15. Do you have any idea?

closed 2023-04-14 by suguruhamada1003910

suguruhamada1003910, 2023-04-14:
The delay is needed to avoid overwritten DBUS_SESSION_BUS_ADDRESS.

```
TimerAction(
         period=5.0,
         actions=[
               swd_diff_drive_controller_description2(),
         ]),
```