IDEC-ezWheel/swd_ros_controllers #68: SWDs ignore input of cmd_vel when direction of rotation is changed

suguruhamada1003910, 2023-04-17:
Hi support.

I'm developing navigation program with SWDs. Then, SWD may swing greatly from side to side. I'm investigating the cause.
https://onedrive.live.com/?id=6BBB21AD79CDF1A1%215401&cid=6BBB21AD79CDF1A1

I noticed that when the SWD rotates clockwise after turning right, or rotates clockwise after turning left, it moves like ignoring the input to cmd_vel.
https://1drv.ms/f/s!AqHxzXmtIbtrqh-q1WP-jk5iQ-gY?e=FOleuA

I suspect this is the cause of the problem. The navigation program is looking at the gyro sensor, and when the program try to move to the right or left, the SWD ignores the value of cmd_vel for a while and does not rotate, so the program input to cmd_vel to try to rotate further, and I think it is going too far.

Do you have any solution or do you know the cause of the behavior of SWD?

Note:
When I try to move SWD right, stop and right, SWD is running according to cmd_vel immidiately.
https://1drv.ms/f/s!AqHxzXmtIbtrqh4-d7fgdl9IbecD?e=E5AJ5J

ez-Support, 2023-04-20:
Did you change the configuration of the gearbox (4 instead of 14) ratio in config.json:
configuration.json
```
[
   {
      "name": "SWD_CORE",
      "nbStepRevolutionElec": 6,
      "nbPolePair": 5,
      "reduction": 14.0,
      "diameter": 125.0
   }
]
```

suguruhamada1003910, 2023-04-20:
> Is the problem occuring also with the joystick ? It looks like a latency problem

Yes. This problem is occurring with joystick.

> Did you change the configuration of the gearbox (4 instead of 14) ratio in configuration.json file:
Yes. I already changed the value of reduction for SWD125. I'm using SWD125.

I cannot reproduce this problem with SWD Core.

ez-Support, 2023-04-20:
Can you try using a higher velocity demand with for example `cmd_vel=0.2` instead of 0.1.

ez-Support, 2023-04-20:
After some investigations, we offer you to set-up the integrative value of SWD's PID controller.
This should improve its behavior as it has a lower ratio (/4).

Please try to change the value of the integrative parameter to `2` in the SWD:
`motctrl_speed_pid_i ==> U32 0x2156:00`

You can do it in the commissioning for instance.
Example:
`can_open_client.setValueUInt8(0x2156_00_20, 2)`

shamada-idec, 2023-04-21:
> After some investigations, we offer you to set-up the integrative value of SWD's PID controller. This should improve its behavior as it has a lower ratio (/4).
>
> Please try to change the value of the [...]

Thank you for reply.
I tried `can_open_client.setValueUInt8(0x2156_00_20, 2)` but it fails. You said `motctrl_speed_pid_i ==> U32 0x2156:00`. It is right what I use UInt8 function?

BTW, what is the default value of `motctrl_speed_pid_i` ?

ez-Support, 2023-04-21:
No indeed, sorry for the mistake, you should use the method: `setValueUInt32()`
To get the value that was set before, you can do a get and print the value, e.g. `getValueUInt32()`

suguruhamada1003910, 2023-04-21:
I could change the value! I try to adjust program again.

closed 2023-04-21 by suguruhamada1003910

ez-Support, 2023-04-24:
I come back to you, to see how is the robot going with the new parameters ?

suguruhamada1003910, 2023-04-25:
> I come back to you, to see how is the robot going with the new parameters ?

I changed the value of P `motctrl_speed_pid_p` and I `motctrl_speed_pid_i`.
As a result, I think the behavior of rotation is good!!

ez-Support, 2023-04-25:
Nice !!! 
What did you set ad P parameter ?

shamada-idec, 2024-12-26:
@ez-Support 

I would like to change P, I, and D.
As you said, I think setValueUint32() can be used as below.

can_open_client.setValueUInt32(0x2155_00_20, 2) #P
can_open_client.setValueUInt32(0x2156_00_20, 2) #I
can_open_client.setValueUInt32(0x2157_00_20, 2) #D

Then, what is the meeting of `20` in `0x215x_00_20`?
I checked the ObjectDictionary on Web page of SWD but I could not understand the meaning.

https://download.ez-wheel.com/SWD_CANopen_ObjectDictionary_v2_0_2.html#8533

[image]

ez-Support, 2025-01-06:
Hello Suguru,
Indeed you can use the methods above, but I think this is more relevant in our case:

```
swd_params, error = commissioning.manufacturer_client.getSWDParameters()
    swd_params.motctrl_speed_pid_p = 151
    swd_params.motctrl_speed_pid_i = 401
    swd_params.motctrl_speed_pid_d = 51
    commissioning.manufacturer_client.setSWDParameters(swd_params)
 
```

In CANopen when you are using a SDO requests, you need to mention the **Index**, **Sub-index** and **Size** of the objects you want to address. In this case 0x2155_00_20:

- 0x2155 = Index
- 00 = Sub-index
- 20 = Size (in hexadecimal <=> 0x20 = 32 (d) bits, as the value is an UNSIGNED32.

shamada-idec, 2025-01-06:
Thank you for advice. I completely understand how to use  ObjectDictionary!