IDEC-ezWheel/swd_ros_controllers #69: PDS State cannot be enabled

suguruhamada1003910, 2023-04-26:
I would like to run SWD Core by "o" command of remote.py but SWD does not run.
The result of remote.py is below. How should I do?

```
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 22069SC007A1 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.3 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT state on error   : 1 ............................................................................ [OK]
NMT State            : PREOP  ....................................................................... [OK]
PDS State            : SWITCH_ON_DISABLED  .......................................................... [OK]
NbError              : 0 ............................................................................ [OK]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, True, True, True, True, True, True  .............................. [OK]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: False, position_polarity: False ........................... [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 0 ............................................................................ [OK]
OdometryValue        : 0 ............................................................................ [OK]
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
PDS                  : OPERATION_ENABLED  ........................................................... [KO]
PDS State            : SWITCH_ON_DISABLED  .......................................................... [OK]

```

suguruhamada1003910, 2023-04-27:
```
DBUS Instance        : swd_left ..................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
CAN device           : can0 ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x10001 ...................................................................... [OK]
Serial number        : 22069SC007A1 ................................................................. [OK]
Revision number      : 4 ............................................................................ [OK]
SWVersion            : 1.0.3 ........................................................................ [OK]
HWVersion            : 3.0.0 ........................................................................ [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT state on error   : 1 ............................................................................ [OK]
NMT State            : PREOP  ....................................................................... [OK]
PDS State            : NOT_READY_TO_SWITCH_ON  ...................................................... [OK]
NbError              : 0 ............................................................................ [OK]
LastError            : 0 : Error_reset_or_no_error .................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, False, False, False, False, True, True  .......................... [OK]
Peripheral           : True, True, True, False, True, True  ......................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 0 ............................................................................ [OK]
OdometryValue        : 0 ............................................................................ [OK]
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
<COM> Connect NodeID = i Disconnect = d Restore param = j
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n
<Mode> local = w remote = z
<Diagnostic> a
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
EMCY list : 
   1 : 	[6020] EZW_SYSTEM_ERROR_CODE_CANOPEN_PARAMETER_ERROR
```

Another error is displayed...

suguruhamada1003910, 2023-04-27:
When I connect as below, SWD can be ran by input of remote.py.

PC --- USB-CAN Converter --- SWD1, SWD2 and SafetyHub

However, when I connect as below, SWD cannot be ran by input of remote.py.

PC --- USB-CAN Converter --- SWD1 only

ez-Support, 2023-05-02:
Can you share the diagnostic for swd_right too ?
To see both wheels state.

suguruhamada1003910, 2023-05-12:
@ez-Support 

I am sorry for the delayed reply.

>Can you share the diagnostic for swd_right too ?
>To see both wheels state.
There is not right motor because only one SWD is connected with PC. So, "swd_right" is always fails.

Current wiring is below.
PC --- USB-CAN Converter --- SWD1 only

ez-Support, 2023-05-25:
Can you update the commissioning of the wheel, with SRDO disactivated, i.e. comment this line in the commisioning file:

    #
    # Update SRDO parameters
    #
`    update_SRDO_parameters()`

The use the remote.py to put the SWD in NMT OPERATIONAL and PDS in OPERATION ENABLED

suguruhamada1003910, 2023-06-05:
> Can you update the commissioning of the wheel, with SRDO disactivated, i.e. comment this line in the commisioning file:
> 
> ```
> #
> # Update SRDO parameters
> #
> ```
> 
> ` update_SRDO_parameters()`
> 
> The use the remote.py to put the SWD in NMT OPERATIONAL and PDS in OPERATION ENABLED

I'm sorry for the delayed reply.
I will test it.

suguruhamada1003910, 2023-06-06:
PDS state could be enabled!! Thank you for support!!

closed 2023-06-06 by suguruhamada1003910