IDEC-ezWheel/swd_ros_controllers #70: Listing ROS Topics

alrad32, 2023-05-25:
Hello,
I am working with the starterkit. To begin development on the platfrom, I am trying to get access to the topics, nodes, etc that are running on the ros docker image. I tried accessing the ros-noetic docker image that runs upon startup via `docker exec -it ros-noetic bash`. However, all ros commands (rostopic, rosnode, etc) are not found. How can I communicate with the running ros master via the host machine (the latter has no ros installation)?
Thanks!

GMezWheel, 2023-05-25:
Hi @alrad32 ,

In order to begin development on the platform, I suggest you to launch Visual Studio Code as editor on your host and connect to the starter kit via remote explorer addon.
=> select /home/swd_sk as working directory
Then, open a terminal into Visual Studio Code and enter into the docker : 
`docker exec -it ros-noetic bash`
Then, you can now run ROS command or build you workspace ROS and modify your code.

_NOTA 1 : the option "ForwardX11 yes" in your Visual Studio Code ssh config allows X11 redirection. Then allow xhost+ and configure DISPLAY env to allow X11 redirection from the docker in order to start rviz for example._

_NOTA 2 : it is also possible to start rviz from Windows by open a Remote Desktop connection to the starter kit. Then, on the debian Desktop, click on the docker terminal icon to open a terminal into the docker with X11 redirection._

alrad32, 2023-05-25:
Thanks for the reply.
I tried out your method, but still a simple `rostopic list` command (from within the docker image shell) returns a "rostopic: command not found". I checked the running containers and I can see that the "ros-noetic" image is up and running. Also, when listing all environemnt variables, there are no ros related ones. I can confirm that nodes such as hector slam are also running when I access the webpage via the master_ip. What am I missing here?

alrad32, 2023-05-25:
I don't see the log messages when entering the docker image. However, sourcing the 2 files solves the problem. Thanks a lot!
Where should I add the commands to run them by default with the docker image?

GMezWheel, 2023-05-25:
Add those lines at the end of your ~/.bashrc :
```
# Set docker environment
[ -s /.dockername ] && source /opt/ezw/install/setup.bash
```

closed 2023-06-19 by alrad32