IDEC-ezWheel/swd_ros_controllers #72: Fowarding GUI Apps from starter-kit

hazemy, 2023-06-19:
Hello,

I am trying to forward GUI applicaitons (rviz, gazebo, etc) from the docker container running on the starter-kit via SSH to a client machine that is connected to a display. I have setup X11Fowarding in the sshd_config, enabled xhost to accept incoming connections, and ran a docker container with the DISPLAY env set as the default container IP (172.17.0.1). I believe some of the setup steps are still missing or are misconfigured.
Could you please provide detailed steps for achieving such forwarding?

ez-Support, 2023-06-20:
Hello,

You might need to export your display environment variable

In your debian terminal get your DISPLAY value:
`echo $DISPLAY`
In your docker terminal, export the DISPLAY, e.g.:
`export DISPLAY=localhost:10.0`

hazemy, 2023-06-29:
Hello,
Thanks for the suggestion. The RDP option works well for me and I can now access the robot.

closed 2023-06-29 by hazemy