IDEC-ezWheel/swd_ros_controllers #73: Open gazebo GUI

Carpool1911, 2023-06-27:
Hallo,
I have cloned https://github.com/ezWheelSAS/swd_ros_controllers repo. I can run catkin_make sucessfully. I have to open Gazebo alongside with swd_ros_controller.launch. So, I have added few lines in swd_ros_controller.launch file. Could you guide me to how to open Gazebo with ezwheel and navigate via keyboard key. I dont know if I am mission to menstion in launch file. Here is my launch file.
<?xml version="1.0"?>

<launch>
  <include file="$(find gazebo_ros)/launch/empty_world.launch" />
  <arg name="model" />
  <arg name="baseline_m" default="0.485" />
  <arg name="pub_freq_hz" default="20" />
  <arg name="left_swd_config_file" default="/opt/ezw/usr/etc/ezw-smc-core/swd_left_config.ini" />
  <arg name="right_swd_config_file" default="/opt/ezw/usr/etc/ezw-smc-core/swd_right_config.ini" />
  <arg name="command_timeout_ms" default="500" />
  <arg name="control_mode" default="Twist" />
  <arg name="base_frame" default="base_link" />
  <arg name="odom_frame" default="odom" />
  <arg name="wheel_max_speed_rpm" default="75" />
  <arg name="wheel_safety_limited_speed_rpm" default="40" />
  <arg name="have_backward_sls" default="false" />
  <arg name="left_encoder_relative_error" default="0.2" />
  <arg name="right_encoder_relative_error" default="0.2" />
  <arg name="publish_odom" default="true" />
  <arg name="publish_tf" default="true" />
  <arg name="publish_safety_functions" default="true" />
  <node name="spawn_model" pkg="gazebo_ros" type="spawn_model"
        args="-file $(find swd_ros_controllers)/urdf/swd_starter_kit.urdf -urdf -model swd_starter_kit"
        output="screen" />
    <node name="spawn_model" pkg="gazebo_ros" type="spawn_model"
        args="-file $(find swd_ros_controllers)/urdf/swd_starter_kit.urdf -urdf -model swd_starter_kit"
        output="screen" />
  <node pkg="swd_ros_controllers" name="swd_diff_drive_controller" type="swd_diff_drive_controller" output="screen">
    <param name="baseline_m" value="$(arg baseline_m)" />
    <param name="pub_freq_hz" value="$(arg pub_freq_hz)" />
    <param name="left_swd_config_file" value="$(arg left_swd_config_file)" />
    <param name="right_swd_config_file" value="$(arg right_swd_config_file)" />
    <param name="command_timeout_ms" value="$(arg command_timeout_ms)" />
    <param name="control_mode" value="$(arg control_mode)" />
    <param name="base_frame" value="$(arg base_frame)" />
    <param name="odom_frame" value="$(arg odom_frame)" />
    <param name="wheel_max_speed_rpm" value="$(arg wheel_max_speed_rpm)" />
    <param name="wheel_safety_limited_speed_rpm" value="$(arg wheel_safety_limited_speed_rpm)" />
    <param name="have_backward_sls" value="$(arg have_backward_sls)" />
    <param name="left_encoder_relative_error" value="$(arg left_encoder_relative_error)" />
    <param name="right_encoder_relative_error" value="$(arg right_encoder_relative_error)" />
    <param name="publish_odom" value="$(arg publish_odom)" />
    <param name="publish_tf" value="$(arg publish_tf)" />
    <param name="publish_safety_functions" value="$(arg publish_safety_functions)" />

  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find swd_ros_controllers)/config/swd_starter_kit.rviz"
        if="$(arg gui)"/>
  </node>
</launch>

ez-Support, 2023-06-28:
Hello,

You have to open your gazebo simulation on your laptop (not on the iMX PC), as it requires a lot of ressources.

In gazebo, it will not use the swd_diff_drive_controller for the motion of the robot. Thus, I'm not sure what do you intend to do on Gazebo ? Can you tell us more on your application and configuration ?

Thanks

Carpool1911, 2023-06-29:
Hello ez-Support team,
thank you clarify. I want to simulation with ez-Wheel robot model in gazebo. I am trying to move ez-Wheel robot with keyboard keys in Gazebo. If you have any reference or documentation, could you share it ? 

Thank you

Carpool1911, 2023-07-04:
Hello ez-Support team,
thank you for your response. If I have any further question, I will let you know. :)

closed 2023-07-04 by Carpool1911