IDEC-ezWheel/swd_ros_controllers #74: Navigation Stack Computational Load

hazemy, 2023-06-29:
Hello,
I am currently experimenting with the navigation stack on the starter kit. Running the path planners on the robot seems to consume very large computational resources (even without rviz). After reducing the control loop rate and map update rate (in order to fit their deadlines) and decreasing the map resolution, the robot still doesn't manage to reach a given goal. Running rviz alone consumes significant resources and reduces the responsiveness of other running processes. Is this a known issue with the computational resources available on the IMX PC? If so, what could be some suggestions to mitigate the issue?

ez-Support, 2023-06-30:
Hello Hazemy, thanks for your question.

Indeed, the iMX8 has limited ressources, nevertheless it is enough for ROS process algorithmes.
For graphical applications, we recommand you to use ROS on your local laptop (rviz for instance).

Thus, you have to redirect the ROS data flows to the wlan like that:

`export CYCLONEDDS_URI=file://$HOME/cyclonedds.xml`
`sudo vi ~/cyclonedds.xml`

copy paste this in `cyclonedds.xml` file:
```
<?xml version="1.0" encoding="utf-8"?>
<CycloneDDS xmlns="https://cdds.io/config"
xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
xsi:schemaLocation="https://cdds.io/config https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">

  <Domain Id="any">
    <General>
      <NetworkInterfaceAddress>wlan0</NetworkInterfaceAddress>
    </General>
  </Domain>
</CycloneDDS>
```

Then, you will be able to see the starterkit topics and so on on your laptop

hazemy, 2023-07-07:
Hello,
Thanks a lot for the suggestion. I managed to have a better performance with the navigation stack (and visualization) running remotely, and keeping the ros master on the EzWheel PC. The navigation is still not very smooth but I think it could get better with more tuning and, potentially, with another computer connected via LAN to the onboard PC to avoid wireless latency.
For the tuning, is there an estimate to the maximum acceleration of the starter kit?

ez-Support, 2023-07-07:
It might take around 2seconds to reach swd's full speed.
The motor design is made for low consumption

closed 2023-07-07 by ez-Support