IDEC-ezWheel/swd_ros_controllers #75: Starterkit Connection Blackout

alrad32, 2023-08-08:
Hello,
I was modifying the `/etc/network/interfaces`, in particular the wireless connections part. However, now I cannot detect neither the starterkit access point (from the list of WiFi networks available on my PC), nor does the attched WiFi dongle seem to be connected to the internet router (as seen from the detected devices on the router's setup page). What options do I currently have? Can I connect via ethernet? Preferably, can I attach a display to the starter kit PC? Any other ideas?

alrad32, 2023-08-10:
Hello,
Support on the matter is highly appreciated.

ez-Support, 2023-08-10:
Hello, 

- On the iMX6 you have a HDMI port to connect a screen, and you can use USB to connect a keyboard.
- On the iMX8 you have a serial port using the USB connector.

ez-Support, 2023-08-10:
Indeed, you can also use the ethernet port .

alrad32, 2023-08-10:
Thank you so much for the reply. I was able to access the iMX8 PC via the serial port. I checked the network interfaces and it seems that wlan0 (which corresponds to the starter kit's access point) is down. I tried restarting the interface and restarting the network manager but the interface still does not start. In the `/etc/network/interfaces`, the interface details are as follows: 
```
auto wlan0
iface wlan0 inet static
    hostapd /etc/hostapd/hostapd.conf
    address 10.10.0.1/24
```
How can I check the configuration for the access point?

alrad32, 2023-08-14:
The issue is resolved by proper setup of the interfaces file. Thanks!

closed 2023-08-14 by alrad32