IDEC-ezWheel/swd_ros_controllers #79: safe_torque_off is not ture when the STO_1 is OFF

shamada-idec, 2025-03-04:
When I wire SWD as below and only STO_1 is OFF, `safe_torque_off` in `safety` topic is always `False`. On the other hand, when only STO_2 is OFF, `safe_torque_off` becomes `True` or `False` according to STO_2.

[image]

Commissioning setting is below.

[image]

Is this bug?

ez-Support, 2025-03-04:
Yes, the ROS safety topic only monitors one out of the two inputs, and this is the same for the SLS and SDI.
This should be fixed in the differential controller.
Feel free to fix it and open a pull request :)

shamada-idec, 2025-03-13:
I created PR for ROS1 and ROS2. Could you review and merge them?

ez-Support, 2025-03-14:
Hello, 

Thanks for your help. First, we will have to review and test the modifications. 
If it is successfull we will merge them, and then we will have to create debian packages with the update.