IDEC-ezWheel/swd_ros_controllers #81: `packages.ez-wheel.com:8081` is working ?

ar90n, 2025-05-19:
We are trying to evaluate [SWD® StarterKit](https://www.ez-wheel.com/en/development-kit-for-agv-and-amr) but we cannot do that. Because we cannot access to `packages.ez-wheel.com:8081`. When we try to access, the timeout occurs.

Is `packages.ez-wheel.com:8081` still alive? If URL has been changed, could you share it?

ez-Support, 2025-05-20:
Did you add ez-Wheel repository to your Apt sources configuration file as sudo in: `/etc/apt/sources.list`?
Please see instructions [here](https://github.com/IDEC-ezWheel/swd_ros_controllers?tab=readme-ov-file#method-1-ubuntu-package-manager) 

Or copy/paste in your teminal: 
echo "deb http://packages.ez-wheel.com:8081/ubuntu/dists/**focal** main" | sudo tee -a /etc/apt/sources.list
Where **[focal](http://packages.ez-wheel.com:8081/ubuntu/dists/)** depends on your Ubuntu version (focal, jammy, noble) :

ar90n, 2025-05-21:
@ez-Support 
Thanks for your reply. I can access to that host. It seems that my issue was caused by network environment.

closed 2025-05-21 by ar90n